dotintent/react-native-ble-plx #1080: Connection Error 2 "Operation was cancelled" on IOS16

Triple-gold, 2023-08-25:
Hi,
I found the one issue, on the IOS 15, i use **device.connect()** to connect the device, there is fast working, usually only takes a few seconds.

but, on the IOS16, **device.connect()**  will return the Error  **'Connect Error', '{"message":"Operation was cancelled","errorCode":2,"attErrorCode":null,"iosErrorCode":null,"androidErrorCode":null,"reason":null,"name":"BleError"}', [BleError: Operation was cancelled]**

I have tried add **device.cancelConnection()** before the **device.connect()** to cancel the  pending connection attempt. Situation has improved but also will return the error **Operation was cancelled**

What causes the connect device fail in IOS16?

Skytuna, 2023-08-25:
Facing the same issue right now. There must be at least some workaround I guess?

dominik-czupryna-withintent, 2023-09-29:
The new major release updates the core packages that support the latest versions of the operating system and fixes some old bugs including new native API for iOS. Please confirm if the issue still persists.

JoaoNM, 2023-10-20:
I'm also getting this error, trying things on iOS 17.

lucasrossell, 2023-10-20:
also getting this on iOS 16

JoaoNM, 2023-10-21:
I used device.cancelConnection(), which did help, but now I’m getting another issue where the connection unexpectedly drops.

Triple-gold, 2023-11-01:
> I'm also getting this error, trying things on iOS 17.

> also getting this on iOS 16

> I used device.cancelConnection(), which did help, but now I’m getting another issue where the connection unexpectedly drops.

Everyone, the following solutions are provided for your reference. While it may not solve the problem at its root, it can help you temporarily solve Bluetooth Connection Error 2 "Operation was cancelled"  issues. I also hope that someone can provide a fix based on the following solutions.

My background：
1. Use React Native
2. Use react-native-ble-plx connect device
3. All time keep scan device
4. Where try connect device in IOS16^，connect will connect and disconnect in one time and forward error "Operation was cancelled"
5. Use device.connect() to connect devices.

Based on my usage scenario, prior to iOS 15, I was using **async** to directly connect the desired device with **device.connect()**. However, starting from iOS 16, I encountered the error **"Operation was cancelled"**. During my continuous search for information, I added **device.cancelConnection()** before **device.connect()**, which increased the success rate of the connection but was still not satisfactory.

Later on, I used **async** to execute them, which significantly improved the success rate, although the **"Operation was cancelled"** error still occurred. After numerous attempts, I discovered that if I retry the connection when the **"Operation was cancelled"** error occurs again in the above solution, it no longer produces an error!

I apologize for not being able to provide my original code due NDA, but below is the execution flow of my program:

**async** run
     **device.cancelConnect()**
     **async** run
          **device.connect()**
if !**device.isConnected()**
     pass
else
     **device.connect()**

I understand that this solution may raise some confusion, and I cannot provide extensive interpretation. However, it is indeed a real working solution.

closed 2024-01-02 by Triple-gold