dotintent/react-native-ble-plx #1085: It is well known that BLE and React Native Expo does not mix

jdcrunchman, 2023-09-18:
## Prerequisites

Please answer the following questions for yourself before submitting an issue. **YOU MAY DELETE THE PREREQUISITES SECTION.**

- [ ] I am running the latest version
   I'm running Mac OS 13.5.2.    My package.json file is here....

{
  "name": "ble_test",
  "version": "1.0.0",
  "main": "AppEntry.js",
  "scripts": {
    "start": "expo start",
    "android": "expo start --android",
    "ios": "expo start --ios",
    "web": "expo start --web"
  },
  "dependencies": {
    "expo": "~49.0.10",
    "expo-status-bar": "~1.6.0",
    "react": "18.2.0",
    "react-native": "0.72.4",
    "react-native-ble-manager": "^10.1.3",
    "react-native-ble-plx": "^2.0.3",
    "react-native-bluetooth-serial": "^1.0.0-rc1",
    "react-native-flatlist": "^0.0.0",
    "react-native-web": "^0.19.8",
    "expo/build/launch": "~40.0.0"
  },
  "devDependencies": {
    "@babel/core": "^7.20.0"
  },
  "private": true
}

This lists the versions of each of the Libraries in my React Project.

- [ ] I checked the documentation and found no answer - not only that,  but this issue was mentioned in a YouTube video - https://youtu.be/ssokvToiYU0?si=rNQPQUu6VWsqrkc3
which clearly states that React native Expo does not support BLE.

- [ ] I checked to make sure that this issue has not already been filed
I glanced through some of the issues and some are related to this issue.

- [ ] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow] - it was posted there.
- [ ] 
(https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active) or on our [Gitter](https://gitter.im/RxBLELibraries/react-native-ble) channel....  this issue appeared in both forums mentioned.

## Question

Please describe the question.    My main question,   is this issue getting addressed,  and when is it estimated it can be resolved.   Who is working on this issue,  so I can provide them with some insight on the issue.

The Youtube Link above addressed this issue,  and shows a workaround that is beyond my pay grade to implement.   I'm only a novice React native developer and it goes into sophisticated XCode workarounds,   lots and lots of Json files have to be edited,  any one of which can cause my code not to work.    I have great need for BLE to run under React Expo,  and in XCode hell trying to reduce the thousands of errors I'm getting.

I hope you developers can resolve this issue soon,   and are they addressing this issue now?

Thanx
John Draper

dominik-czupryna-withintent, 2023-09-29:
As far as I know local prebuild is necessary to use custom code in Expo Go and our package is fully dependent on native code. Check https://www.npmjs.com/package/@config-plugins/react-native-ble-plx

dominik-czupryna-withintent, 2023-11-02:
Expo support has been added to the library, but local prebuild is still required.

closed 2023-11-02 by dominik-czupryna-withintent