dotintent/react-native-ble-plx #1101: Question : Android minSdkVersion

rvasseur31, 2023-09-29:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

What causes the minSdkVersion on Android to be set to 23?

dominik-czupryna-withintent, 2023-09-29:
@gmiszewski-intent

bj97301, 2023-09-29:
Need to update your expo client to 49.  This fixed it for me on v3 of this package.

bj97301, 2023-09-29:
I upped my expo to 49 but still see this error as well on Android:
[image]

eddyv19, 2023-10-03:
You need to use expo-build-properties and set the minSdkVersion in your app.json file. This will make the problem go away.  I had the same issue.

bj97301, 2023-10-03:
> You need to use expo-build-properties and set the minSdkVersion in your app.json file. This will make the problem go away. I had the same issue.

Gonna try this soon. Thank you very much ❤️❤️❤️

bj97301, 2023-10-03:
This seems to have done the trick.  Thank you @eddyv19.  

```bash
npx expo install expo-build-properties
```

```js
   plugins: [
      [
        'expo-build-properties',
        {
          android: {
            minSdkVersion: 23,
          },
        },
      ],
```

bj97301, 2023-10-03:
@dominik-czupryna-withintent want me to update the expo installation docs to include this?

dominik-czupryna-withintent, 2023-10-04:
@bj97301  We are currently checking the lib compatibility with downgraded minSDK to version 21, so it is possible that we will solve this problem in a different way (version 21 is currently the default version for the react-native template).

bj97301, 2023-10-10:
> @bj97301 We are currently checking the lib compatibility with downgraded minSDK to version 21, so it is possible that we will solve this problem in a different way (version 21 is currently the default version for the react-native template).

Got it.  Welp, let me know if I can help.  I love this package and what you guys are doing.

dominik-czupryna-withintent, 2023-10-18:
minSdkVersion has been lowered to 21

closed 2023-10-18 by dominik-czupryna-withintent