dotintent/react-native-ble-plx #1102: Crash App when open

HarryDoan, 2023-10-03:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

**This is my code:**
```
import React, { useState, useEffect } from 'react';
import { View, Text, Button, FlatList } from 'react-native';
import { BleManager } from 'react-native-ble-plx';

const BleScanner = () => {
  const [devices, setDevices] = useState([]);
  const [scanning, setScanning] = useState(false);
  const manager = new BleManager();

  useEffect(() => {
    // Request Bluetooth permissions on mount
    manager.startDeviceScan(null, null, (error, peripheral) => {
      if (error) {
        console.error(error);
        return;
      }
      handleDiscoverPeripheral(peripheral);
    });

    return () => {
      // Stop scanning and clean up on unmount
      manager.stopDeviceScan();
    };
  }, []);

  const startScan = () => {
    if (!scanning) {
      setDevices([]);
      manager.startDeviceScan(null, null, (error, peripheral) => {
        if (error) {
          console.error(error);
          return;
        }
        handleDiscoverPeripheral(peripheral);
      });
      setScanning(true);
    }
  };

  const stopScan = () => {
    if (scanning) {
      manager.stopDeviceScan();
      setScanning(false);
    }
  };

  const handleDiscoverPeripheral = (peripheral) => {
    const existingDevices = [...devices];
    const deviceIndex = existingDevices.findIndex((device) => device.id === peripheral.id);

    if (deviceIndex === -1) {
      existingDevices.push(peripheral);
      setDevices(existingDevices);
    }
  };

  return (
    <View>
      <Text>BLE Devices:</Text>
      <FlatList
        data={devices}
        keyExtractor={(item) => item.id}
        renderItem={({ item }) => (
          <View>
            <Text>Name: {item.name || 'N/A'}</Text>
            <Text>UUID: {item.id}</Text>
            <Text>RSSI: {item.rssi}</Text>
          </View>
        )}
      />
      <Button title="Start Scan" onPress={startScan} disabled={scanning} />
      <Button title="Stop Scan" onPress={stopScan} disabled={!scanning} />
    </View>
  );
};

export default BleScanner;
```
**I set in my Info.split file to permission Bluetooth scan, but i don't know why my app always crashes**

simonbothen-radinn, 2023-10-04:
You are re-creating a `BleManager` every rerender. 

Not sure if this will solve your issue, but it might be worth a try to cache it.

HarryDoan, 2023-10-04:
I came across this code example in the library repository. But it doesn't work in real IOS devices.

dominik-czupryna-withintent, 2023-10-05:
When app crashes, xCode shows why it happened, can you show these logs? The example application has been tested on multiple devices in debug and release mode and so far no device has had any problems. Could you tell me more about the device it doesn't work on?

HarryDoan, 2023-10-05:
i solved this error, cause i not define permissions in Info.split file.

closed 2023-10-05 by HarryDoan

RaulFaife, 2024-12-13:
> i solved this error, cause i not define permissions in Info.split file.

Which permissions were missing? I believe I'm having a similar issue