dotintent/react-native-ble-plx #1103: 🐛 Android crashes on bluetooth device disconnect

bj97301, 2023-10-03:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

shouldnt crash

### Current Behavior

crashes

### Library version

3.0.0

### Device

pixle 7a

### Environment info

```shell
see screenshot
```

### Steps to reproduce

1. connect to device
2. disconnect(i just shut down the power on the Bluetooth device)
3. crashes
env info:
[image]

crash:
[image]

### Formatted code sample or link to a repository

```js
sorry let me know if you really need that but I think the stack trace should be good nuff right?
```

### Relevant log output

```shell
see screenshots
```

bj97301, 2023-10-03:
Let me know if you need anything else.  Love the package.  Keep up the amazing work 🤘

andycamp, 2023-10-03:
We are also getting this exception after upgrading to 3.0.0. Thanks.

contactsimonwilson, 2023-10-05:
I'm also facing the same issue

eliaslecomte, 2023-10-06:
I'm having a similar crash but just on connect, after something like 10 seconds 🤔.

lucaswitch, 2023-10-06:
I can confirm that it's happening on 3.0.0 on Android 12 just like @eliaslecomte says after 5~6 seconds after monitor.

lucaswitch, 2023-10-06:
The solution is downgrade to latest version before 3.0.0 on Android.

lucaswitch, 2023-10-06:
Here is the repro.
Happening when monitor characteristics at same time. (Don't know if it's related).
```js
// Monitor
device.monitorCharacteristicForService(
      'some service id',
      'some characteristic id',
      handleOnCharacteristicChanged
)

device.monitorCharacteristicForService(
      'some service id',
      'some other characteristic id',
      handleOnCharacteristicChanged
)
// the after some  moments seconds the crash happens.
```
Here is the full stack trace of the error?
[image]

bj97301, 2023-10-08:
> Here is the repro. Happening when monitor characteristics at same time. (Don't know if it's related).
>
> ```js
> // Monitor
> device.monitorCharacteristicForService(
>       'some service id',
>       'some [...]

Interesting. I’ll see if it doesn’t happen to me if I don’t monitor shits.

lucaswitch, 2023-10-09:
Seems to be a problem with the RXJava:

On Android side we use the [RxAndroidBle](https://github.com/Polidea/RxAndroidBle) library of Polidea. After migration towards RxJava 2 some of the errors are not routed properly to their listeners and thus this will result in a BLE Undeliverable Exception. The root cause lies in the threading of the Android OS. As workaround RxJava has a hook where you can set the global errorhandler. For more info see [RxJava docs](https://github.com/ReactiveX/RxJava/wiki/What's-different-in-2.0#error-handling) .

A default workaround implementation in the Flutter app (needs to be in the Java / Kotlin part e.g. mainactivity) is shown below. For an example (in Java) see Polidea RxAndroidBle [sample](https://github.com/Polidea/RxAndroidBle/tree/master/sample/src/main/java/com/polidea/rxandroidble2/sample).

BleException is coming from Polidea RxAndroidBle, so make sure your application declares the following depedency: implementation "com.polidea.rxandroidble2:rxandroidble:1.11.1"

RxJavaPlugins.setErrorHandler { throwable ->
  if (throwable is UndeliverableException && throwable.cause is BleException) {
    return@setErrorHandler // ignore BleExceptions since we do not have subscriber
  }
  else {
    throw throwable
  }
}

bj97301, 2023-10-10:
@dominik-czupryna-withintent Any thoughts on this one?  I am hoping to test out your 3.0.0 package but this is blocking me at the moment.

dominik-czupryna-withintent, 2023-10-10:
Currently, we have this error reproduced and described, waiting for fix. We plan to release a fix this or next week.

lucaswitch, 2023-10-10:
Apparently is caused by RxJava, the events need to have always a subscriber or it will lead to undeliverable exception crashes.

closed 2023-10-16 by gmiszewski-intent

eliaslecomte, 2023-10-17:
So it now no longer crashes with the 3.1.0 release but I am loosing connection after 4/5 seconds on Android.

simonitfi, 2023-10-19:
Also it seems onDisconnected is not called on Android. https://github.com/dotintent/react-native-ble-plx/issues/1126

andycamp, 2023-10-19:
Confirming that onDisconnect is not called on android.

eliaslecomte, 2023-10-30:
Should this be reopened or should I report a new issue?

commit d2cf6454 (2023-10-10): fix: resolves #1103 - crash on disconnect
commit 95ae38b7 (2023-10-10): fix: resolves #1103 - crash on disconnect
commit c1cdb036 (2023-10-16): fix: resolves #1103 - crash on disconnect (#1111)

* chore: recreate native error
* test: android device disconnecting example
* fix: resolves #1103 - crash on disconnect
* refactor: sorted imports
* refactor: types
* fix: fast reuse of connect to device leads to 'device not connected error'
* fix: fast reuse of connect to device leads to 'device not connected error' (only for Android)
* refactor: change numbers into variables