dotintent/react-native-ble-plx #1104: Characteristic value which is received by monitorCharacteristicForDeviceQuestion, is cannot decrypt.

AzranAzwer, 2023-10-06:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

Characteristic value which is received by monitorCharacteristicForDeviceQuestion is chunked into 4-5 based on api response of bluetooth terminal device. So i need to concatenate to one. if i concatenate, its giving me wrong decryption data.

The value i am getting from BLE is, more than its size.
i ll post sample response below.

`AkDA6Z1tuxi+4svJtHEEWaixsCcIPTAPq1wC3/8SJ0DK6v/mHXjtvGAXoZirGcJrtpMp+ahVdpTSDdB3kpqvC7qoXMGYFfsw2Gtap56Uml5teBneZASl7PvT4kULOjvdeOFX0QlK79u5s07YeUXR/ZYx+i6hHXYUoEQDE4nAZA==`
`AkDA6Z1tuxi+4svJtHEEWaixsCcIPTAPq1wC3/8SJ0DK6v/mHXjtvGAXoZirGcJrtpMp+ahVdpTSDdB3kpqvC7qoXMGYFfsw2Gtap56Uml5teBneZASl7PvT4kULOjvdeOFX0QlK79u5s07YeUXR/ZYx+i6hHXYUoEQDE4nAZA==`
`AkDA6Z1tuxi+4svJtHEEWaixsCcIPTAPq1wC3/8SJ0DK6v/mHXjtvGAXoZirGcJrtpMp+ahVdpTSDdB3kpqvC7qoXMGYFfsw2Gtap56Uml5teBneZASl7PvT4kULOjvdeOFX0QlK79u5s07YeUXR/ZYx+i6hHXYUoEQDE4nAZA==`
`AkDA6Z1tuxi+4svJtHEEWaixsCcIPTAPq1wC3/8SJ0DK6v/mHXjtvGAXoZirGcJrtpMp+ahVdpTSDdB3kpqvC7qoXMGYFfsw2Gtap56Uml5teBneZASl7PvT4kULOjvdeOFX0QlK79u5s07YeUXR/ZYx+i6hHXYUoEQDE4nAZA==`
`AkDA6Z1tuxi+4svJtHEEWaixsCcIPTAPq1wC3/8SJ0DK6v/mHXjtvGAXoZirGcJrtpMp+ahVdpTSDdB3kpqvC7qoXMGYFfsw2Gtap56Uml5teBneZASl7PvT4kULOjvdeOFX0QlK79u5s07YeUXR/ZYx+i6hHXYUoEQDE4nAZA==`

so How should i concatenate the response? 
If i cannot concatenate, how should i receive all data at once or how should i achieve my goal ?

AzranAzwer, 2023-10-06:
@dominik-czupryna-withintent could you please look at this one ?

dominik-czupryna-withintent, 2023-10-06:
To be honest I’m not sure what you want to do. I have no idea what type of data you are sending. The value of your characteristic can only be one at a time. So when you are watching characteristic notifications you will always get this one value. Each of these values has to be decoded. So you are getting the characteristic value it is base64 encoded, so you are decoding it. Your response has been changed from `AkDA6Z1tuxi+4svJtHEEWaixsCcIPTAPq1wC3/8SJ0DK6v/mHXjtvGAXoZirGcJrtpMp+ahVdpTSDdB3kpqvC7qoXMGYFfsw2Gtap56Uml5teBneZASl7PvT4kULOjvdeOFX0QlK79u5s07YeUXR/ZYx+i6hHXYUoEQDE4nAZA==`
into something like
```
Ðw¯G  @Àé¾âËÉ´qY¨±°=0«\ßÿ'@Êêÿæxí¼`¡Âk¶)ù¨UvÒ
   º¨\Á0ØkZ§^mxÞd¥ìûÓâE
                       :;ÝxáWÑ  JïÛ¹³NØyEÑý1ú.¡v DÀd
```

So probably it's not the output that you are expecting. I’m guessing you are sending the uint8 array. So in that case you have to parse this value one more time into uint8! Something like
```
    Uint8Array.from(
      base64.decode(
   'AkDA6Z1tuxi+4svJtHEEWaixsCcIPTAPq1wC3/8SJ0DK6v/mHXjtvGAXoZirGcJrtpMp+ahVdpTSDdB3kpqvC7qoXMGYFfsw2Gtap56Uml5teBneZASl7PvT4kULOjvdeOFX0QlK79u5s07YeUXR/ZYx+i6hHXYUoEQDE4nAZA=='
      ),
      char => String(char).charCodeAt(0) // parse char into numer
    )
```
And you will get an array like:
[2, 64, 192, 233, 157, 109, 187, 24, 190, 226, 203, 201, 180, 113, 4, 89, 168, 177, 176, 39, 8, 61, 48, 15, 171, 92, 2, 223, 255, 18, 39, 64, 202, 234, 255, 230, 29, 120, 237, 188, 96, 23, 161, 152, 171, 25, 194, 107, 182, 147, 41, 249, 168, 85, 118, 148, 210, 13, 208, 119, 146, 154, 175, 11, 186, 168, 92, 193, 152, 21, 251, 48, 216, 107, 90, 167, 158, 148, 154, 94, 109, 120, 25, 222, 100, 4, 165, 236, 251, 211, 226, 69, 11, 58, 59, 221, 120, 225, 87, 209, 9, 74, 239, 219, 185, 179, 78, 216, 121, 69, 209, 253, 150, 49, 250, 46, 161, 29, 118, 20, 160, 68, 3, 19, 137, 192, 100]
And this is what you should concat with the next characteristic values.

Is this what you are asking or have I misunderstood you?

dominik-czupryna-withintent, 2023-10-18:
Does the problem still occurs?

closed 2023-11-02 by dominik-czupryna-withintent

priyasamyal, 2023-12-13:
@AzranAzwer  Any update on this. I am also getting this. kind of response
BmwBAP/aBwEBAR0AAA==

Don't know how to parse this. Kindly help

@dominik-czupryna-withintent