dotintent/react-native-ble-plx #1117: Unable to resolve module @babel/runtime/helpers/interopRequireDefault from🐛

Kralg, 2023-10-13:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

I am trying the example project to test your library.

### Current Behavior

I am getting error when npm start and pressing a (android): I am getting: error: Error: Unable to resolve module @babel/runtime/helpers/interopRequireDefault from

### Library version

3.0.0

### Device

Samsung s23 android

### Environment info

```shell
info Fetching system and libraries information...
System:
  OS: Windows 10 10.0.22621
  CPU: (12) x64 12th Gen Intel(R) Core(TM) i7-1255U
  Memory: 3.30 GB / 31.64 GB
Binaries:
  Node:
    version: 18.17.0
    path: C:\Program Files\nodejs\node.EXE
  Yarn:
    version: 1.22.15
    path: C:\Program Files\nodejs\yarn.CMD
  npm:
    version: 10.2.0
    path: C:\Program Files\nodejs\npm.CMD
  Watchman: Not Found
SDKs:
  Android SDK: Not Found
  Windows SDK: Not Found
IDEs:
  Android Studio: AI-222.4459.24.2221.9971841
  Visual Studio:
    - 17.6.33829.357 (Visual Studio Community 2022)
Languages:
  Java: Not Found
  Ruby: Not Found
npmPackages:
  "@react-native-community/cli": Not Found
  react:
    installed: 18.2.0
    wanted: 18.2.0
  react-native:
    installed: 0.72.4
    wanted: 0.72.4
  react-native-windows: Not Found
npmGlobalPackages:
  "*react-native*": Not Found
Android:
  hermesEnabled: true
  newArchEnabled: false
iOS:
  hermesEnabled: Not found
  newArchEnabled: Not found
```

### Steps to reproduce

1. Pull git repo.
2. npm start
3. Press a to start on Android.

### Formatted code sample or link to a repository

```js
https://github.com/dotintent/react-native-ble-plx
```

### Relevant log output

```shell
error: Error: Unable to resolve module @babel/runtime/helpers/interopRequireDefault from D:\AK\Pikotera\IN17\App\ReactApp_\ble_plx\react-native-ble-plx\src\Characteristic.js: @babel/runtime/helpers/interopRequireDefault could not be found within the project.
> 1 | // @flow
  2 | 'use strict'
  3 |
  4 | import type { BleManager } from './BleManager'
```

dominik-czupryna-withintent, 2023-10-13:
Have you installed the dependency both in exemple and in the main directory?

Kralg, 2023-10-14:
No, yes the solution was to move back to the main git directory and install dependencies.
The app now launches. Just some comments: when it launches I get white background and white text in the devices list so only rectagles of each element appears. I had to change background to see devices names and ids.
However after pressing on a device that definitely has services and characteristics, in the new window I see null but name, mtu and some other parameters seems to be correct. Can you advice on what problem can it be? This device is nrf ble serial uart services device.

closed 2023-10-17 by abursztynowska