dotintent/react-native-ble-plx #1124: How can I scan external keyboard

HarunKilic, 2023-10-17:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

Product: [Eyoyo EY-016](https://www.eyoyousa.com/eyoyo-ey-016-2d-finger-ring-barcode-scanner-mini-wearable-3-in-1-usb-wired-amp-24g-wireless-amp-bluetooth-scanner-image-1d-qr-bar-code-reader-pdf417-data-matrix-screen-scan-for-ipad-smartphone-pc-p00212p1.html)
I can find this device in native bluetooth search on samsung device. 
But I can't find it with this library, it has bluetooth 4.1 compability.

dominik-czupryna-withintent, 2023-10-18:
Can you show GATT server of this device? You can check it using an app like nRF Connect and conning to the device (https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en&gl=US).

dominik-czupryna-withintent, 2023-10-18:
Did you check `Ask for permissions` section in https://dotintent.github.io/react-native-ble-plx/ ?

HarunKilic, 2023-10-18:
Okay after some debugging, I was using wrong service uuid. Is there a way to identify the "right" service?

closed 2023-10-23 by dominik-czupryna-withintent