dotintent/react-native-ble-plx #1126: 🐛 onDisconnected callback is newer triggering

simonitfi, 2023-10-19:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

//monitor disconnections
deviceConnection.onDisconnected((error, disconnectedDevice) => {
    console.log('onDisconnected Device disconnected: ' + disconnectedDevice.id);
});

onDisconnected should fire when disconnect occures.

### Current Behavior

New get called onDisconnected.

### Library version

3.1.0

### Device

Android

### Environment info

```shell
-
```

### Steps to reproduce

-

### Formatted code sample or link to a repository

```js
-
```

### Relevant log output

```shell
Only catch I got is from monitor characteristics:
monitorCharacteristicForDevice catch [BleError: Device AB:BC:DD:EF:12:34 was disconnected] Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 8 (GATT_INSUF_AUTHORIZATION or GATT_CONN_TIMEOUT)
```

### Additional information

Worked on 2.x.x versions.

Is this working on other users on 3.1.0? Please

andycamp, 2023-10-19:
I can confirm we are experiencing the same issue with onDisconnect it never called on android, but it's called on ios.

andycamp, 2023-10-19:
We have also seen this issue on 3.0.0. Works fine on 2.x.x versions.

lucas-bortoli, 2023-10-23:
I'm having the same issue. The callback is never called with version 3.0.0 on Android.

simonitfi, 2023-10-26:
Yes the library version 3.x is at the moment pretty useless until these critical bugs are fixed.

dominik-czupryna-withintent, 2023-10-26:
Yes, the error has been confirmed, it occurs when disconnection occurs for a reason other than calling `disconnect` from `BleManager` (like Device side disconnect). And it's probably related to https://github.com/dotintent/react-native-ble-plx/issues/1103 . We are currently looking for a solution to the problem.

dominik-czupryna-withintent, 2023-10-30:
Hi, we have changed the trigger of the disconnect action and released an RC version (https://www.npmjs.com/package/react-native-ble-plx/v/3.1.2-rc.0), could you check in your cases whether the change solves your problem?

andycamp, 2023-10-30:
onDisconnect is triggered for us on the rc version.

agutierrezgit, 2023-11-01:
Hi @dominik-czupryna-withintent - I was having this issue and at least in my case the RC version fixed it. Thanks a lot

lucas-bortoli, 2023-11-01:
@dominik-czupryna-withintent  The RC version fixed the problem. The onDisconnected callback is now triggering when the peripheral device disconnects. Thanks a lot!

simonitfi, 2023-11-01:
Also I could confirm that with quick tests I have done the onDisconnected callback is now triggering.

angelos3lex, 2023-11-09:
So far so good for us too.

closed 2023-11-29 by dominik-czupryna-withintent