dotintent/react-native-ble-plx #1130: [Android 14] Can not reconnect after lost connection

HoangThiKimPhung, 2023-10-30:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

Hi everyone,
My App has been working well but not on Android 14
When lost connection, I tried to disconnect and connect again but it still didn't work
Only work when I unpair it before start connect again. It's not good for user when the prompt pairing appear continuously
Hope someone might have an idea for this

`"reason":"Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 19 (GATT_CONN_TERMINATE_PEER_USER)"`
`Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 8 (GATT_INSUF_AUTHORIZATION or GATT_CONN_TIMEOUT`

dominik-czupryna-withintent, 2023-10-30:
We have changed the connect events process a bit. Could you check if the error also occurs in the version https://www.npmjs.com/package/react-native-ble-plx/v/3.1.2-rc.0 ?

bj97301, 2023-12-08:
Seems to work as long as i make a new build and update all my packages.  @dominik-czupryna-withintent you rock!  Thank you.

milobob, 2023-12-11:
OK I have a stupid question. If my device goes out of range in the background, I want it to reconnect. I assume the 
```
  disconnectSubscription = bleManager.onDeviceDisconnected(device.id, async(error, data) => {
    connectBLEDevice();
  });

``` 
will be fired and the `connectBLEDevice()` function will be executed... which calls in my case, among other things, the `bleManager.connectToDevice(...)` function.
 
 So, that should take care of reconnecting in the background right? Do I need to "scan" more than once? or should this function:
```
 bleManager.startDeviceScan(
      localServiceIds,..)

```      
only be called once? (e.g. at the first time connecting?)
So my question is: call the startDeviceScan only once at first time connecting, or every time a device goes out of range? This is a basic question about the difference between scanning and reconnecting using connectToDevice. I know it's a basic question but I'm sure I'm not the only one wondering!

HoangThiKimPhung, 2023-12-11:
I’ve the same question with @milobob

milobob, 2023-12-11:
> I’ve the same question with @milobob

That's reassuring to hear that I'm not the only one with this question! I have read the react-native-ble-plx documentation so many times, and poured over the internet to try to find an answer, but I think maybe I'm just missing the forest for the trees on this basic question! It could be that react-native-ble-plx is so advanced that it can even scan for peripherals in the background (I'm sure some people need this - although most don't!)... but the explanation of the more basic question of how to reconnect in the background is not super clear. Maybe some more clarity in the documentation about how to reconnect in the background for iOS and Android for a single peripheral would be helpful since I imagine this is the primary use-case. Or it could be that the scan in the background is necessary to maintain a connection. It makes intuitive sense that if the connection to the peripheral is lost, we need to "scan" around to find it, but intuition doesn't always match reality. So I'm really not sure!

intent-kacper-cyranowski, 2024-07-17:
Hi everyone

We have added a tutorial section that describes how to use [`onDeviceDisconnected`](https://dotintent.github.io/react-native-ble-plx/#monitoring-device-connection), if you find something ambiguous or missing, let us know, and we will try to cover it. There is no need to use `startDeviceScan` if you already have the device to which you want to connect.

Background mode is not supported at this moment. We have a [guide for iOS](https://github.com/dotintent/react-native-ble-plx/wiki/Background-mode-(iOS)), but nothing for android. We are planning to add support for that. 

As the issue seems to be resolved, I will close this one. In case of further questions, please open a separate issue.

closed 2024-07-17 by intent-kacper-cyranowski