dotintent/react-native-ble-plx #1133: Project Setup

syncre02, 2023-11-11:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Library version

3.1.1

### Platform

Android

### Setup Description

Getting the createClient is null error. I followed the documentation as seen on the website but I still have the same issue.

TypeError: Cannot read property 'createClient' of null, js engine: hermes
at node_modules\react-native\Libraries\Core\ExceptionsManager.js:null in reportException
at node_modules\react-native\Libraries\Core\ExceptionsManager.js:null in handleException
at node_modules\react-native\Libraries\Core\setUpErrorHandling.js:null in handleError
at node_modules\@react-native\polyfills\error-guard.js:null in ErrorUtils.reportFatalError
at node_modules\metro-runtime\src\polyfills\require.js:null in guardedLoadModule
at node_modules\metro-runtime\src\polyfills\require.js:null in metroRequire
at http://10.0.2.2:19001/node_modules/expo/AppEntry.bundle?platform=android&dev=true&minify=false&app=com.syncre.windowblinds&modulesOnly=false&runModule=true:null in global

### Steps to Reproduce

1. creating new BLEManager just creates the error

### Relevant log output

```shell
TypeError: Cannot read property 'createClient' of null, js engine: hermes
at node_modules\react-native\Libraries\Core\ExceptionsManager.js:null in reportException
at node_modules\react-native\Libraries\Core\ExceptionsManager.js:null in handleException
at node_modules\react-native\Libraries\Core\setUpErrorHandling.js:null in handleError
at node_modules\@react-native\polyfills\error-guard.js:null in ErrorUtils.reportFatalError
at node_modules\metro-runtime\src\polyfills\require.js:null in guardedLoadModule
at node_modules\metro-runtime\src\polyfills\require.js:null in metroRequire
at http://10.0.2.2:19001/node_modules/expo/AppEntry.bundle?platform=android&dev=true&minify=false&app=com.syncre.windowblinds&modulesOnly=false&runModule=true:null in global
```

### Contents of the `package.json` file

{
  "name": "expo-ble",
  "version": "1.0.0",
  "main": "node_modules/expo/AppEntry.js",
  "scripts": {
    "start": "REACT_NATIVE_PACKAGER_HOSTNAME=192.168.0.141 expo start --dev-client",
    "android": "expo start --android",
    "ios": "expo start --ios",
    "web": "expo start --web"
  },
  "dependencies": {
    "@config-plugins/react-native-ble-plx": "^5.0.0",
    "@expo/react-native-action-sheet": "^4.0.1",
    "@react-navigation/bottom-tabs": "^6.4.1",
    "@react-navigation/drawer": "^6.5.1",
    "@react-navigation/native": "^6.0.14",
    "@react-navigation/native-stack": "^6.9.2",
    "@shopify/react-native-skia": "0.1.171",
    "axios": "1.2.0",
    "expo": "47.0.8",
    "expo-auth-session": "~3.8.0",
    "expo-crypto": "~12.0.0",
    "expo-dev-client": "~2.0.1",
    "expo-image-manipulator": "~11.0.0",
    "expo-location": "~15.0.1",
    "expo-media-library": "~15.0.0",
    "expo-random": "~13.0.0",
    "expo-secure-store": "~12.0.0",
    "expo-status-bar": "1.4.2",
    "jwt-decode": "^3.1.2",
    "luxon": "^3.2.0",
    "react": "18.1.0",
    "react-dom": "18.1.0",
    "react-native": "0.70.5",
    "react-native-gesture-handler": "~2.8.0",
    "react-native-reanimated": "~2.12.0",
    "react-native-redash": "^18.1.0",
    "react-native-safe-area-context": "4.4.1",
    "react-native-screens": "~3.18.2",
    "react-native-vision-camera": "^2.15.2",
    "twrnc": "^3.3.2",
    "vision-camera-code-scanner": "^0.2.0",
    "vision-camera-ocr": "^1.0.0",
    "react-native-ble-plx": "^2.0.3"
  },
  "devDependencies": {
    "@babel/core": "^7.12.9"
  },
  "private": true
}

garciavalter, 2023-11-13:
Since you are using expo did you follow these instructions? https://github.com/dotintent/react-native-ble-plx#expo-sdk-43

Remember this part that it's one of most important: 

`Then you should build the version using native modules (e.g. with npx expo prebuild command). And install it directly into your device with npx expo run:android.`

dominik-czupryna-withintent, 2023-11-13:
@garciavalter is right but
@syncre02 
Could you describe your installation process step by step? Maybe I missed something in the documentation.

syncre02, 2023-11-25:
Nevermind, the issue was that I didn't initalize it properly (I have to ask for premisisons first).
Sorry for the delay on the reply.

closed 2023-11-25 by syncre02

bj97301, 2023-12-05:
The permissions issue wasn't the fix for me.  I needed to rebuild the app and install a new expo dev client thingy on my phone with the latest package.

CyxouD, 2024-02-15:
thank you, @bj97301! My project is not an Expo project, but I also experienced this issue and it was resolved only after rebuilding app in Android Studio.