dotintent/react-native-ble-plx #1135: Inconsistency in connectToDevice Timeout Behavior Between iOS and Android

DevGeorge, 2023-11-15:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

We have encountered a significant inconsistency in the behavior of the timeout parameter in the connectToDevice method between iOS and Android platforms when using the react-native-ble-plx library.

Behavior:
On iOS, the timeout parameter functions as expected. It sets a limit on the duration for attempting a Bluetooth connection to a device.

On Android, however, the timeout parameter behaves differently and unexpectedly. Rather than acting solely as a connection attempt limiter, it appears to be functioning as a maximum duration for the connection itself. This means that once a connection is established, it automatically disconnects after the specified timeout duration. This behavior is inconsistent with the iOS implementation and the expected functionality of a connection timeout.

Steps to Reproduce:

Initialize the BLE manager and call the connectToDevice method with a timeout parameter on an Android device.
Observe that the device connects successfully but then disconnects after the specified timeout duration.
Perform the same steps on an iOS device and observe that the timeout only limits the connection attempt duration without affecting the connection's stability post-establishment.

silsergey, 2023-12-06:
I can confirm the issue (3.1.2) on Android (Samsung S23). If the timeout is not specified, the device will keep the connection.
When the timeout is specified, the connection drops after some seconds with `error: null` as if it was disconnected by cancelDeviceConnection

**A bit more information**
The communication is done by writeCharacteristicWithoutResponseForDevice and monitorCharacteristicForDevice. May be this is helpful. No issues for iOS

DaAitch, 2023-12-22:
We have the same problem 👍.

Seems like on Android the timeout-timer is not reset after successful device connection or so, so it will trigger a disconnect after the timeout no matter connected successfully or not. On iOS it works.

I checked https://github.com/dotintent/react-native-ble-plx/blob/master/android/src/main/java/com/bleplx/adapter/BleModule.java#L1263 and I believe the `Observable` is leaking and never resolves? Shouldn't it be like `connect.take(1).subscribe(...)` or something like this? (it's been years I was using rx, sorry 🙄).

closed 2024-07-31 by intent-kacper-cyranowski