dotintent/react-native-ble-plx #1136: How to improve throughput

TreeOfLearning, 2023-11-16:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

Hello,

We are currently seeing a throughput of around 3KB/s for receiving data from a monitored characteristic. I'm trying to figure out if there's anything I can do to speed this up. The suggestions I can find on the web are:

1. Enable DLE (data length extension)
2. Use the smallest possible connection interval 
3. Use the largest possible MTU
4. Use the LE 2M PHY mode 

I only seem to be able to control the MTU with react-native-ble-plx. 

Are any of these other options supported (perhaps automatically enabled?) by the library? Are there any other suggestions on how to improve the throughput?

Thanks

ethansisk, 2023-11-22:
https://developer.apple.com/library/archive/qa/qa1931/_index.html Apple has some pretty specific requirements for connection intervals/latency on connections made to their devices. I don't believe that anything from the Core Bluetooth library allows for manually altering many connection parameters, not sure about natively on Android. If you are in control of the code running on the peripheral that the phone is trying to connect to, you would be able to make tweaks that way.

aliberski, 2025-01-13:
The library does not provide direct control over DLE settings. However, modern BLE devices and operating systems typically enable DLE by default.

You can negotiate the MTU size using the [requestMtu](https://github.com/dotintent/react-native-ble-plx/blob/d6196f090394a26461e4c8d33d7dd7ec1140ebc6/src/Device.js#L135) or [requestMTUForDevice](https://github.com/dotintent/react-native-ble-plx/blob/d6196f090394a26461e4c8d33d7dd7ec1140ebc6/src/BleManager.js#L440) method. Keep in mind that the effective MTU will be the minimum of the requested size and the maximum size supported by both devices.

More info [HERE](https://github.com/dotintent/react-native-ble-plx/wiki/MTU-Negotiation)

closed 2025-01-13 by aliberski