dotintent/react-native-ble-plx #1138: 🐛 device.manufacturerData is undefined, however device.rawScanRecord has valid manufacturerData data.

supurloop, 2023-11-16:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

The device object should be populating the device.manufacturerData field with a Base64 encoded string that is the manufacturer data in the advertisement. This had been working on both iOS and Android with 2.0.3, but after upgrading to 3.1.1 only Android sets the device.manufacturerData field, on iOS it is undefined.

### Current Behavior

The device.manufacturerData should be a Base64 encoded string, it is undefined. I am certain that the received advertisement contains manufacturer data.  It works on Android, and even on iOS I can still correctly extract the missing device.manufacturerData from device.rawScanRecord field so I know the data is being received by iOS.  Again, this was working with 2.0.3 of the library on both iOS and Android, and is working on Android with 3.1.1 as well.

### Library version

3.1.1

### Device

iPhone 8/iOS 16.7.2

### Environment info

```shell
yarn info

{
  name: '@client',
  'dist-tags': {
    latest: '0.1.0'
  },
  versions: [
    '0.1.0',
  ],
  time: {
    '0.1.0': '2023-11-16T13:56:52Z'
  },
  description: 'Client',
  author: {
    name: 'Client',
    email: 'name@company.com',
    url: 'https://github.com/client'
  },
  license: 'UNLICENSED',
  homepage: 'https://github.com/client/client#readme',
  repository: {
    type: 'git',
    url: 'git+https://github.com/client/client.git'
  },
  bugs: {
    url: 'https://github.com/client/client/issues'
  },
  version: '0.1.0',
  dist: {
    integrity: 'sha512-',
    shasum: 'b4c849b787aafd9d',
    tarball: 'https://npm.pkg.github.com/download/@client/client/0.1.0/b4c8'
  },
  main: 'lib/commonjs/index',
  scripts: {
    format: 'prettier --write "src/**/*.{js,ts,tsx}" "*.js"',
    lint: 'eslint . --ignore-path .eslintignore --ext .js,.tsx,.ts',
    prepare: 'bob build',
    test: 'npm run typecheck && npm run lint',
    typecheck: 'tsc --noEmit'
  },
  dependencies: {
    '@client/ubjson': '^0.0.1',
    'aes-js': '^3.1.2',
    assert: '^1.5.0',
[... 140 lines cut ...]
        '303': 'Services not discovered for device {deviceID}',
        '400': 'Characteristic discovery failed for device {deviceID} and service {serviceUUID}',
        '401': 'Characteristic {characteristicUUID} write failed for device {deviceID} and service {serviceUUID}',
        '402': 'Characteristic {characteristicUUID} read failed for device {deviceID} and service {serviceUUID}',
        '403': 'Characteristic {characteristicUUID} notify change failed for device {deviceID} and service {serviceUUID}',
        '404': 'Characteristic {characteristicUUID} not found',
        '405': 'Characteristics not discovered for device {deviceID} and service {serviceUUID}',
        '406': 'Cannot write to characteristic {characteristicUUID} with invalid data format: {internalMessage}',
        '500': 'Descriptor {descriptorUUID} discovery failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}',
        '501': 'Descriptor {descriptorUUID} write failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}',
        '502': 'Descriptor {descriptorUUID} read failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}',
        '503': 'Descriptor {descriptorUUID} not found',
        '504': 'Descriptors not discovered for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}',
        '505': 'Cannot write to descriptor {descriptorUUID} with invalid data format: {internalMessage}',
        '506': 'Cannot write to descriptor {descriptorUUID}. It\'s not allowed by iOS and therefore forbidden on Android as well.',
        '600': 'Cannot start scanning operation',
        '601': 'Location services are disabled' },
     _scanEventSubscription: { remove: [Function: remove] } } }
```

### Additional information

The Base64 decode of the above log output shows that there is manufacturerData.  The value of the manufacturerData is valid for the particular device.

{"serviceData":null,"isConnectable":true,"txPowerLevel":null,"serviceUUIDs":["0000fca5-0000-1000-8000-00805f9b34fb"],"manufacturerData":"6gwBELDSeGVjnQ==","overflowServiceUUIDs":null,"localName":null,"solicitedServiceUUIDs":null}

MahmoudNafiseh, 2023-11-18:
This happens on our application too, downgraded to 2.0.3 and it works. I assume this is because of the switch to the MultiplatformBleAdapter library?

MahmoudNafiseh, 2023-11-21:
@KamilNatonek I tried your PR using patch-package and I don't think that fixes the issue

KamilNatonek, 2023-11-21:
@MahmoudNafiseh thanks for info, I will continue working on it

KamilNatonek, 2023-11-21:
@MahmoudNafiseh in the meantime can you provide your current `<ScannedPeripheral>` output ? Sample one

luca-tomasetti, 2023-11-22:
+1 same for me as well. 
I also tried your PR patch @KamilNatonek and I still received `undefined` in the manufacturerData

KamilNatonek, 2023-11-22:
@luca-tomasetti @MahmoudNafiseh Check this out -> https://github.com/dotintent/react-native-ble-plx/issues/605#issuecomment-580655596
I've added console logs to our example app and these are my logs confirming the theory from the link.

` LOG  Device name: MIScooter2526,Device manufacturerData: null`  -> first record
` LOG  Device name: MIScooter2526,Device manufacturerData: TkIuAgAAAM8=` -> second record

supurloop, 2023-11-22:
#605 is not the same problem I am reporting. On iOS with 3.1.1 manufactuerData is always undefined, not just the first time.

MahmoudNafiseh, 2023-11-22:
@KamilNatonek this is my output data:

```
device.manufacturerData: undefined
```

```
{ rawScanRecord: 'eyJpc0Nvbm5lY3RhYmxlIjp0cnVlLCJvdmVyZmxvd1NlcnZpY2VVVUlEcyI6bnVsbCwic2VydmljZURhdGEiOm51bGwsInNvbGljaXRlZFNlcnZpY2VVVUlEcyI6bnVsbCwibG9jYWxOYW1lIjoiQkFORCBWMiAzMCIsIm1hbnVmYWN0dXJlckRhdGEiOiJnc2JTWG5CVEowSDlZTExpVmowZ3FPNTB3ZWZ1SUp1TnZTXC9nZllKWkVRNkhaZz09IiwidHhQb3dlckxldmVsIjpudWxsLCJzZXJ2aWNlVVVJRHMiOlsiNTM0N2FiYTItNjBkNi00MmFhLTg4YjktMTNiZGU5OWE0NWE5Il19',
  id: '3DF4DD74-5027-1722-0970-AB439897BA03',
  rssi: -67,
  name: 'BAND V2 30',
  mtu: 23,
  _manager: 
   { _eventEmitter: 
      { _nativeModule: 
         { stopDeviceScan: { [Function: nonPromiseMethodWrapper] type: 'async' },
           requestConnectionPriorityForDevice: { [Function: promiseMethodWrapper] type: 'promise' },
           readRSSIForDevice: { [Function: promiseMethodWrapper] type: 'promise' },
           requestMTUForDevice: { [Function: promiseMethodWrapper] type: 'promise' },
           devices: { [Function: promiseMethodWrapper] type: 'promise' },
           connectedDevices: { [Function: promiseMethodWrapper] type: 'promise' },
           connectToDevice: { [Function: promiseMethodWrapper] type: 'promise' },
           cancelDeviceConnection: { [Function: promiseMethodWrapper] type: 'promise' },
           isDeviceConnected: { [Function: promiseMethodWrapper] type: 'promise' },
           discoverAllServicesAndCharacteristicsForDevice: { [Function: promiseMethodWrapper] type: 'promise' },
           servicesForDevice: { [Function: promiseMethodWrapper] type: 'promise' },
           characteristicsForDevice: { [Function: promiseMethodWrapper] type: 'promise' },
           characteristicsForService: { [Function: promiseMethodWrapper] type: 'promise' },
           descriptorsForDevice: { [Function: promiseMethodWrapper] type: 'promise' },
           descriptorsForService: { [Function: promiseMethodWrapper] type: 'promise' },
           descriptorsForCharacteristic: { [Function: promiseMethodWrapper] type: 'promise' },
           readCharacteristicForDevice: { [Function: promiseMethodWrapper] type: 'promise' },
           readCharacteristicForService: { [Function: promiseMethodWrapper] type: 'promise' },
           readCharacteristic: { [Function: promiseMethodWrapper] type: 'promise' },
           writeCharacteristicForDevice: { [Function: promiseMethodWrapper] type: 'promise' },
           writeCharacteristicForService: { [Function: promiseMethodWrapper] type: 'promise' },
           writeCharacteristic: { [Function: promiseMethodWrapper] type: 'promise' },
           monitorCharacteristicForDevice: { [Function: promiseMethodWrapper] type: 'promise' },
           monitorCharacteristicForService: { [Function: promiseMethodWrapper] type: 'promise' },
           monitorCharacteristic: { [Function: promiseMethodWrapper] type: 'promise' },
           readDescriptorForDevice: { [Function: promiseMethodWrapper] type: 'promise' },
           readDescriptorForService: { [Function: promiseMethodWrapper] type: 'promise' },
           readDescriptorForCharacteristic: { [Function: promiseMethodWrapper] type: 'promise' },
           readDescriptor: { [Function: promiseMethodWrapper] type: 'promise' },
           writeDescriptorForDevice: { [Function: promiseMethodWrapper] type: 'promise' },
           writeDescriptorForService: { [Function: promiseMethodWrapper] type: 'promise' },
           writeDescriptorForCharacteristic: { [Function: promiseMethodWrapper] type: 'promise' },
           writeDescriptor: { [Function: promiseMethodWrapper] type: 'promise' },
           cancelTransaction: { [Function: nonPromiseMethodWrapper] type: 'async' },
           setLogLevel: { [Function: nonPromiseMethodWrapper] type: 'async' },
           logLevel: { [Function: promiseMethodWrapper] type: 'promise' },
           createClient: { [Function: nonPromiseMethodWrapper] type: 'async' },
           destroyClient: { [Function: nonPromiseMethodWrapper] type: 'async' },
           enable: { [Function: promiseMethodWrapper] type: 'promise' },
           disable: { [Function: promiseMethodWrapper] type: 'promise' },
           state: { [Function: promiseMethodWrapper] type: 'promise' },
           startDeviceScan: { [Function: nonPromiseMethodWrapper] type: 'async' },
           addListener: { [Function: nonPromiseMethodWrapper] type: 'async' },
           removeListeners: { [Function: nonPromiseMethodWrapper] type: 'async' },
           DisconnectionEvent: 'DisconnectionEvent',
           ScanEvent: 'ScanEvent',
           StateChangeEvent: 'StateChangeEvent',
           ConnectedEvent: 'ConnectedEvent',
           ConnectingEvent: 'ConnectingEvent',
           RestoreStateEvent: 'RestoreStateEvent',
           ReadEvent: 'ReadEvent',
           getConstants: [Function] } },
     _uniqueId: 3,
     _activePromises: {},
     _activeSubscriptions: { '1': { remove: [Function: remove] } },
[... 17 lines cut ...]
        '204': 'Device {deviceID} not found',
        '205': 'Device {deviceID} is not connected',
        '206': 'Device {deviceID} could not change MTU size',
        '300': 'Services discovery failed for device {deviceID}',
        '301': 'Included services discovery failed for device {deviceID} and service: {serviceUUID}',
        '302': 'Service {serviceUUID} for device {deviceID} not found',
        '303': 'Services not discovered for device {deviceID}',
[... cut ...]
        '504': 'Descriptors not discovered for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}',
        '505': 'Cannot write to descriptor {descriptorUUID} with invalid data format: {internalMessage}',
        '506': 'Cannot write to descriptor {descriptorUUID}. It\'s not allowed by iOS and therefore forbidden on Android as well.',
        '600': 'Cannot start scanning operation',
        '601': 'Location services are disabled' },
     _scanEventSubscription: { remove: [Function: remove] } } }
```

MahmoudNafiseh, 2023-11-22:
@KamilNatonek I just installed the example app on my iOS device, I do get an undefined when I use your PR + the same logs you put

luca-tomasetti, 2023-11-22:
> @luca-tomasetti @MahmoudNafiseh Check this out -> [#605 (comment)](https://github.com/dotintent/react-native-ble-plx/issues/605#issuecomment-580655596) I've added console logs to our example app and these are my logs confirming the theory from the link.
> 
> ` LOG Device name: MIScooter2526,Device manufacturerData: null` -> first record ` LOG Device name: MIScooter2526,Device manufacturerData: TkIuAgAAAM8=` -> second record

As @supurloop my problem is not related with #605, on iOS with 3.1.1 I always received  `manufacturerData === undefined`, not just the first time. If I use Android and 3.1.1, the problem doesn't occur and manufacturerData is populated as expected.

closed 2023-11-24 by dominik-czupryna-withintent