dotintent/react-native-ble-plx #1139: [Android 14] startDeviceScan can Not find the ble device after it is disconnected

edwardhsueh, 2023-11-17:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

Hi, 
in android 14, after my ble device is disconnected and power off, startDeviceScan fails find the ble device again after it is powered on again. on other Android version, startDeviceScan function works well as expected
I have use the most recent version of react-native-ble-plx(3.1.2-rc.0)

1. I call startDeviceScan as shown below

 self.manager.startDeviceScan(null,
                                        null, (error, device) => {
                                        ....
}

2. When startDeviceScan  fails to  find the ble device,  I go to android blutooth menu, I can find the device on the list. I also use another App (nrf connect), it can also find the ble device
3. I found an interest thing , if I pair the my ble device in the android bluetooth menu first, it seems that startDeviceScan  can find my ble device.

Do you have any idea about how I can fix the problem. Many Thanks

edwardhsueh, 2023-11-17:
additional,
- My phone is google pixel 8
- startDeviceScan can also work after turning off Bluetooth and then turning on again

edwardhsueh, 2023-11-18:
Finally, the root cause is that I should stopDeviceScan only and if only startDeviceScan is called

closed 2023-11-19 by edwardhsueh

pariharkaran, 2024-04-24:
Facing same issue on Google Pixel 7 (Android 14), above solution didn't worked for me. If anyone found solution please comment here

JoshuaTLOZ, 2024-05-27:
Samsung A52. I use it to control a midi ble device... Update to android14 and midi commander crashes when I try to search ble devices. Work in android13