dotintent/react-native-ble-plx #1140: Question: omitting a line from AndroidManifest.xml

milobob, 2023-11-17:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

What would happen if the line:
`<uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>`
was omitted? What would be the expected behavior? Asking for a friend :)

Specifically, does this just influence whether it will be installed on certain phones in Google Play store? Or could it have implications for BLE in the background functionality?

dominik-czupryna-withintent, 2023-11-29:
"If you say the feature is required for your app, then the Google Play store will hide your app from users on devices lacking those features. For this reason, you should only set the required attribute to true if your app can't work without the feature." ~ https://developer.android.com/develop/connectivity/bluetooth/bt-permissions#features

closed 2023-11-29 by dominik-czupryna-withintent

milobob, 2023-11-29:
OK that's what 

> "If you say the feature is required for your app, then the Google Play store will hide your app from users on devices lacking those features. For this reason, you should only set the required [...]

OK thanks! That's what I figured.

However, if that's true, then I'm confused why the readme states: - `isBackgroundEnabled` (_boolean_): Enable background BLE support on Android. Adds `<uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>` to the `AndroidManifest.xml`. Default `false`.
Why is the AndroidManifest.xml file associated with background BLE support on Android? May I suggest edit the wording to be more clear about why the AndroidManifest.xml is edited, since by your comment above it does not seem strictly necessary for background BLE support on Android. For example, change:
` `isBackgroundEnabled` (_boolean_): Enable background BLE support on Android. Adds `<uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>` to the `AndroidManifest.xml`. Default `false`.
` to:
`  `isBluetoothRequired` (_boolean_):  Adds `<uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>` to the `AndroidManifest.xml`, which helps hide the app on Google Play store from some users with devices lacking these features. Default `false`.`