dotintent/react-native-ble-plx #1141: 🐛 Attempt to invoke interface method 'void com.bleplx.adapter.BleAdapter.stopDeviceScan()' on a null object reference

scarlettmiss, 2023-11-20:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

The app not to crash and start device scan to be called successfully

### Current Behavior

The app is crashig with above error. We havent seen it a lot and we have yet to reproduce on our own. Didn't happen on the v2.x

### Library version

3.1.2-rc.0

### Device

Galaxy Tab A8,Galaxy S20 FE 5G,Galaxy Tab S6 Lite,Lenovo Tab P11 (2nd Gen),Galaxy Tab A (2016)

### Environment info

```shell
hermesEnabled:true
react native: "0.72.6"
```

### Steps to reproduce

we have not yet reproduced on our side

### Formatted code sample or link to a repository

```js
class BleDeviceLocator {
    private bleManager: BleManager;

    constructor() {
        this.bleManager = new BleManager();
    }

    public startBluetoothScanning(onDeviceFound: (device: BlePlxDevice) => void, repeatFound: boolean = false): void {
        this.bleManager.startDeviceScan(null, null, (error: BleError | null, device: BlePlxDevice | null) => {
            .
            .
            .
        });
    }
}
```

### Relevant log output

```shell
Fatal Exception: java.lang.NullPointerException
Attempt to invoke interface method 'void com.bleplx.adapter.BleAdapter.stopDeviceScan()' on a null object reference
com.bleplx.BlePlxModule.stopDeviceScan (BlePlxModule.java:222)
java.lang.reflect.Method.invoke (Method.java)
com.facebook.react.bridge.JavaMethodWrapper.invoke (JavaMethodWrapper.java:372)
com.facebook.react.bridge.JavaModuleWrapper.invoke (JavaModuleWrapper.java:188)
com.facebook.jni.NativeRunnable.run (NativeRunnable.java)
android.os.Handler.handleCallback (Handler.java:790)
android.os.Handler.dispatchMessage (Handler.java:99)
com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage (MessageQueueThreadHandler.java:27)
android.os.Looper.loop (Looper.java:164)
com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run (MessageQueueThreadImpl.java:228)
java.lang.Thread.run (Thread.java:764)
```

kindapath, 2023-11-24:
Same, and I can add that everything works in dev mode, but when i do `gradlew assembleRelease`, install apk and open the app it crashes with this error.

dominik-czupryna-withintent, 2023-11-24:
I'm having trouble reproducing this 😞 

I created a sample project:
https://github.com/dominik-czupryna-withintent/ble-plx-issues-1144-1142-1141
with as mentioned above:
```js
    "react-native": "0.72.6",
    "react-native-ble-plx": "^3.1.2-rc.0",
```

I generated a release app
https://github.com/dominik-czupryna-withintent/ble-plx-issues-1144-1142-1141/blob/main/app-release.apk

And tested it
https://github.com/dominik-czupryna-withintent/ble-plx-issues-1144-1142-1141/blob/main/Screenrecorder-2023-11-24-13-52-39-436.mp4

The only option in which I can trigger the error mentioned was when I destroyed the Manager instance, but this should not be done as long as it is in use.

Can you show me how to reproduce these errors in the above project?

solheimsolutions, 2023-11-30:
Get the same (NullPointerException) but for BleAdapter.getCurrentState(). Only in prelaunch report. Cannot reproduce on local simulator even if device/SDK are the same as reported from Google
Happened on two devices on google test. Pixel 2(SDK31) and Small Desktop (SDK 32)

Philipp-gr4, 2023-12-31:
Im getting the same error on a Xiaomi Redmi Note 12 Pro+ (Android 13) when i install the generated APK. On the expo dev server and on Android 14 devices everything seems to works. 

I created a new issue with my code sample, maybe this helps to reproduce the error: https://github.com/dotintent/react-native-ble-plx/issues/1159

henla464, 2024-01-06:
I'm also getting this error on a OnePlus 6T (Android version 11, OxygenOS 11.1.2.2). Hope someone can find the cause!

simonitfi, 2024-01-16:
Hello, we have this same on "BleAdapter.getCurrentState()" as well. I think quick fix would be wrap some try catch there on Java-code so it would not crash the app. This seems to be related somehow if the manager is newly created or being destroyed. This is maybe only hunch.

danielholbling, 2024-01-29:
Adding to the crowd that gets the error with BleAdapter.getCurrentState(). Only shows in prelaunch report, works perfectly on test devices.

solheimsolutions, 2024-02-13:
> Adding to the crowd that gets the error with BleAdapter.getCurrentState(). Only shows in prelaunch report, works perfectly on test devices.

I avoided this error by checking if device is physical. If not (BLE does not work on simulator anyway) I set app to not try to use BLE at all. Then the prelaunch tests passed without crashes

jaittoka, 2024-03-13:
We are seeing the same issue. We have identified that it is caused by the `createClient`-function ([BlePlxModule.java](https://github.com/dotintent/react-native-ble-plx/blob/c60eff0d1f040dfe1ef80880c040cf58bab55337/android/src/main/java/com/bleplx/BlePlxModule.java#L100)) not instantiating the `bleAdapter`. That is caused by `getCurrentActivity`-function returning `null` sometimes. 

In our case it was caused by the another react native library which responded to push notification. When it activated our application, the `Activity` is not the main activity of our react-native application.

There are at least couple of problems with `react-native-ble-plx`-library:
* It doesn't actually handle the case where `getCurrentActivity` returns null. It just leaves the module in an inconsistent state that causes Java side to crash at some point later.
* It takes a long lived reference to the value returned by `getCurrentActivity`-function, even though  at the react-native source code it says: `DO NOT HOLD LONG-LIVED REFERENCES TO THE OBJECT RETURNED BY THIS METHOD, AS THIS WILL CAUSE MEMORY LEAKS`.

cixio, 2024-05-04:
I got the same error like @jaittoka 
`
05-03 18:00:57.587  5033  5061 E AndroidRuntime: FATAL EXCEPTION: mqt_native_modules
05-03 18:00:57.587  5033  5061 E AndroidRuntime: Process: de.ira.app, PID: 5033
05-03 18:00:57.587  5033  5061 E AndroidRuntime: java.lang.NullPointerException: Attempt to invoke interface method 'java.lang.String com.bleplx.adapter.BleAdapter.getCurrentState()' on a null object reference
05-03 18:00:57.587  5033  5061 E AndroidRuntime: 	at com.bleplx.BlePlxModule.state(BlePlxModule.java:178)
05-03 18:00:57.587  5033  5061 E AndroidRuntime: 	at java.lang.reflect.Method.invoke(Native Method)
05-03 18:00:57.587  5033  5061 E AndroidRuntime: 	at com.facebook.react.bridge.JavaMethodWrapper.invoke(JavaMethodWrapper.java:372)
05-03 18:00:57.587  5033  5061 E AndroidRuntime: 	at com.facebook.react.bridge.JavaModuleWrapper.invoke(JavaModuleWrapper.java:146)
05-03 18:00:57.587  5033  5061 E AndroidRuntime: 	at com.facebook.jni.NativeRunnable.run(Native Method)
05-03 18:00:57.587  5033  5061 E AndroidRuntime: 	at android.os.Handler.handleCallback(Handler.java:958)
05-03 18:00:57.587  5033  5061 E AndroidRuntime: 	at android.os.Handler.dispatchMessage(Handler.java:99)
05-03 18:00:57.587  5033  5061 E AndroidRuntime: 	at com.facebook.react.bridge.queue.MessageQueueThreadHandler.dispatchMessage(MessageQueueThreadHandler.java:27)
05-03 18:00:57.587  5033  5061 E AndroidRuntime: 	at android.os.Looper.loopOnce(Looper.java:205)
05-03 18:00:57.587  5033  5061 E AndroidRuntime: 	at android.os.Looper.loop(Looper.java:294)
05-03 18:00:57.587  5033  5061 E AndroidRuntime: 	at com.facebook.react.bridge.queue.MessageQueueThreadImpl$4.run(MessageQueueThreadImpl.java:233)
05-03 18:00:57.587  5033  5061 E AndroidRuntime: 	at java.lang.Thread.run(Thread.java:1012)
`

Seems that my app crashes when adding the subscription for manager.onStateChange. It happens only in release mode - and not always. There are users devices on which the app crashes every time (Samsung SM-A515F) and some (many) other devices where the app crashes just sometimes - but every time on startup.

In simulator I can reproduce a crash - but also not every time. I have to install and delete the release build until it crashes (needs 3-5x uninstall and install). Sometimes it crashes only once, sometimes it crashes 2-3 times on app start. But when the app startet one time correctly it never crashes again.

@dominik-czupryna-withintent maybe it is a solution to fix the problems @jaittoka mentioned if you cant reproduce?

intent-kacper-cyranowski, 2024-06-18:
Hi everyone

The issue with the NullPointerException was related to calling a method on non-existent objects. This has been resolved in version 3.2.0.

intent-kacper-cyranowski, 2024-06-24:
I’m closing the issue, in case of anyone with similar error please upgrade to at least 3.2.0 version

closed 2024-06-24 by intent-kacper-cyranowski