dotintent/react-native-ble-plx #1142: 🐛 Attempt to invoke virtual method 'java.lang.String com.bleplx.adapter.Service.getDeviceID()' on a null object reference

scarlettmiss, 2023-11-21:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

the app not to crash and cleanServicesAndCharacteristicsForDevice to be called successfully

### Current Behavior

The app is crashig with above error. We havent seen it a lot and we have yet to reproduce on our own. Didn't happen on the v2.x

### Library version

3.1.2-rc.0

### Device

Galaxy Tab S6 Lite, Galaxy Tab A, Galaxy Tab S7 FE,Lenovo Tab M10 FHD Plus

### Environment info

```shell
hermesEnabled:true
react native: "0.72.6"
```

### Steps to reproduce

we have not yet reproduced on our side

### Formatted code sample or link to a repository

```js
-
```

### Relevant log output

```shell
Caused by java.lang.NullPointerException
Attempt to invoke virtual method 'java.lang.String com.bleplx.adapter.Service.getDeviceID()' on a null object reference
com.bleplx.adapter.BleModule.cleanServicesAndCharacteristicsForDevice (BleModule.java:1548)
com.bleplx.adapter.BleModule.onDeviceDisconnected (BleModule.java:1286)
com.bleplx.adapter.BleModule.lambda$safeConnectToDevice$29$com-bleplx-adapter-BleModule (BleModule.java:1273)
com.bleplx.adapter.BleModule$$ExternalSyntheticLambda34.accept (Unknown Source:8)
io.reactivex.internal.observers.LambdaObserver.onError (LambdaObserver.java:77)
io.reactivex.internal.operators.observable.ObservableDoFinally$DoFinallyObserver.onError (ObservableDoFinally.java:85)
io.reactivex.internal.observers.DisposableLambdaObserver.onError (DisposableLambdaObserver.java:65)
io.reactivex.internal.operators.observable.ObservableDoFinally$DoFinallyObserver.onError (ObservableDoFinally.java:85)
io.reactivex.internal.operators.observable.ObservableUnsubscribeOn$UnsubscribeObserver.onError (ObservableUnsubscribeOn.java:70)
io.reactivex.internal.operators.observable.ObservableSubscribeOn$SubscribeOnObserver.onError (ObservableSubscribeOn.java:63)
io.reactivex.internal.operators.observable.ObservableDoFinally$DoFinallyObserver.onError (ObservableDoFinally.java:85)
io.reactivex.internal.observers.DisposableLambdaObserver.onError (DisposableLambdaObserver.java:65)
io.reactivex.internal.operators.observable.ObservableDelaySubscriptionOther$DelayObserver$OnComplete.onError (ObservableDelaySubscriptionOther.java:99)
io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.checkTerminate (ObservableFlatMap.java:496)
io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.drainLoop (ObservableFlatMap.java:334)
io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.drain (ObservableFlatMap.java:326)
io.reactivex.internal.operators.observable.ObservableFlatMap$InnerObserver.onError (ObservableFlatMap.java:575)
io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.checkTerminate (ObservableFlatMap.java:496)
io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.drainLoop (ObservableFlatMap.java:334)
io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.drain (ObservableFlatMap.java:326)
io.reactivex.internal.operators.observable.ObservableFlatMap$InnerObserver.onError (ObservableFlatMap.java:575)
io.reactivex.internal.disposables.EmptyDisposable.error (EmptyDisposable.java:63)
io.reactivex.internal.operators.observable.ObservableError.subscribeActual (ObservableError.java:38)
io.reactivex.Observable.subscribe (Observable.java:12284)
io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.subscribeInner (ObservableFlatMap.java:165)
io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.onNext (ObservableFlatMap.java:139)
io.reactivex.internal.util.NotificationLite.accept (NotificationLite.java:246)
[... 33 lines cut ...]
io.reactivex.internal.operators.observable.ObservableFlatMap$InnerObserver.onError (ObservableFlatMap.java:575)
io.reactivex.internal.disposables.EmptyDisposable.error (EmptyDisposable.java:63)
io.reactivex.internal.operators.observable.ObservableError.subscribeActual (ObservableError.java:38)
io.reactivex.Observable.subscribe (Observable.java:12284)
io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.subscribeInner (ObservableFlatMap.java:165)
io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.onNext (ObservableFlatMap.java:139)
io.reactivex.internal.util.NotificationLite.accept (NotificationLite.java:246)
io.reactivex.internal.operators.observable.ObservableReplay$UnboundedReplayBuffer.replay (ObservableReplay.java:553)
io.reactivex.internal.operators.observable.ObservableReplay$ReplayObserver.replay (ObservableReplay.java:401)
io.reactivex.internal.operators.observable.ObservableReplay$ReplayObserver.onNext (ObservableReplay.java:366)
io.reactivex.internal.operators.observable.ObservableDoOnEach$DoOnEachObserver.onNext (ObservableDoOnEach.java:101)
io.reactivex.internal.operators.observable.ObservableElementAt$ElementAtObserver.onNext (ObservableElementAt.java:86)
com.jakewharton.rxrelay2.BehaviorRelay$BehaviorDisposable.test (BehaviorRelay.java:364)
com.jakewharton.rxrelay2.BehaviorRelay$BehaviorDisposable.emitNext (BehaviorRelay.java:358)
com.jakewharton.rxrelay2.BehaviorRelay.accept (BehaviorRelay.java:135)
com.polidea.rxandroidble2.internal.connection.DisconnectionRouter.onDisconnectedException (DisconnectionRouter.java:110)
com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange (RxBleGattCallback.java:81)
android.bluetooth.BluetoothGatt$1$5.run (BluetoothGatt.java:363)
android.bluetooth.BluetoothGatt.runOrQueueCallback (BluetoothGatt.java:1003)
android.bluetooth.BluetoothGatt.-$$Nest$mrunOrQueueCallback
android.bluetooth.BluetoothGatt$1.onClientConnectionState (BluetoothGatt.java:358)
android.bluetooth.IBluetoothGattCallback$Stub.onTransact (IBluetoothGattCallback.java:135)
android.os.Binder.execTransactInternal (Binder.java:1321)
android.os.Binder.execTransact (Binder.java:1280)
```

kindapath, 2023-11-23:
same problem but with stopDeviceScan

solheimsolutions, 2023-11-30:
I get the same for .state(). Happens on google prelaunch for few devices. Not able to reproduce locally

11-29 09:23:45.549: E/AndroidRuntime(7426): java.lang.NullPointerException: Attempt to invoke interface method 'java.lang.String com.bleplx.adapter.BleAdapter.getCurrentState()' on a null object reference

intent-kacper-cyranowski, 2024-06-18:
Hi everyone

The issue with the NullPointerException was related to calling a method on non-existent objects. This has been resolved in version 3.2.0.

intent-kacper-cyranowski, 2024-06-24:
I’m closing the issue, in case of anyone with similar error please upgrade to at least 3.2.0 version

closed 2024-06-24 by intent-kacper-cyranowski

angelos3lex, 2025-03-10:
@intent-kacper-cyranowski we did get again same error when upgrading to 3.5 from v2.
```
 Caused by java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String com.bleplx.adapter.Service.getDeviceID()' on a null object reference
       at com.bleplx.adapter.BleModule.cleanServicesAndCharacteristicsForDevice(BleModule.java:1569)
       at com.bleplx.adapter.BleModule.onDeviceDisconnected(BleModule.java:1303)
       at com.bleplx.adapter.BleModule.lambda$safeConnectToDevice$30$com-bleplx-adapter-BleModule(BleModule.java:1290)
       at com.bleplx.adapter.BleModule$$ExternalSyntheticLambda35.accept(:8)
       at io.reactivex.internal.observers.LambdaObserver.onError(LambdaObserver.java:77)
       at io.reactivex.internal.operators.observable.ObservableDoFinally$DoFinallyObserver.onError(ObservableDoFinally.java:85)
       at io.reactivex.internal.observers.DisposableLambdaObserver.onError(DisposableLambdaObserver.java:65)
       at io.reactivex.internal.operators.observable.ObservableDoFinally$DoFinallyObserver.onError(ObservableDoFinally.java:85)
       at io.reactivex.internal.operators.observable.ObservableUnsubscribeOn$UnsubscribeObserver.onError(ObservableUnsubscribeOn.java:70)
       at io.reactivex.internal.operators.observable.ObservableSubscribeOn$SubscribeOnObserver.onError(ObservableSubscribeOn.java:63)
       at io.reactivex.internal.operators.observable.ObservableDoFinally$DoFinallyObserver.onError(ObservableDoFinally.java:85)
       at io.reactivex.internal.observers.DisposableLambdaObserver.onError(DisposableLambdaObserver.java:65)
       at io.reactivex.internal.operators.observable.ObservableDelaySubscriptionOther$DelayObserver$OnComplete.onError(ObservableDelaySubscriptionOther.java:99)
       at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.checkTerminate(ObservableFlatMap.java:496)
       at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.drainLoop(ObservableFlatMap.java:334)
       at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.drain(ObservableFlatMap.java:326)
       at io.reactivex.internal.operators.observable.ObservableFlatMap$InnerObserver.onError(ObservableFlatMap.java:575)
       at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.checkTerminate(ObservableFlatMap.java:496)
       at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.drainLoop(ObservableFlatMap.java:334)
       at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.drain(ObservableFlatMap.java:326)
       at io.reactivex.internal.operators.observable.ObservableFlatMap$InnerObserver.onError(ObservableFlatMap.java:575)
       at io.reactivex.internal.disposables.EmptyDisposable.error(EmptyDisposable.java:63)
       at io.reactivex.internal.operators.observable.ObservableError.subscribeActual(ObservableError.java:38)
       at io.reactivex.Observable.subscribe(Observable.java:12284)
       at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.subscribeInner(ObservableFlatMap.java:165)
       at io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.onNext(ObservableFlatMap.java:139)
       at io.reactivex.internal.util.NotificationLite.accept(NotificationLite.java:246)
       at io.reactivex.internal.operators.observable.ObservableReplay$UnboundedReplayBuffer.replay(ObservableReplay.java:553)
       at io.reactivex.internal.operators.observable.ObservableReplay$ReplayObserver.replay(ObservableReplay.java:401)
       at io.reactivex.internal.operators.observable.ObservableReplay$ReplayObserver.onNext(ObservableReplay.java:366)
       at io.reactivex.internal.operators.observable.ObservableDoOnEach$DoOnEachObserver.onNext(ObservableDoOnEach.java:101)
       at io.reactivex.internal.operators.observable.ObservableElementAt$ElementAtObserver.onNext(ObservableElementAt.java:86)
       at com.jakewharton.rxrelay2.BehaviorRelay$BehaviorDisposable.test(BehaviorRelay.java:364)
       at com.jakewharton.rxrelay2.BehaviorRelay$BehaviorDisposable.emitNext(BehaviorRelay.java:358)
       at com.jakewharton.rxrelay2.BehaviorRelay.accept(BehaviorRelay.java:135)
       at com.polidea.rxandroidble2.internal.connection.DisconnectionRouter.onDisconnectedException(DisconnectionRouter.java:110)
       at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:81)
       at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:381)
       at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:1068)
       at android.bluetooth.BluetoothGatt.-$$Nest$mrunOrQueueCallback()
       at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:376)
       at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:226)
       at android.os.Binder.execTransactInternal(Binder.java:1380)
       at android.os.Binder.execTransact(Binder.java:1311)
```

[image]