dotintent/react-native-ble-plx #1144: 🐛 com.bleplx.adapter.BleModule.cleanServicesAndCharacteristicsForDevice outOfBounds

scarlettmiss, 2023-11-21:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

the app not to crash and cleanServicesAndCharacteristicsForDevice to be called successfully

### Current Behavior

The app is crashig with above error. We havent seen it a lot and we have yet to reproduce on our own. Didn't happen on the v2.x

### Library version

3.1.2-rc.0

### Device

Galaxy Tab A7, Galaxy Tab S7 FE, Galaxy Tab A8, Galaxy Tab S6 Lite, Galaxy S20 FE 5G, Galaxy Tab A

### Environment info

```shell
hermesEnabled:true
react native: "0.72.6"
```

### Steps to reproduce

we have not yet reproduced on our side

### Formatted code sample or link to a repository

```js
-
```

### Relevant log output

```shell
Caused by java.lang.ArrayIndexOutOfBoundsException
Array index out of range: 5
android.util.SparseArray.keyAt (SparseArray.java:317)
com.bleplx.adapter.BleModule.cleanServicesAndCharacteristicsForDevice (BleModule.java:1545)
com.bleplx.adapter.BleModule.onDeviceDisconnected (BleModule.java:1286)
com.bleplx.adapter.BleModule.lambda$safeConnectToDevice$29$com-bleplx-adapter-BleModule (BleModule.java:1273)
com.bleplx.adapter.BleModule$$ExternalSyntheticLambda34.accept (Unknown Source:8)
io.reactivex.internal.observers.LambdaObserver.onError (LambdaObserver.java:77)
io.reactivex.internal.operators.observable.ObservableDoFinally$DoFinallyObserver.onError (ObservableDoFinally.java:85)
io.reactivex.internal.observers.DisposableLambdaObserver.onError (DisposableLambdaObserver.java:65)
io.reactivex.internal.operators.observable.ObservableDoFinally$DoFinallyObserver.onError (ObservableDoFinally.java:85)
io.reactivex.internal.operators.observable.ObservableUnsubscribeOn$UnsubscribeObserver.onError (ObservableUnsubscribeOn.java:70)
io.reactivex.internal.operators.observable.ObservableSubscribeOn$SubscribeOnObserver.onError (ObservableSubscribeOn.java:63)
io.reactivex.internal.operators.observable.ObservableDoFinally$DoFinallyObserver.onError (ObservableDoFinally.java:85)
io.reactivex.internal.observers.DisposableLambdaObserver.onError (DisposableLambdaObserver.java:65)
io.reactivex.internal.operators.observable.ObservableDelaySubscriptionOther$DelayObserver$OnComplete.onError (ObservableDelaySubscriptionOther.java:99)
io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.checkTerminate (ObservableFlatMap.java:496)
io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.drainLoop (ObservableFlatMap.java:334)
io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.drain (ObservableFlatMap.java:326)
io.reactivex.internal.operators.observable.ObservableFlatMap$InnerObserver.onError (ObservableFlatMap.java:575)
io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.checkTerminate (ObservableFlatMap.java:496)
io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.drainLoop (ObservableFlatMap.java:334)
io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.drain (ObservableFlatMap.java:326)
io.reactivex.internal.operators.observable.ObservableFlatMap$InnerObserver.onError (ObservableFlatMap.java:575)
io.reactivex.internal.disposables.EmptyDisposable.error (EmptyDisposable.java:63)
io.reactivex.internal.operators.observable.ObservableError.subscribeActual (ObservableError.java:38)
io.reactivex.Observable.subscribe (Observable.java:12284)
io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.subscribeInner (ObservableFlatMap.java:165)
io.reactivex.internal.operators.observable.ObservableFlatMap$MergeObserver.onNext (ObservableFlatMap.java:139)
io.reactivex.internal.util.NotificationLite.accept (NotificationLite.java:246)
io.reactivex.internal.operators.observable.ObservableReplay$UnboundedReplayBuffer.replay (ObservableReplay.java:553)
io.reactivex.internal.operators.observable.ObservableReplay$ReplayObserver.replay (ObservableReplay.java:401)
io.reactivex.internal.operators.observable.ObservableReplay$ReplayObserver.onNext (ObservableReplay.java:366)
io.reactivex.internal.operators.observable.ObservableDoOnEach$DoOnEachObserver.onNext (ObservableDoOnEach.java:101)
io.reactivex.internal.operators.observable.ObservableElementAt$ElementAtObserver.onNext (ObservableElementAt.java:86)
com.jakewharton.rxrelay2.BehaviorRelay$BehaviorDisposable.test (BehaviorRelay.java:364)
com.jakewharton.rxrelay2.BehaviorRelay$BehaviorDisposable.emitNext (BehaviorRelay.java:358)
com.jakewharton.rxrelay2.BehaviorRelay.accept (BehaviorRelay.java:135)
com.polidea.rxandroidble2.internal.connection.DisconnectionRouter.onDisconnectedException (DisconnectionRouter.java:110)
com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange (RxBleGattCallback.java:81)
android.bluetooth.BluetoothGatt$1$5.run (BluetoothGatt.java:346)
android.bluetooth.BluetoothGatt.runOrQueueCallback (BluetoothGatt.java:964)
android.bluetooth.BluetoothGatt.access$600 (BluetoothGatt.java:48)
android.bluetooth.BluetoothGatt$1.onClientConnectionState (BluetoothGatt.java:341)
android.bluetooth.IBluetoothGattCallback$Stub.onTransact (IBluetoothGattCallback.java:216)
android.os.Binder.execTransactInternal (Binder.java:1220)
android.os.Binder.execTransact (Binder.java:1179)
```

intent-kacper-cyranowski, 2024-07-11:
Hi @scarlettmiss 
Sorry for late response, is your issue is still relevant?

closed 2024-07-17 by intent-kacper-cyranowski

angelos3lex, 2025-03-18:
@intent-kacper-cyranowski Indeed this is still relevant.

```
Caused by java.lang.ArrayIndexOutOfBoundsException
Array index out of range: 12
android.util.SparseArray.keyAt (SparseArray.java:317)
com.bleplx.adapter.BleModule.cleanServicesAndCharacteristicsForDevice (BleModule.java:1566)
com.bleplx.adapter.BleModule.onDeviceDisconnected (BleModule.java:1303)
com.bleplx.adapter.BleModule.lambda$safeConnectToDevice$30$com-bleplx-adapter-BleModule (BleModule.java:1290)
com.bleplx.adapter.BleModule$$ExternalSyntheticLambda35.accept (Unknown Source:8)
io.reactivex.internal.observers.LambdaObserver.onError (LambdaObserver.java:77)
io.reactivex.internal.operators.observable.ObservableElementAt$ElementAtObserver.onNext (ObservableElementAt.java:86)
com.jakewharton.rxrelay2.BehaviorRelay$BehaviorDisposable.test (BehaviorRelay.java:364)
com.jakewharton.rxrelay2.BehaviorRelay$BehaviorDisposable.emitNext (BehaviorRelay.java:358)
com.jakewharton.rxrelay2.BehaviorRelay.accept (BehaviorRelay.java:135)
com.polidea.rxandroidble2.internal.connection.DisconnectionRouter.onDisconnectedException (DisconnectionRouter.java:110)
com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange (RxBleGattCallback.java:81)
android.bluetooth.BluetoothGatt$1$4.run (BluetoothGatt.java:381) 
```

```
Caused by com.polidea.rxandroidble2.exceptions.BleDisconnectedException: Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 8 (GATT_INSUF_AUTHORIZATION or GATT_CONN_TIMEOUT)
       at com.polidea.rxandroidble2.internal.connection.RxBleGattCallback$2.onConnectionStateChange(RxBleGattCallback.java:81)
       at android.bluetooth.BluetoothGatt$1$4.run(BluetoothGatt.java:381)
       at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:1068)
       at android.bluetooth.BluetoothGatt.-$$Nest$mrunOrQueueCallback()
       at android.bluetooth.BluetoothGatt$1.onClientConnectionState(BluetoothGatt.java:376)
       at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:226)
       at android.os.Binder.execTransactInternal(Binder.java:1380)
       at android.os.Binder.execTransact(Binder.java:1311)
```

After updating from v2 to v3.5.0 this is again firing (39 times occurred until now)
(The array index is not the same, could be 6, 5 ,8 etc, not always 12 as on the above reported crash^.)

angelos3lex, 2025-03-18:
@intent-kacper-cyranowski i think this can only happen when there is concurrent access to `discoveredServices`, so while iterating it, it's altered from outside.

```
private void cleanServicesAndCharacteristicsForDevice(@NonNull Device device) {
    for (int i = discoveredServices.size() - 1; i >= 0; i--) {
      if (i < discoveredServices.size()) {
        int key = discoveredServices.keyAt(i);
        Service service = discoveredServices.get(key);
        if (service.getDeviceID().equals(device.getId())) {
          discoveredServices.remove(key);
        }
     }
    }
    for (int i = discoveredCharacteristics.size() - 1; i >= 0; i--) {
      if (i < discoveredCharacteristics.size()) {
        int key = discoveredCharacteristics.keyAt(i);
        Characteristic characteristic = discoveredCharacteristics.get(key);
        if (characteristic.getDeviceId().equals(device.getId())) {
          discoveredCharacteristics.remove(key);
        }
      }
    }

    for (int i = discoveredDescriptors.size() - 1; i >= 0; i--) {
      if (i < discoveredCharacteristics.size()) {
        int key = discoveredDescriptors.keyAt(i);
        Descriptor descriptor = discoveredDescriptors.get(key);
        if (descriptor.getDeviceId().equals(device.getId())) {
          discoveredDescriptors.remove(key);
        }
      }
    }
  }
```
What do you think?