dotintent/react-native-ble-plx #1149: 🐛 device.manufacturerData is undefined in iOS

luca-tomasetti, 2023-12-01:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

After starting a scan with `startDeviceScan()` I'm expecting to receive the manufacturerData populated in the found device.

With a previous version of the library (2.0.3) in parallel with [MultiPlatformBleAdapter](https://github.com/dotintent/MultiPlatformBleAdapter)  0.1.9, the manufacturerData is normally populated, both with iOS and Android (used the same devices for triggering the bug).

### Current Behavior

The `manufacturerData` field in the found device is not populated, but undefined.

Everything is fine with an Android phone, but using iOS the manufacturerData is not defined, just the rawScanRecord.

This issue is related to #1138, which was supposed to be fixed in the latest version (3.1.2) but I am experiencing the same issue.

### Library version

3.1.2

### Device

iPhone 12 Pro (17.1.1)

### Environment info

```shell
System:
  OS: macOS 14.0
  CPU: (10) arm64 Apple M2 Pro
  Memory: 90.86 MB / 16.00 GB
  Shell:
    version: "5.9"
    path: /bin/zsh
Binaries:
  Node:
    version: 18.17.1
    path: ~/.nvm/versions/node/v18.17.1/bin/node
  Yarn: Not Found
  npm:
    version: 9.6.7
    path: ~/.nvm/versions/node/v18.17.1/bin/npm
  Watchman:
    version: 2023.09.04.00
    path: /opt/homebrew/bin/watchman
Managers:
  CocoaPods:
    version: 1.13.0
    path: /usr/local/bin/pod
SDKs:
  iOS SDK:
    Platforms:
      - DriverKit 23.0
      - iOS 17.0
      - macOS 14.0
      - tvOS 17.0
      - watchOS 10.0
  Android SDK: Not Found
IDEs:
  Android Studio: 2022.3 AI-223.8836.35.2231.10811636
  Xcode:
    version: 15.0/15A240d
    path: /usr/bin/xcodebuild
Languages:
  Java:
    version: 11.0.20
    path: /Library/Java/JavaVirtualMachines/zulu-11.jdk/Contents/Home/bin/javac
  Ruby:
    version: 2.6.10
    path: /usr/bin/ruby
npmPackages:
  "@react-native-community/cli": Not Found
  react:
    installed: 18.2.0
    wanted: 18.2.0
  react-native:
    installed: 0.72.5
    wanted: ^0.72.5
  react-native-macos: Not Found
npmGlobalPackages:
  "*react-native*": Not Found
Android:
  hermesEnabled: true
  newArchEnabled: false
iOS:
  hermesEnabled: true
  newArchEnabled: false
```

### Steps to reproduce

1. startDeviceScan()
2. receiving data on an iOS phone

### Formatted code sample or link to a repository

```js
const bleManager = new BleManager()

const services = [/* my services */]
const options = { allowDuplicates: true }
bleManager.startDeviceScan(services, options, (err, device) => {
    console.log(device.rawScanRecord) // <-- always populated
    console.log(device.manufacturerData) // <-- populated only in Android
})
```

### Relevant log output

```shell
Console.log of device in iOS: 

{"_manager": {"_activePromises": {}, "_activeSubscriptions": {}, "_errorCodesToMessagesMapping": {"0": "Unknown error occurred. This is probably a bug! Check reason property.", "1": "BleManager was destroyed", "100": "BluetoothLE is unsupported on this device", "101": "Device is not authorized to use BluetoothLE", "102": "BluetoothLE is powered off", "103": "BluetoothLE is in unknown state", "104": "BluetoothLE is resetting", "105": "Bluetooth state change failed", "2": "Operation was cancelled", "200": "Device {deviceID} connection failed", "201": "Device {deviceID} was disconnected", "202": "RSSI read failed for device {deviceID}", "203": "Device {deviceID} is already connected", "204": "Device {deviceID} not found", "205": "Device {deviceID} is not connected", "206": "Device {deviceID} could not change MTU size", "3": "Operation timed out", "300": "Services discovery failed for device {deviceID}", "301": "Included services discovery failed for device {deviceID} and service: {serviceUUID}", "302": "Service {serviceUUID} for device {deviceID} not found", "303": "Services not discovered for device {deviceID}", "4": "Operation was rejected", "400": "Characteristic discovery failed for device {deviceID} and service {serviceUUID}", "401": "Characteristic {characteristicUUID} write failed for device {deviceID} and service {serviceUUID}", "402": "Characteristic {characteristicUUID} read failed for device {deviceID} and service {serviceUUID}", "403": "Characteristic {characteristicUUID} notify change failed for device {deviceID} and service {serviceUUID}", "404": "Characteristic {characteristicUUID} not found", "405": "Characteristics not discovered for device {deviceID} and service {serviceUUID}", "406": "Cannot write to characteristic {characteristicUUID} with invalid data format: {internalMessage}", "5": "Invalid UUIDs or IDs were passed: {internalMessage}", "500": "Descriptor {descriptorUUID} discovery failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "501": "Descriptor {descriptorUUID} write failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "502": "Descriptor {descriptorUUID} read failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "503": "Descriptor {descriptorUUID} not found", "504": "Descriptors not discovered for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "505": "Cannot write to descriptor {descriptorUUID} with invalid data format: {internalMessage}", "506": "Cannot write to descriptor {descriptorUUID}. It's not allowed by iOS and therefore forbidden on Android as well.", "600": "Cannot start scanning operation", "601": "Location services are disabled"}, "_eventEmitter": {"_nativeModule": [Object]}, "_scanEventSubscription": {"remove": [Function remove]}, "_uniqueId": 0}, "id": "213E44A1-1EDB-037C-04C2-1B57E6377322", "mtu": 23, "name": null, "rawScanRecord": "eyJzZXJ2aWNlRGF0YSI6bnVsbCwidHhQb3dlckxldmVsIjpudWxsLCJzZXJ2aWNlVVVJRHMiOlsiMDAwMGZmYzAtMDAwMC0xMDAwLTgwMDAtMDA4MDVmOWIzNGZiIiwiMDAwMDEzODUtMDAwMC0xMDAwLTgwMDAtMDA4MDVmOWIzNGZiIl0sIm1hbnVmYWN0dXJlckRhdGEiOm51bGwsIm92ZXJmbG93U2VydmljZVVVSURzIjpudWxsLCJpc0Nvbm5lY3RhYmxlIjp0cnVlLCJzb2xpY2l0ZWRTZXJ2aWNlVVVJRHMiOm51bGwsImxvY2FsTmFtZSI6bnVsbH0=", "rssi": -85}
```

### Additional information

I'm using  [MultiPlatformBleAdapter](https://github.com/dotintent/MultiPlatformBleAdapter)  0.2.0

luca-tomasetti, 2023-12-02:
I did some more testing and it seems like the bug appears after the major update to 3.0.0 and [MultiPlatformBleAdapter](https://github.com/dotintent/MultiPlatformBleAdapter) 0.2.0.

With 2.0.3 and MultiPlatformBleAdapter 0.1.9 the behaviour is as expected.

projectnatz, 2023-12-14:
I also experience the same problem (3.1.2). Does anybody know a quick patch to add?

roys-qlog-co, 2024-04-21:
for me 2.0.3 and 3.1.2 not showing manufacturerData
using iphone 11

intent-kacper-cyranowski, 2024-07-16:
Hi everyone
Is anyone still struggling with the issue?

luca-tomasetti, 2024-07-17:
Hi @intent-kacper-cyranowski, for now I'm using this package with version 2.0.3 and MultiPlatformBleAdapter 0.1.9.

I will update my project with your new version (3.2.1) and check it tomorrow with an iOS device.

luca-tomasetti, 2024-07-18:
I updated to the latest version and after testing on iOS 17.5.1, the bug seems resolved :partying_face: 

I will perform more tests later but everything seems as expected!

closed 2024-07-23 by intent-kacper-cyranowski