dotintent/react-native-ble-plx #1150: 🐛 Getting null in serviceUUID and servicedata in device.discoverAllServicesAndCharacteristics()

priyasamyal, 2023-12-07:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

Expecting to receive serviceUUID and service data array after successful connect. But getting null.

[image]

### Current Behavior

Unable to fetch the services of A&D Smart BP device.

I am getting device id, name and serviceUUID **in manager.startDeviceScan() response**. 
[image]

But after calling  **device.connect()**  serviceUUID got null i response.
[image]

### Library version

3.1.2

### Device

iPhone 12

### Environment info

```shell
System:
  OS: macOS 13.4
  CPU: (10) arm64 Apple M2 Pro
  Memory: 82.33 MB / 16.00 GB
  Shell:
    version: 3.2.57
    path: /bin/bash
Binaries:
  Node:
    version: 20.5.0
    path: /opt/homebrew/bin/node
  Yarn:
    version: 1.22.21
    path: ~/Documents/Projects/FoneMed/MobileApp/upgrade/mobileapp/node_modules/.bin/yarn
  npm:
    version: 9.8.0
    path: /opt/homebrew/bin/npm
  Watchman:
    version: 2023.08.07.00
    path: /opt/homebrew/bin/watchman
Managers:
  CocoaPods:
    version: 1.12.1
    path: /usr/local/bin/pod
SDKs:
  iOS SDK:
    Platforms:
      - DriverKit 22.4
      - iOS 16.4
      - macOS 13.3
      - tvOS 16.4
      - watchOS 9.4
  Android SDK: Not Found
IDEs:
  Android Studio: 2022.3 AI-223.8836.35.2231.10406996
  Xcode:
    version: 14.3.1/14E300c
    path: /usr/bin/xcodebuild
Languages:
  Java:
    version: 11.0.21
    path: /opt/homebrew/opt/openjdk@11/bin/javac
  Ruby:
    version: 2.6.10
    path: /usr/bin/ruby
npmPackages:
  "@react-native-community/cli": Not Found
  react:
    installed: 18.2.0
    wanted: 18.2.0
  react-native:
    installed: 0.72.5
    wanted: 0.72.5
  react-native-macos: Not Found
npmGlobalPackages:
  "*react-native*": Not Found
Android:
  hermesEnabled: false
  newArchEnabled: false
iOS:
  hermesEnabled: false
  newArchEnabled: false
```

### Steps to reproduce

1. Enable A&D smart BP device
2. Check the connect() function response after detecting your device from startDeviceScan() 
3. Device response from connect function, It is not showing the serviceUUID

### Formatted code sample or link to a repository

```js
manager.connectedDevices().then(res => {
                    console.log(res, 'connectedDevices...');
                });
                manager.startDeviceScan(null, null, (error, device) => {
                    console.log(error, device, 'error, device.....');

                    if (error) {
                        // Handle error (scanning will be stopped automatically)
                        return;
                    } else {
                        manager.stopDeviceScan();
                        device.isConnected().then(status => {
                            if (status) {
                                console.log('device already connecteddd');
                            } else {
                                manager.stopDeviceScan();
                                device
                                    .connect()
                                    .then(device => {
                                        console.log(device, 'connecteddd');
                                        return device.discoverAllServicesAndCharacteristics();
                                    })
                                    .then(device => {
                                        console.log(device, 'Thennnnn');
                                    })
                                    .catch(error => {
                                        console.log(error, 'errors');
                                        // Handle errors
                                    });
                            }
                        });
                    }

                    // Check if it is a device you are looking for based on advertisement data
                    // or other criteria.
                    if (device.name === 'TI BLE Sensor Tag' || device.name === 'SensorTag') {
                        // Stop scanning as it's not necessary if you are scanning for one device.
                        // Proceed with connection.
                    }
                });
```

### Relevant log output

```shell
Expecting data in serviceUUID key.
```

dominik-czupryna-withintent, 2024-02-16:
After using the function https://dotintent.github.io/react-native-ble-plx/#blemanagerdiscoverallservicesandcharacteristicsfordevice you know services and characteristics, not directly after connect.

keilaiendraicak, 2024-03-19:
Any update to resolve this?

mreimann1, 2024-05-29:
I also have experienced this issue when using device.discoverAllServicesAndCharacteristics(). I didn't notice until reading this issue, however, that you can get the service UUIDs from startDeviceScan. So perhaps this is a workaround 

Devs: Can you please fix the behavior of this function to provide a serviceUUIDs?

optecmuza, 2024-06-07:
It seems this issue is still present. Can someone update on it?

intent-kacper-cyranowski, 2024-06-14:
Hi everyone
As mention above by @dominik-czupryna-withintent connecting to the device and discovering it's characteristics are separate processes that are covered in introduction section of the documentation. We can't make an assumption that everyone that is using this library wants to take resources of devices to do so, especially when these informations may be known upfront. At this moment there are no plans of changing that as it would affect many users.

manishmobmaxime, 2024-06-17:
I am facing the same issue, anyone got the solution?

gitsakos, 2024-07-05:
I am facing this issue as well, only on android.  Finding this library unusable on android.

intent-kacper-cyranowski, 2024-07-11:
Hi everyone
We will try to provide fix for this issue in next release. In the meantime for everyone facing this issue please use [servicesForDevice](https://dotintent.github.io/react-native-ble-plx/#blemanagerservicesfordevice) to get services data.

```js
 bleManagerInstance.servicesForDevice(device.id).then(services => {
    console.log('services', JSON.stringify(services))
 })
```

derekstavis, 2024-08-04:
Similar to the comment above, I noticed that if I call `device.services()` after `device.discoverAllServicesAndCharacteristics()` it populates the services list as expected.

closed 2025-01-21 by aliberski