dotintent/react-native-ble-plx #1153: 🐛 JSON value '(     65,     84,     90,     13 )' of type NSMutableArray cannot be converted to NSString

rakshitbharat, 2023-12-09:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

I am getting this error

```
JSON value '(
    65,
    84,
    90,
    13
)' of type NSMutableArray cannot be converted to NSString
```

### Current Behavior

getting with notifiaction

### Library version

latest

### Device

iPhone 15 Pro

### Environment info

```shell
react native
```

### Steps to reproduce

simple when listining to ios

### Formatted code sample or link to a repository

```js
this.connectedDevice.monitorCharacteristicForService(
      serviceUUID,
      characteristicUUID,
      d => {
        console.log(d);
      },
    );
```

### Relevant log output

```shell
JSON value '(
    65,
    84,
    90,
    13
)' of type NSMutableArray cannot be converted to NSString
```

dominik-czupryna-withintent, 2024-02-16:
Could you tell what device was connected? What data was transferred? What service was observed and what characteristics?

intent-kacper-cyranowski, 2024-06-19:
Hi @rakshitbharat 
Do you still need help with the issue?

intent-kacper-cyranowski, 2024-07-01:
Closing due to inactivity

closed 2024-07-01 by intent-kacper-cyranowski