dotintent/react-native-ble-plx #1155: Cannot scan Eddystone service UUID  on Android with versions > 2.0.3

pwellner, 2023-12-14:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

`startDeviceScan(["0000feaa-0000-1000-8000-00805f9b34fb"], ...)` should find BLE devices advertising with this standard Eddystone UUID when running on iOS and on Android

### Current Behavior

When using `v3.1.2` devices with this UUID are NOT found on Android, but are found correctly on iOS.
When using `v2.0.3` devices with this UUID are found correctly on Android and on iOS

### Library version

3.1.2

### Device

Android 12 OnePlus Nord N20 5G CPH2459

### Environment info

```shell
react-native `0.72.6` and expo `^49.0.21`
```

### Steps to reproduce

Connect Android device to development system with Android studio installed, and have an Eddystone beacon available nearby.

```
$ git clone git@github.com:pwellner/ble-plx-scan-bug.git
$ yarn
$ npx expo prebuild
$ npx expo run:android --device
```
Run test program on device.  Press the "GET PERMISSIONS" button, accept all permissions, then  press the "SCAN SELECTED" button, and observe the log files.

Devices with Eddystone UUID `"0000feaa-0000-1000-8000-00805f9b34fb"` are not logged.

```
$ git checkout HEAD~1   # (only change is to revert package to v2.0.3)
$ yarn
$ npx expo prebuild
$ npx expo run:android --device
```

Run test program again and observe the logs.  Devices with Eddystone UUID  `"0000feaa-0000-1000-8000-00805f9b34fb"` are all found correctly.

### Formatted code sample or link to a repository

```js
https://github.com/pwellner/ble-plx-scan-bug
```

### Relevant log output

```shell
Devices with UUID 0000feaa-0000-1000-8000-00805f9b34fb appear correctly on iOS and on Android when using v2.0.3, but they do not appear when using v3.1.2 on Android
```

pwellner, 2023-12-19:
Contact me if you need a beacon to test this.

pwellner, 2024-01-26:
Note: This problem persists with or without `legacyScan: false`

Could this issue be related to https://github.com/dotintent/MultiPlatformBleAdapter/issues/102 ?

dominik-czupryna-withintent, 2024-02-15:
The bug has been confirmed and we are working on it.

EmmaZachara, 2024-03-01:
Hi @pwellner,
We have worked on the RN side of the issue, but we will also have to engage the Android team. This is planned for next week, so we will get back to you with an update as soon as possible. 
I appreciate your patience!

pwellner, 2024-03-01:
Good news, Thanks.  Let me know if there's anything I can do to help.

pwellner, 2024-03-11:
More info:
With `expo 50.0.11` and `react-native 0.73.4` on Android, the old `react-native-ble-plx 2.0.3 startDeviceScan()` is able to scan Eddystone beacons with `uuid = null`, but the new `react-native-ble-plx 3.1.2` is never able to find them, even with `uuid = null`

EmmaZachara, 2024-03-26:
Hi @pwellner 
Thank you for the additional info. Our Android dev will get back to working on that tomorrow. He will write his status and further approach.

pwellner, 2024-05-03:
@dominik-czupryna-withintent, would you consider accepting a PR to fix this bug?  I would like to fix this if you or your colleagues are open to it.  To get me started, could you point me to the Android dependency repository that you think will need attention?

dominik-czupryna-withintent, 2024-05-05:
Hi @pwellner, 
of course, feel free to open new PR, we merge smaller PR with each release. For more complex PR we make a release for them. Unfortunately, I wasn't able to solve this issue, and our BLE Android developer is focused on another project (at the moment the task is still in progress and unfortunately, I am not able to tell a date when it will be ready). In branch https://github.com/dotintent/react-native-ble-plx/tree/bugfix/Issue-1155 you will find your code moved to the example app (react-native-ble-plx/example/), which uses directly the code from the main directory and allows you to build the app with the changes made. Under the hood for Android, we use https://github.com/dariuszseweryn/RxAndroidBle, and our iOS is based on https://github.com/Polidea/RxBluetoothKit. 

If you have any additional questions, feel free to ask, I will help if I can.

pwellner, 2025-02-16:
Hi @paulca55,  Nope,  I'm using the older version `^2.0.3`

paulca55, 2025-02-16:
@pwellner What version of Expo are you using? I'm using Expo 51 is there anything I need to do to keep using 'v2.0.3' since the config plugin is now included in Expo?

pwellner, 2025-02-16:
`2.0.3` is working for me with Expo 51 and `"@config-plugins/react-native-ble-plx": "^7.0.0"`

paulca55, 2025-02-16:
Thanks @pwellner I'll give that a try.

bagraercan, 2025-02-16:
I am using     **"expo": "~52.0.33"**, what should i do to use **react-native-ble-plx** properly? Version 3.0.5 does not scan devices with an uuid and version 2.0.3 gives error when doing **expo prebuild**.

paulca55, 2025-02-17:
I get the following error when trying to install `@config-plugins/react-native-ble-plx` at `v7.0.0.`.

> Could not resolve dependency: peer expo@"^49" from @config-plugins/react-native-ble-plx@7.0.0

We got around this when upgrading to Expo 50 by adding the following to `package.json` and using `react-native-ble-plx` `v2.0.3` which worked.

```
"overrides": {
    "@config-plugins/react-native-ble-plx": {
      "expo": "^50.0.17"
    }
}
```

I've tried bumping the expo override to `^51.0.39` which prevents the error, but I'm back to Android not scanning.