dotintent/react-native-ble-plx #1157: [Android] After migration from 2.0.3 to 3.1.2 device disconnect after 5 second.

p4bl1t0, 2023-12-18:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

I've upgrade from v2 to v3. Now while connecting from an Android phone the BLE device disconnect 5 seconds without any other error. On the event "onDeviceDisconnected" error is equal to null. 

The documentation said that this is because of a call to cancelConnection but no call is made. 

Using exactly the same code with version 2 works all fine so I asking for hint to help me to find the problem with connection. It's has to be a changed in the library.

In JavaScript I doesn't have any log information. 

```
12-18 12:21:54.414 13688 17245 D BluetoothGatt: connect() - device: F0:5C:7C:7A:CB:15, auto: false
12-18 12:21:54.415 13688 17245 D BluetoothGatt: registerApp()
12-18 12:21:54.415 13688 17245 D BluetoothGatt: registerApp() - UUID=922321d0-501a-41d0-8816-5711e7ab408f
12-18 12:21:54.415  2504  2521 D BtGatt.GattService: registerClient() - UUID=922321d0-501a-41d0-8816-5711e7ab408f
12-18 12:21:54.416  2504  2936 I bt_stack: [INFO:gatt_api.cc(950)] GATT_Register e30e9475-f3cf-4038-dfbc-17075ee9c589
12-18 12:21:54.416  2504  2936 I bt_stack: [INFO:gatt_api.cc(970)] allocated gatt_if=8
12-18 12:21:54.417  2504  2610 D BtGatt.GattService: onClientRegistered() - UUID=922321d0-501a-41d0-8816-5711e7ab408f, clientIf=8
12-18 12:21:54.417 13688 13705 D BluetoothGatt: onClientRegistered() - status=0 clientIf=8
12-18 12:21:54.418  2504  2521 D BtGatt.GattService: clientConnect() - address=F0:5C:7C:7A:CB:15, isDirect=true, opportunistic=false, phy=1
12-18 12:21:54.419  2504  2610 D bt_btif_config: btif_get_address_type: Device [f0:5c:7c:7a:cb:15] address type 1
12-18 12:21:54.419  2504  2610 D bt_btif_config: btif_get_device_type: Device [f0:5c:7c:7a:cb:15] type 2
12-18 12:21:54.420  2504  2936 I bt_stack: [INFO:gatt_api.cc(1122)] GATT_Connect gatt_if=8, address=f0:5c:7c:7a:cb:15 is_direct 1
12-18 12:21:54.451  2504  2936 I bt_stack: [INFO:connection_manager.cc(277)] lower the scan parameters used for connecting
12-18 12:21:54.452  2504  2936 I bt_stack: [INFO:btm_ble_bgconn.cc(406)] Already connection initiated (or) No Pending connections(or) Resources of lcb are exhausted
12-18 12:21:54.452  2504  2936 W bt_btm  : btm_acl_created hci_handle=9 link_role=0  transport=2
12-18 12:21:54.452  2504  2936 I bt_stack: [INFO:btm_ble_bgconn.cc(406)] Already connection initiated (or) No Pending connections(or) Resources of lcb are exhausted
12-18 12:21:54.523  2504  2936 I bt_stack: [INFO:gatt_main.cc(919)] gatt_data_process op_code = 8, msg_len = 6
12-18 12:21:54.524  2504  2936 E bt_btif : bta_dm_acl_change new acl connetion:count = 1
12-18 12:21:54.524  2504  2936 W bt_btif : bta_dm_acl_change info: 0x0
12-18 12:21:54.524  2504  2936 W bt_l2cap: L2CA_SetDesireRole() new:x1, disallow_switch:0
12-18 12:21:54.524  2504  2936 E bt_stack: [ERROR:bta_gattc_cache.cc(733)] bta_gattc_cache_load: can't open GATT cache file /data/misc/bluetooth/gatt_cache_f05c7c7acb15 for reading, error: No such file or directory
12-18 12:21:54.525  2504  2936 I bt_stack: [INFO:hearing_aid.cc(1804)] GetDeviceCount: Not initialized yet
12-18 12:21:54.525  2504  2936 I bt_stack: [INFO:gatt_api.cc(635)] GATTC_Discover conn_id=0x0008, disc_type=1, s_handle=0x0001, e_handle=0xffff
12-18 12:21:54.526  2504  2610 D bt_btif_dm: remote version info [f0:5c:7c:7a:cb:15]: 0, 0, 0
12-18 12:21:54.532  2504  2610 D BtGatt.GattService: onConnected() - clientIf=8, connId=8, address=F0:5C:7C:7A:CB:15
12-18 12:21:54.533 13688 13705 D BluetoothGatt: onClientConnectionState() - status=0 clientIf=8 device=F0:5C:7C:7A:CB:15
12-18 12:21:54.535  2504 17327 D BluetoothMetrics: btclass1f00
12-18 12:21:54.535  1062  1562 D NativeSensors: CapSensor:  mPendingEvent.distance=1.000000
12-18 12:21:54.537 16806 16806 W BluetoothEventManager: AclStateChangedHandler: activeDevice is null
12-18 12:21:54.539  2524  2757 W BluetoothEventManager: AclStateChangedHandler: activeDevice is null
12-18 12:21:54.541 13688 17245 D BluetoothGatt: configureMTU() - device: F0:5C:7C:7A:CB:15 mtu: 253
12-18 12:21:54.542  2504  2521 D BtGatt.GattService: configureMTU() - address=F0:5C:7C:7A:CB:15 mtu=253
12-18 12:21:54.546 12461 12461 D MCLOG   : [MC_Service] onStartcommand()
12-18 12:21:54.546  2504 17327 D Checkin : publish the event [tag = MOT_BT event name = CONN_SUC]
12-18 12:21:54.562   970   970 D Zygote  : Forked child process 17328
12-18 12:21:54.568  1537  1829 I ActivityManager: Start proc 17328:com.google.android.projection.gearhead:shared/u0a240 for broadcast {com.google.android.projection.gearhead/com.google.android.apps.auto.wireless.bluetooth.WifiBluetoothReceiver}
12-18 12:21:54.575 17328 17328 E gearhead:share: Unknown bits set in runtime_flags: 0x8000
12-18 12:21:54.584 17328 17328 E gearhead:share: Not starting debugger since process cannot load the jdwp agent.
12-18 12:21:54.613  1062  1562 D NativeSensors: CapSensor:  mPendingEvent.distance=2.000000
12-18 12:21:54.624 17328 17328 I gearhead:share: The ClassLoaderContext is a special shared library.
12-18 12:21:54.624  1089  3015 E WifiHAL : Received fatal event, sending alert
12-18 12:21:54.625  1537 12332 W WifiVendorHal: onDebugErrorAlert 4104
12-18 12:21:54.713 17328 17328 D ProcessLifecycleG3: androidx.startup is not available, initializing using ProcessLifecycleOwnerInitializer instead.
12-18 12:21:54.718  2504  2936 I bt_stack: [INFO:gatt_main.cc(919)] gatt_data_process op_code = 17, msg_len = 13
12-18 12:21:54.762 17328 17328 I CAR.GH  : onCreate
12-18 12:21:54.796 17328 17328 I GH.PrimesMgr: transmitting to prod
12-18 12:21:54.804 17328 17353 W gearhead:share: Accessing hidden field Ljava/util/Collections$SynchronizedCollection;->mutex:Ljava/lang/Object; (greylist-max-o, reflection, denied)
12-18 12:21:54.804 17328 17353 W gearhead:share: Accessing hidden method Ljava/util/Collections$SynchronizedSet;-><init>(Ljava/util/Set;Ljava/lang/Object;)V (greylist-max-o, reflection, denied)
12-18 12:21:54.804 17328 17353 W gearhead:share: Accessing hidden method Ljava/util/Collections$SynchronizedCollection;-><init>(Ljava/util/Collection;Ljava/lang/Object;)V (greylist-max-o, reflection, denied)
12-18 12:21:54.806 17328 17328 I CAR.GH  : Process: SHARED_SERVICE
12-18 12:21:54.815  2504  2936 I bt_stack: [INFO:gatt_main.cc(919)] gatt_data_process op_code = 17, msg_len = 21
12-18 12:21:54.886  1537  1614 I system_server: Background concurrent copying GC freed 194192(8799KB) AllocSpace objects, 41(11MB) LOS objects, 42% free, 27MB/47MB, paused 136us total 157.840ms
[... 197 lines cut ...]
12-18 12:21:59.832 13688 14889 I ReactNativeJS: 'ble.js: ', 'F0:5C:7C:7A:CB:15', [ 'F9:F2:19:9D:7A:BA', 'F0:5C:7C:7A:CB:15' ], '', '', '', '', '', '', ''
12-18 12:21:59.834 13688 14889 I ReactNativeJS: 'ble.js: ', 'message to write: ', 'NzIzN0ExQjUtNzkwQS1EMDQ0LTAxMjMtNDU2Nzg5QUJDREUw', '', '', '', '', '', '', ''
12-18 12:21:59.886  2504  2936 I bt_stack: [INFO:gatt_main.cc(919)] gatt_data_process op_code = 19, msg_len = 0
12-18 12:21:59.983  2504  2936 I bt_stack: [INFO:gatt_main.cc(919)] gatt_data_process op_code = 19, msg_len = 0
12-18 12:21:59.984  2504  2936 I bt_stack: [INFO:gatt_main.cc(919)] gatt_data_process op_code = 27, msg_len = 5
12-18 12:21:59.984  2504  2936 E bt_btif : bta_gattc_process_indicate, ignore HID ind/notificiation
12-18 12:21:59.990 13688 14889 I ReactNativeJS: 'ble.js: ', 'char write response:', 'NzIzN0ExQjUtNzkwQS1EMDQ0LTAxMjMtNDU2Nzg5QUJDREUw', 'with time_out of ', 8000, '', '', '', '', ''
12-18 12:21:59.995 13688 14889 I ReactNativeJS: 'StackLog: ', 'Connecting: Connection Successful!', '', '', '', '', '', '', '', ''
12-18 12:22:00.024  2524  2524 W View    : requestLayout() improperly called by com.android.systemui.statusbar.HeadsUpStatusBarView{f602a08 I.E...... ......ID 0,0-194,53 #7f0a01f7 app:id/heads_up_status_bar_view} during layout: running second layout pass
12-18 12:22:00.564  2504  2521 D BtGatt.GattService: getDeviceType() - device=F4:4E:FD:10:9A:59, type=1
12-18 12:22:00.564  2504  2521 D BtGatt.GattService: getDeviceType() - device=9C:19:C2:32:C6:5A, type=3
[... cut ...]
12-18 12:22:00.570  2504  2521 D BtGatt.GattService: registerForNotification() - address=F0:5C:7C:7A:CB:15 enable: false
12-18 12:22:00.570  2504 17445 D BluetoothMetrics: btclass1f00
12-18 12:22:00.570  2504  2521 E BtGatt.GattService: registerForNotification() - No connection for F0:5C:7C:7A:CB:15...
12-18 12:22:00.572  2504 17445 D Checkin : publish the event [tag = MOT_BT event name = DISC]
12-18 12:22:00.575 13688 17245 D BluetoothGatt: close()
12-18 12:22:00.575 13688 17245 D BluetoothGatt: unregisterApp() - mClientIf=8
```

budowski, 2023-12-31:
Happens to me as well (only on Android). Works fine on version 2.0.3

eliaslecomte, 2024-01-09:
I have the same issue 🤔.

GerardoStella, 2024-01-11:
I have the same issue

p4bl1t0, 2024-01-12:
@dominik-czupryna-withintent @dimninik can you help us to find the issue?

leoafarias, 2024-01-19:
Don't want to do just a +1, but this seems like an important issue.

andreyukD, 2024-02-07:
Same situation for me, Android 2.0.3 works fine, 3.1.2 gets disconnected after max 30s
apparently, disconnection time depends on timeout in the `connectToDevice` method
```ts
await bleManagerDevice.connectToDevice(device.id, { timeout: 15000 }); 
// disconnect will be after 15s. but increasing timeout to more than 30000 won't increase connection time 
```

p4bl1t0, 2024-02-07:
@andreyukD thats works for you?

andreyukD, 2024-02-07:
@p4bl1t0 version 3.1.2 works only for a few seconds (max 30 seconds - if I set timeout: 30000), if I set timeout 5000 it will work for a 5s, but if I set for instance timeout: 60000 it will be disconnected after ~30 seconds 

version 2.0.3. works fine

macksal, 2024-02-19:
I found the disconnect originates from the onFinally clause here. https://github.com/dotintent/react-native-ble-plx/blob/c60eff0d1f040dfe1ef80880c040cf58bab55337/android/src/main/java/com/bleplx/adapter/BleModule.java#L1227-L1234

The problem is that the timeout is added just below, constructed from the original observable.

I believe that when the observable made on L1260 errors, it will unsubscribe from the original one constructed on L1227, which triggers the doFinally and disconnects.

I don't know Rx for languages other than js. I suppose doFinally is meant to cancel the connection when the connection observable is unsubscribed from. I think the connection can be interrupted as the subscription made on L1264 can be disposed later through the `replaceSubscription` method, so something is needed there. I'm not sure doFinally is the right mechanism to do that, though.

#1135 is a duplicate.

p4bl1t0, 2024-02-19:
Great find @macksal 

It's a pitty that dotintent team is so silenced

EmmaZachara, 2024-02-20:
Hi @macksal, @p4bl1t0 ,
Thank you for reporting the issue. We are analyzing it and will get back to you with more information soon.

dominik-czupryna-withintent, 2024-02-25:
Hi,
I have quite a big problem with this error, because I am unable to reproduce it, but it is present in couple issues. I also don't want the problem to be fixed blindly without example. In thread #1174 @joanmarcs mentioned that the problem occurs on Samsung s22 Ultra and One plus, unfortunately I do not have either of these devices. I tried to reproduce this problem on Xiaomi mi 6 and 10, Motorola g82 and Samsung M13 in both development and production applications, but unfortunately I was unable to do so. A certain problem is that the event appears in your logs: "BluetoothGatt: cancelOpen() - device: F0:5C:7C:7A:CB:15", which identifies that the connection should be terminated. 
Following the android source https://android.googlesource.com/platform/frameworks/base/+/364ed4d/core/java/android/bluetooth/BluetoothGatt.java:

```
    * Disconnects an established connection, or cancels a connection attempt
    * currently in progress.
    *
    * <p>Requires {@link android.Manifest.permission#BLUETOOTH} permission.
    */
    public void disconnect() {
    if (DBG) Log.d(TAG, "cancelOpen() - device: " + mDevice.getAddress());
    if (mService == null \|\| mClientIf == 0) return;
     
    try {
    mService.clientDisconnect(mClientIf, mDevice.getAddress());
    } catch (RemoteException e) {
    Log.e(TAG,"",e);
    }
    }
```
This function is mainly called in two cases. The first one is when the cancelDeviceConnection function is called, and the second one is when the timeout (https://dotintent.github.io/react-native-ble-plx/#connectionoptions) time that you specify when connecting run out. An important change is the fact that when using v2 the timeout parameter does not work (at least for me) and the connection is not terminated, while in v3 this mechanism works.

BluetoothGatt logs when timeout is set in v3:
```
2024-02-25 23:34:05.806 24239-24498 BluetoothGatt           com.i1157                            D  connect() - device: 55:D6:CF:22:13:7A, auto: false, eattSupport: false
2024-02-25 23:34:05.806 24239-24498 BluetoothGatt           com.i1157                            D  registerApp()
2024-02-25 23:34:05.806 24239-24498 BluetoothGatt           com.i1157                            D  registerApp() - UUID=b2e5621e-be0b-48f8-bc20-fc8f69b7f0a8
2024-02-25 23:34:05.810 24239-24259 BluetoothGatt           com.i1157                            D  onClientRegistered() - status=0 clientIf=9
2024-02-25 23:34:06.301 24239-24259 BluetoothGatt           com.i1157                            D  onClientConnectionState() - status=0 clientIf=9 device=55:D6:CF:22:13:7A
2024-02-25 23:34:06.697 24239-24259 BluetoothGatt           com.i1157                            D  onPhyUpdate() - status=0 address=55:D6:CF:22:13:7A txPhy=2 rxPhy=2
2024-02-25 23:34:06.745 24239-24259 BluetoothGatt           com.i1157                            D  onConnectionUpdated() - Device=55:D6:CF:22:13:7A interval=36 latency=0 timeout=500 status=59
2024-02-25 23:34:08.141 24239-24259 BluetoothGatt           com.i1157                            D  onConnectionUpdated() - Device=55:D6:CF:22:13:7A interval=6 latency=0 timeout=500 status=0
2024-02-25 23:34:25.918 24239-24498 BluetoothGatt           com.i1157                            D  discoverServices() - device: 55:D6:CF:22:13:7A
2024-02-25 23:34:25.921 24239-24329 BluetoothGatt           com.i1157                            D  onSearchComplete() = Device=55:D6:CF:22:13:7A Status=0
2024-02-25 23:35:06.343 24239-24498 BluetoothGatt           com.i1157                            D  cancelOpen() - device: 55:D6:CF:22:13:7A
2024-02-25 23:35:06.347 24239-24466 BluetoothGatt           com.i1157                            D  onClientConnectionState() - status=0 clientIf=9 device=55:D6:CF:22:13:7A
2024-02-25 23:35:06.350 24239-24498 BluetoothGatt           com.i1157                            D  close()
2024-02-25 23:35:06.350 24239-24498 BluetoothGatt           com.i1157                            D  unregisterApp() - mClientIf=9
```

BluetoothGatt logs when timeout is set in v2:
```
2024-02-25 23:56:24.045 28846-31711 BluetoothGatt           com.i1157                            D  connect() - device: 58:C1:3C:30:27:E0, auto: false, eattSupport: false
2024-02-25 23:56:24.045 28846-31711 BluetoothGatt           com.i1157                            D  registerApp()
2024-02-25 23:56:24.045 28846-31711 BluetoothGatt           com.i1157                            D  registerApp() - UUID=3efda5b8-4e31-48c1-b8d1-8d06573be3bf
2024-02-25 23:56:24.047 28846-29093 BluetoothGatt           com.i1157                            D  onClientRegistered() - status=0 clientIf=10
2024-02-25 23:56:24.053 28846-29093 BluetoothGatt           com.i1157                            D  onClientConnectionState() - status=0 clientIf=10 device=58:C1:3C:30:27:E0
2024-02-25 23:56:24.054 28846-29093 RxBle#BluetoothGatt$1   com.i1157                            D  onConnectionStateChange newState=2 status=0
2024-02-25 23:56:29.766 28846-31711 BluetoothGatt           com.i1157                            D  discoverServices() - device: 58:C1:3C:30:27:E0
2024-02-25 23:56:29.769 28846-29093 BluetoothGatt           com.i1157                            D  onSearchComplete() = Device=58:C1:3C:30:27:E0 Status=0
2024-02-25 23:56:29.770 28846-29093 RxBle#BluetoothGatt$1   com.i1157                            D  onServicesDiscovered status=0
[... cut ...]
2024-02-26 00:27:27.095  7741-8480  BluetoothGatt           com.i1157                            D  onClientConnectionState() - status=0 clientIf=9 device=4B:30:1D:5D:DE:82
2024-02-26 00:27:27.099  7741-8238  BluetoothGatt           com.i1157                            D  close()
2024-02-26 00:27:27.099  7741-8238  BluetoothGatt           com.i1157                            D  unregisterApp() - mClientIf=9
```

Could you please show me the process you implemented to handle the connection and what devices you use? I'm stuck at the moment and don't know what I could do to get cancelOpen() called in different case.

andreyukD, 2024-02-29:
I've debugged both versions a bit, so I have two screens:
  1. ConnectionScreen (starts device scan -> connects to the BLE device -> navigates to ReadCharacteristicScreen)
  2. ReadCharacteristicScreen (every second reads from the BLE device a characteristic)
---
1. Connection screen:
```ts
const processJoinBleQr = (
  data: BarCodeScannerResult,
  setBleDevices: React.Dispatch<React.SetStateAction<Device[]>>,
  navigation: UserStackScreenProps<Routes.AddFromQRScreen>['navigation']
) => {
  requestPermissions((isGranted) => {
    if (isGranted) {
      const onDeviceDisconnected = async (device: Device) => {
        const deviceId = device.id;
        if (device) {
          const subscription = device.onDisconnected((error, device) => {
            Alert.alert(translate('ui.bluetoothDevice.deviceWasDisconnected'), `ID: ${deviceId}`);
            setBleDevices((prev) => prev.filter((el) => el.id !== deviceId));
            subscription.remove();
          });
        }
      };

      const disconnectFromDevice = (id: string) => {
        bleManagerDevice.cancelDeviceConnection(id);
        setBleDevices((prev) => prev.filter((el) => el.id !== id));
      };

      const stopScanning = () => {
        bleManagerDevice.stopDeviceScan();
      };

      bleManagerDevice.startDeviceScan(undefined, null, async (error, device) => {
        if (error) {
          Alert.alert(error.message);
          return navigation.goBack();
        }

        if (
          device &&
          [device.localName, device.name].some((el) => el?.startsWith('BLE-MY-DEVICE-123'))
        ) {
          await stopScanning();

          try {
            const isConnected = await device.isConnected(); // check if device was previously connected

            const deviceConnection = isConnected
              ? device
              : await bleManagerDevice.connectToDevice(device.id, {
                  timeout: 15000,
                });

            setBleDevices((prev) =>
              !prev.find((el) => el.id === deviceConnection.id) ? [...prev, deviceConnection] : prev
            );
            await deviceConnection.discoverAllServicesAndCharacteristics();
            stopScanning();
            !isConnected && onDeviceDisconnected(deviceConnection);

            return navigation.replace(Routes.MainStack, {
              screen: Routes.ReadCharacteristicScreen,
              params: { deviceId: device.id },
            });
          } catch (e) {
            stopScanning();
            if (await device.isConnected()) disconnectFromDevice(device.id);
            Alert.alert(JSON.stringify(e, Object.getOwnPropertyNames(e)));
            return navigation.goBack();
[... 94 lines cut ...]

If I trigger manual disconnect

2024-02-29 23:10:16.042 10297-21717 BluetoothGatt           com.foo                      D  cancelOpen() - device: 70:F7:84:E4:FC:9D
2024-02-29 23:10:16.065 10297-20360 BluetoothGatt           com.foo                      D  onClientConnectionState() - status=0 clientIf=10 device=70:F7:84:E4:FC:9D
2024-02-29 23:10:16.070 10297-21717 BluetoothGatt           com.foo                      D  close()
2024-02-29 23:10:16.070 10297-21717 BluetoothGatt           com.foo                      D  unregisterApp() - mClientIf=10
```

Logs from 3.1.2 (auto disconnect)
```
2024-02-29 23:13:13.778 12979-6311  BluetoothGatt           com.foo                      D  connect() - device: 45:54:2A:33:14:1C, auto: false
2024-02-29 23:13:13.778 12979-6311  BluetoothGatt           com.foo                      D  registerApp()
2024-02-29 23:13:13.778 12979-6311  BluetoothGatt           com.foo                      D  registerApp() - UUID=b32a98f2-966d-4f38-b959-faea350bc154
2024-02-29 23:13:13.783 12979-5081  BluetoothGatt           com.foo                      D  onClientRegistered() - status=0 clientIf=10
2024-02-29 23:13:13.966 12979-6297  BluetoothGatt           com.foo                      D  onClientConnectionState() - status=0 clientIf=10 device=45:54:2A:33:14:1C
2024-02-29 23:13:14.373 12979-5089  BluetoothGatt           com.foo                      D  onConnectionUpdated() - Device=45:54:2A:33:14:1C interval=6 latency=0 timeout=500 status=0
2024-02-29 23:13:14.404 12979-6311  BluetoothGatt           com.foo                      D  discoverServices() - device: 45:54:2A:33:14:1C
2024-02-29 23:13:15.894 12979-6335  BluetoothGatt           com.foo                      D  onSearchComplete() = Device=45:54:2A:33:14:1C Status=0
2024-02-29 23:13:15.994 12979-5084  BluetoothGatt           com.foo                      D  onConnectionUpdated() - Device=45:54:2A:33:14:1C interval=39 latency=0 timeout=500 status=0
2024-02-29 23:13:29.035 12979-6311  BluetoothGatt           com.foo                      D  cancelOpen() - device: 45:54:2A:33:14:1C
2024-02-29 23:13:29.057 12979-5089  BluetoothGatt           com.foo                      D  onClientConnectionState() - status=0 clientIf=10 device=45:54:2A:33:14:1C
2024-02-29 23:13:29.062 12979-6311  BluetoothGatt           com.foo                      D  close()
2024-02-29 23:13:29.062 12979-6311  BluetoothGatt           com.foo                      D  unregisterApp() - mClientIf=10
```

dominik-czupryna-withintent, 2024-02-29:
@andreyukD
```
await bleManagerDevice.connectToDevice(device.id, {
                  timeout: 15000,
                })
```
timeout paramter works in 3.1.2 and doesn't work on 2.0.3 and you are declared that the device should be disconnected after 15 seconds and it is (29.035 - 13.778 = 15.257). So should it be autodisconnected or not?

andreyukD, 2024-03-01:
@dominik-czupryna-withintent 
I mentioned that in 3.1.2 disconnection time depends on timeout in the connectToDevice method
and logs confirm this (29 - 13 = ~15). But this is a bug and it shouldn't behave like that

dominik-czupryna-withintent, 2024-03-01:
Okay, I understood, error confirmed. We're working on it.

lucaslt89, 2024-04-17:
As stated by @macksal above, removing the timeout to the connection is a functional workaround and the connection doesn't drop.

joanmarcs, 2024-06-05:
The problem persist in the new version 3.2.0 @dominik-czupryna-withintent

closed 2024-07-31 by intent-kacper-cyranowski

adumat, 2024-09-27:
when this pr will be released?

alex-freeaim, 2024-10-17:
This continues to be an issue in version 3.2.0.

adumat, 2024-10-17:
@alex-freeaim the pr is merged but no new release was made after the merge 😐

tomaseliasm, 2024-10-17:
@adumat @alex-freeaim you can use the latest version pointing to the master branch
"react-native-ble-plx": "dotintent/react-native-ble-plx#master",

alex-freeaim, 2024-10-18:
Okay, I made a fix already that I tested on android and ios, so I can wait until the next release. Below is my fix, where it will time out after 10 seconds.

```
                setTimeout(async () => {
                    const deviceConnected = await this.bleManager.isDeviceConnected(deviceId);
                    if(!deviceConnected) {
                        console.log("Timeout waiting for device to connect");
                        this.bleManager.cancelDeviceConnection(deviceId);
                    }
                }, 10000);
                this.connectedDevice = await this.bleManager.connectToDevice(deviceId);
```