dotintent/react-native-ble-plx #1158: Cannot scan for devices (With error message)

iSaBo96, 2023-12-25:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

Hello everybody,

i want to test the example app. So i checked out the app and run the app on my android device.
I use Android 12.

When i log the startDevice information, i get the following message:

`{"_manager": {"_activePromises": {}, "_activeSubscriptions": {"1": [Object]}, "_errorCodesToMessagesMapping": {"0": "Unknown error occurred. This is probably a bug! Check reason property.", "1": "BleManager was destroyed", "100": "BluetoothLE is unsupported on this device", "101": "Device is not authorized to use BluetoothLE", "102": "BluetoothLE is powered off", "103": "BluetoothLE is in unknown state", "104": "BluetoothLE is resetting", "105": "Bluetooth state change failed", "2": "Operation was cancelled", "200": "Device {deviceID} connection failed", "201": "Device {deviceID} was disconnected", "202": "RSSI read failed for device {deviceID}", "203": "Device {deviceID} is already connected", "204": "Device {deviceID} not found", "205": "Device {deviceID} is not connected", "206": "Device {deviceID} could not change MTU size", "3": "Operation timed out", "300": "Services discovery failed for device {deviceID}", "301": "Included services discovery failed for device {deviceID} and service: {serviceUUID}", "302": "Service {serviceUUID} for device {deviceID} not found", "303": "Services not discovered for device {deviceID}", "4": "Operation was rejected", "400": "Characteristic discovery failed for device {deviceID} and service {serviceUUID}", "401": "Characteristic {characteristicUUID} write failed for device {deviceID} and service {serviceUUID}", "402": "Characteristic {characteristicUUID} read failed for device {deviceID} and service {serviceUUID}", "403": "Characteristic {characteristicUUID} notify change failed for device {deviceID} and service {serviceUUID}", "404": "Characteristic {characteristicUUID} not found", "405": "Characteristics not discovered for device {deviceID} and service {serviceUUID}", "406": "Cannot write to characteristic {characteristicUUID} with invalid data format: {internalMessage}", "5": "Invalid UUIDs or IDs were passed: {internalMessage}", "500": "Descriptor {descriptorUUID} discovery failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "501": "Descriptor {descriptorUUID} write failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "502": "Descriptor {descriptorUUID} read failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "503": "Descriptor {descriptorUUID} not found", "504": "Descriptors not discovered for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}", "505": "Cannot write to descriptor {descriptorUUID} with invalid data format: {internalMessage}", "506": "Cannot write to descriptor {descriptorUUID}. It's not allowed by iOS and therefore forbidden on Android as well.", "600": "Cannot start scanning operation", "601": "Location services are disabled"}, "_eventEmitter": {"_nativeModule": [Object]}, "_scanEventSubscription": {"remove": [Function remove]}, "_uniqueId": 3}, "id": "61:0B:C3:C9:70:9E", "isConnectable": true, "localName": null, "manufacturerData": "TAAQBjYaiJB/9Q==", "mtu": 23, "name": null, "overflowServiceUUIDs": null, "rawScanRecord": "AgEaAgoLC/9MABAGNhqIkH/1AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA=", "rssi": -66, "serviceData": null, "serviceUUIDs": null, "solicitedServiceUUIDs": null, "txPowerLevel": 11}`

Does anybody know why and whats wrong there?
Additional: Bluetooth is on, all permissions granted (i checked this in the app settings)
To make sure i have granted permissions, i add a additonal button to ask for the permissions and granted

Steps to reproduce:
- Enable bluetooth
- Checkout the example app
- Run "npm start"
- Run "npm run android"

dominik-czupryna-withintent, 2024-02-16:
You have logged the function with all manager options, which also shows all possible errors. Could you tell me what you wanted to do? Information about your device is also included here:
[image].
Why are you checking a function instead the output of its listener?

flexbox, 2024-06-17:
Hey @iSaBo96 

If your problem is solved can you please close the issue? 
It sparks joy for maintainers ✨

intent-kacper-cyranowski, 2024-06-19:
Hi @iSaBo96 
Do you still need help with the issue?

closed 2024-06-19 by iSaBo96