dotintent/react-native-ble-plx #1160: 🐛  OnDisconnected() never reporting error

Cheryet, 2024-01-05:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

When a device is disconnected from BLE, The OnDisconnected() callback reports an error related to disconnection. When a disconnection is successful via a call to bleManager.cancelDeviceConnection() no error should be shown. When a connection is lost, an error should be reported.

### Current Behavior

While setting up my disconnection handler I am never seeing an error for device.OnDisconnected() or manager.onDeviceDisconnected(DeviceId). Although the callback gets called, the Error parameter is always null in my case.

### Library version

3.1.2-rc.0

### Device

IPhone 14 (IOS 17.2), Samsung Galaxy S8 (Android 9)

### Environment info

```shell
System:
    OS: macOS 13.6.1
    CPU: (8) arm64 Apple M2
    Memory: 251.31 MB / 16.00 GB
    Shell: 5.9 - /bin/zsh
  Binaries:
    Node: 14.18.2 - ~/.nvm/versions/node/v14.18.2/bin/node
    Yarn: Not Found
    npm: 6.14.15 - ~/.nvm/versions/node/v14.18.2/bin/npm
    Watchman: 2023.12.04.00 - /opt/homebrew/bin/watchman
  Managers:
    CocoaPods: 1.10.2 - /Users/cheryet/.rbenv/shims/pod
  SDKs:
    iOS SDK:
      Platforms: DriverKit 23.2, iOS 17.2, macOS 14.2, tvOS 17.2, watchOS 10.2
    Android SDK: Not Found
  IDEs:
    Android Studio: 2022.2 AI-222.4459.24.2221.10121639
    Xcode: 15.1/15C65 - /usr/bin/xcodebuild
  Languages:
    Java: 11.0.19 - /usr/bin/javac
  npmPackages:
    @react-native-community/cli: Not Found
    react: 18.2.0 => 18.2.0 
    react-native: ^0.71.2 => 0.71.2 
    react-native-macos: Not Found
  npmGlobalPackages:
    *react-native*: Not Found
```

### Steps to reproduce

1. Create disconect handler for device disconnection
2. connect to device via ble
3. remove power from ble hardware to produce loss of connection
4. no error is shown

### Formatted code sample or link to a repository

```js
connectToDevice(peripheralId, onDisconnect) {
    try {
      await this.manager.connectToDevice(peripheralId, {
        timeout: 30000,
        refreshGatt: 'OnConnected',
      });
[... 439 lines cut ...]
        '302': 'Service {serviceUUID} for device {deviceID} not found',
        '303': 'Services not discovered for device {deviceID}',
        '400': 'Characteristic discovery failed for device {deviceID} and service {serviceUUID}',
        '401': 'Characteristic {characteristicUUID} write failed for device {deviceID} and service {serviceUUID}',
        '402': 'Characteristic {characteristicUUID} read failed for device {deviceID} and service {serviceUUID}',
        '403': 'Characteristic {characteristicUUID} notify change failed for device {deviceID} and service {serviceUUID}',
        '404': 'Characteristic {characteristicUUID} not found',
        '405': 'Characteristics not discovered for device {deviceID} and service {serviceUUID}',
        '406': 'Cannot write to characteristic {characteristicUUID} with invalid data format: {internalMessage}',
        '500': 'Descriptor {descriptorUUID} discovery failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}',
        '501': 'Descriptor {descriptorUUID} write failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}',
        '502': 'Descriptor {descriptorUUID} read failed for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}',
        '503': 'Descriptor {descriptorUUID} not found',
        '504': 'Descriptors not discovered for device {deviceID}, service {serviceUUID} and characteristic {characteristicUUID}',
        '505': 'Cannot write to descriptor {descriptorUUID} with invalid data format: {internalMessage}',
        '506': 'Cannot write to descriptor {descriptorUUID}. It\'s not allowed by iOS and therefore forbidden on Android as well.',
        '600': 'Cannot start scanning operation',
        '601': 'Location services are disabled' },
     _scanEventSubscription: null } }
'From Client: Device is disconnected. Error: ', null
```

### Additional information

I have tried multiple versions of BLE PLX to attempt to fix this issue.

zachTrio, 2024-01-24:
I've updated to 3.1.2 in hopes it would fix this issue as it was in the patch notes.

This is also still happening to me, despite the issue being resolved in that release.

#1126

zachTrio, 2024-02-19:
> How long have you been waiting for this event?

Well, I am not receiving the event during the tens of minutes I may set the phone down after testing.

The fact it's not happening in the same time frame as the iOS disconnect is suggesting it's not going to happen at all.

Cheryet, 2024-02-20:
> How long have you been waiting for this event? 

Similar here, Not receiving the event during the 10+ minuets after a disconnection has been made.

tbarusseau, 2025-02-12:
Hello, I have the same issue. Nowhere in my codebase do I call the `cancelDeviceConnection` method. My `bleManager` is also not being destroyed or garbage collected.

Wondering which steps I can take to identify why my device actually gets disconnected?

Note that I'm running the latest version: `3.5.0`

DmitryOsambit, 2025-03-31:
I'm having the same issue.
According to the code in the `BlePlxModule.java`, the reason for this is simply that the error is never passed to the `DisconnectionEvent` payload. 

```java
new OnEventCallback<ConnectionState>() {
        @Override
        public void onEvent(ConnectionState connectionState) {
          if (connectionState == ConnectionState.DISCONNECTED) {
            WritableArray event = Arguments.createArray();
            event.pushNull();
            WritableMap device = Arguments.createMap();
            device.putString("id", deviceId);
            event.pushMap(device);
            sendEvent(Event.DisconnectionEvent, event);
          }
        }
      }
```

Hence we are not getting any error in the React Native `BleManager.js` code:
```javascript
onDeviceDisconnected(deviceIdentifier: DeviceId, listener: (error: ?BleError, device: Device) => void): Subscription {
    const disconnectionListener = ([error, nativeDevice]: [?string, NativeDevice]) => {
      if (deviceIdentifier !== nativeDevice.id) {
        return
      }
      listener(error ? parseBleError(error, this._errorCodesToMessagesMapping) : null, new Device(nativeDevice, this))
    }

    const subscription: Subscription = this._eventEmitter.addListener(
      BleModule.DisconnectionEvent,
      disconnectionListener
    )

    const id = this._nextUniqueID()
    const wrappedSubscription = {
      remove: () => {
        if (this._activeSubscriptions[id] != null) {
          delete this._activeSubscriptions[id]
          subscription.remove()
        }
      }
    }
    this._activeSubscriptions[id] = wrappedSubscription
    return wrappedSubscription
  }
```