dotintent/react-native-ble-plx #1161: TypeError: null is not an object (evaluating '_BleModule.BleModule.createClient')🐛

Elnatanv, 2024-01-11:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

I want to run `new BleManager()` without getting the "createClient" error. (on react native + expo)

### Current Behavior

After updating to version 3.1.2, im getting an error: `TypeError: null is not an object (evaluating '_BleModule.BleModule.createClient')`  when running `new BleManager().` 

I was using before version 2.0.3 and it worked without any problems. i could fire `new BleManager()` i could scan, connect,read and write. everything worked as expected. 

### Library version

3.1.2

### Device

Android 13

### Environment info

```shell
Expo - 49.0.21
react native:
System:
  OS: Windows 10 10.0.22621
  CPU: (8) x64 11th Gen Intel(R) Core(TM) i5-1135G7 @ 2.40GHz
  Memory: 14.08 GB / 31.77 GB
Binaries:
  Node:
    version: 18.18.0
    path: C:\Program Files\nodejs\node.EXE
  Yarn: Not Found
  npm:
    version: 10.2.2
    path: C:\Program Files\nodejs\npm.CMD
  Watchman: Not Found
SDKs:
  Android SDK:
    API Levels:
      - "31"
      - "33"
    Build Tools:
      - 30.0.3
      - 31.0.0
      - 33.0.2
    System Images:
      - android-33 | Google APIs Intel x86_64 Atom
    Android NDK: Not Found
  Windows SDK: Not Found
IDEs:
  Android Studio: AI-222.4459.24.2221.9971841
  Visual Studio: Not Found
Languages:
  Java: 17.0.7
  Ruby:
    version: 3.1.2
    path: C:\Ruby31-x64\bin\ruby.EXE
npmPackages:
  "@react-native-community/cli": Not Found
  react:
    installed: 18.2.0
    wanted: 18.2.0
  react-native:
    installed: 0.72.6
    wanted: 0.72.6
[... 187 lines cut ...]
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:null in renderWithHooks
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:null in mountIndeterminateComponent
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:null in beginWork$1
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:null in performUnitOfWork
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:null in workLoopSync
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:null in renderRootSync
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:null in performSyncWorkOnRoot
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:null in flushSyncCallbacks
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:null in flushSyncCallbacksOnlyInLegacyMode
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:null in scheduleUpdateOnFiber
at node_modules\react-native\Libraries\Renderer\implementations\ReactNativeRenderer-dev.js:null in dispatchSetState
at app\utils\AuthProvider\AuthProvider.js:null in AuthProvider
at app\utils\AuthProvider\AuthProvider.js:null in auth.onAuthStateChanged$argument_0
at node_modules\firebase\node_modules\@firebase\auth\dist\rn\index-3d9ec48b.js:null in AuthImpl.prototype.registerStateListener
at node_modules\react-native\node_modules\promise\setimmediate\core.js:null in tryCallOne
at node_modules\react-native\node_modules\promise\setimmediate\core.js:null in setImmediate$argument_0
at node_modules\react-native\Libraries\Core\Timers\JSTimers.js:null in queueReactNativeMicrotask
at node_modules\react-native\Libraries\Core\Timers\JSTimers.js:null in _callTimer
at node_modules\react-native\Libraries\Core\Timers\JSTimers.js:null in _callReactNativeMicrotasksPass
at node_modules\react-native\Libraries\Core\Timers\JSTimers.js:null in setTimeout$argument_0
at node_modules\react-native\Libraries\BatchedBridge\MessageQueue.js:null in __callReactNativeMicrotasks
at node_modules\react-native\Libraries\BatchedBridge\MessageQueue.js:null in flushedQueue
at node_modules\react-native\Libraries\BatchedBridge\MessageQueue.js:null in __guard
at node_modules\react-native\Libraries\BatchedBridge\MessageQueue.js:null in flushedQueue
```

bj97301, 2024-01-30:
I believe you need to update your expo dev client(make a new build on eas)

Elnatanv, 2024-02-13:
Updating expo solved the issue. Thank you.

closed 2024-02-16 by dominik-czupryna-withintent