dotintent/react-native-ble-plx #1162: 🐛 Connection Error 2 "Operation was cancelled" on android

carllewisc, 2024-01-12:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

Connect with a device when an APK generated on assembleRelease.

### Current Behavior

When I try to connect with a device using an APK generated on assembleRelease or from the Play Store, the response is false. But if I try this process in development, I can connect with a device without a problem, and the response is true. 

This happens on Android.

 'Connect Error', '{"message":"Operation was cancelled","errorCode":2,"attErrorCode":null,"iosErrorCode":null,"androidErrorCode":null,"reason":null,"name":"BleError"}', [BleError: Operation was cancelled]

### Library version

2.0.1, 3.1.2

### Device

Google Pixel 6 (Android 14)

### Environment info

```shell
System:
    OS: macOS 12.6.2
    CPU: (10) arm64 Apple M1 Pro
    Memory: 212.72 MB / 32.00 GB
    Shell: 5.8.1 - /bin/zsh
  Binaries:
    Node: 16.14.0 - ~/.nvm/versions/node/v16.14.0/bin/node
    Yarn: 1.22.19 - /opt/homebrew/bin/yarn
    npm: 9.3.1 - /opt/homebrew/bin/npm
    Watchman: 2023.12.04.00 - /opt/homebrew/bin/watchman
  Managers:
    CocoaPods: 1.12.1 - /Users/macbook/.rvm/gems/ruby-2.7.6/bin/pod
  SDKs:
    iOS SDK:
      Platforms: DriverKit 22.2, iOS 16.2, macOS 13.1, tvOS 16.1, watchOS 9.1
    Android SDK: Not Found
  IDEs:
    Android Studio: 2022.2 AI-222.4459.24.2221.9971841
    Xcode: 14.2/14C18 - /usr/bin/xcodebuild
  Languages:
    Java: 11.0.18 - /usr/bin/javac
  npmPackages:
    @react-native-community/cli: Not Found
    react: 17.0.1 => 17.0.1
    react-native: 0.64.4 => 0.64.4
    react-native-macos: Not Found
  npmGlobalPackages:
    *react-native*: Not Found
```

### Steps to reproduce

1. … Generate apk on assembleRelease
2. … Trying to connect with a device

### Formatted code sample or link to a repository

```js
const create = () => {
  const manager = new BleManager()

  discoverAllServicesAndCharacteristicsForDevice = (deviceId, transactionId = null) => {
    return manager.discoverAllServicesAndCharacteristicsForDevice(deviceId, transactionId)
      .then((device) => { return device })
  }
  
    connect = (deviceId) => {
    return this.connectToDevice(deviceId)
      .then(() => { return this.discoverAllServicesAndCharacteristicsForDevice(deviceId) })
      .then((device) => { return { ok: true, data: { device } } })
      .catch((err) => {
        const { errorCode } = err || {}
        return { ok: false, data: { err, bleOff: errorCode === BleErrorCode.BluetoothPoweredOff } }
      })
  }

}

      for (let index = 0; index < arrayFiltered.length; index++) {
        const item = arrayFiltered[index];
        const connectResponse = yield call(bleApi.connect, item.id)

        if (Config.api.isLogSendingActive) {
          yield call(ApiGlobal.sendLog, { message: 'WizardSagas _scanForBleId - Connection result', data: { deviceId: item.id, connectResponseOk: connectResponse.ok } });
        }

        if (connectResponse.ok) {
          const deviceInfo = yield call(_bleGetDeviceInfoReadForDeviceInfo, bleApi, 'mini', item.id, false)
          if (deviceInfo?.type === 'MINI') {
            bleId = deviceInfo.bleId
            break
          }
        }
      }
```

### Relevant log output

```shell
'Connect Error', '{"message":"Operation was cancelled","errorCode":2,"attErrorCode":null,"iosErrorCode":null,"androidErrorCode":null,"reason":null,"name":"BleError"}', [BleError: Operation was cancelled]
```

dominik-czupryna-withintent, 2024-02-16:
Could you also show in what context the `create` function is called? The generated application throws a lot of unnecessary operations and it is possible that some race condition is revealed.

intent-kacper-cyranowski, 2024-06-25:
Hi @carllewisc 
Is your issue still relevant, do you need any help?

carllewisc, 2024-07-01:
Hi @intent-kacper-cyranowski

This issue was resolved

closed 2024-07-01 by intent-kacper-cyranowski

dmacpro91, 2024-07-12:
> Hi @intent-kacper-cyranowski
> 
> This issue was resolved

We are also experiencing something similar. What ended up resolving it for you?

Mr-Robot-err-404, 2025-01-22:
mind sharing the solution?