dotintent/react-native-ble-plx #1164: [HELP NEEDED] Auto connect feature

Sanjaya94, 2024-01-12:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

Is there any way to implement an auto-connect feature once the BLE device is in the range?

dominik-czupryna-withintent, 2024-02-16:
https://dotintent.github.io/react-native-ble-plx/#connectionoptionsautoconnect

EmmaZachara, 2024-04-19:
Hi @Sanjaya94, did that cover your needs? Do you have any other questions or can we close this issue?

flexbox, 2024-06-12:
@Sanjaya94 do you have a more specific use case?

I have more or less the same workflow but I am not sure.
on iOS, after successfully paired a device I want to "Auto connect".

1. turn on the device _—my case Nacon R5P_ 
2. iOS detect the device on Settings > Bluetooth
3. but it's impossible to "Auto connect" because the device is somehow not visible.

I need to reset the pairing to connect again.

Here is an example:
https://github.com/dotintent/react-native-ble-plx/assets/360936/da83b41c-84bc-4190-b49f-b85cd159c9cd

@dominik-czupryna-withintent 
The reference you pointed out is only for android. Do you know if we can use it anyway on all platforms? because the docs said it's `[Not used]` 😅 
https://github.com/dotintent/react-native-ble-plx/blob/8893a5e11b80fd294aa4da3e50416542e7feb983/src/index.d.ts#L173

intent-kacper-cyranowski, 2024-06-19:
Hi @flexbox 
On the provided video it looks like your device is not discovered because it's already connected to the phone. In case of a device that is already connected to your phone, but not used by your app, I think you should look at [connectedDevices](https://dotintent.github.io/react-native-ble-plx/#blemanagerconnecteddevices).
Let me know if this helps you, in case of any problems feel free to open another issue

flexbox, 2024-06-25:
@intent-kacper-cyranowski 🚀 Thank you very much!

I can successfully use `connectedDevices` to get the connected devices at the Operating System level.
[image]

I was confuse because I used the example code from `example/src/services/BLEService/BLEService.ts`
https://github.com/dotintent/react-native-ble-plx/blob/8893a5e11b80fd294aa4da3e50416542e7feb983/example/src/services/BLEService/BLEService.ts#L312

and the service return an error with `getConnectedDevices` if it's not connected, so that's why I encounter an error every time 😅

intent-kacper-cyranowski, 2024-06-25:
I'm glad that I could help 😄 
I'm going to close the issue as it's author seems inactive

closed 2024-06-25 by intent-kacper-cyranowski