dotintent/react-native-ble-plx #1166: Handle Raw Data in notification

Viswa1999, 2024-01-17:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

I'm running a React Native with ble-plx library. For handling a large amount of data through notification with a time interval of 5 milliseconds is not possible. Here is the question, How to handle the large amount of data in ble-plx? Should I change anything in the libraries? How can I get the data in the time interval of 5 ms in React native?

dominik-czupryna-withintent, 2024-02-16:
Does your device use Indications or Notifications? This question is quite complicated because with Notifications it is possible to lose data. It is possible that the queue on your device is overflowing.

intent-kacper-cyranowski, 2024-06-19:
Hi @Viswa1999 
Do you still need help with the issue?

laxminarayana-capgemini, 2024-06-21:
Hi @intent-kacper-cyranowski,
No, Thanks. I have already completed the Notification part

closed 2024-06-21 by intent-kacper-cyranowski