dotintent/react-native-ble-plx #1168: 🐛 android still connected to bt device after force kill of app

bj97301, 2024-01-27:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

shouldnt remember state after killing the app

### Current Behavior

 thinks its still connected on android but not ios

### Library version

    "react-native-ble-plx": "2.0.3",     "@config-plugins/react-native-ble-plx": "^7.0.0",

### Device

 android pixle 7a latest os

### Environment info

```shell
Bryan@bj-macbook-pro truMoto % react-native info
info Fetching system and libraries information...
(node:81111) Warning: Accessing non-existent property 'padLevels' of module exports inside circular dependency
(Use `node --trace-warnings ...` to show where the warning was created)
System:
  OS: macOS 14.2.1
  CPU: (10) x64 Apple M1 Max
  Memory: 74.75 MB / 32.00 GB
  Shell:
    version: "5.9"
    path: /bin/zsh
Binaries:
  Node:
    version: 16.20.1
    path: ~/.nvm/versions/node/v16.20.1/bin/node
  Yarn:
    version: 1.22.19
    path: ~/.yarn/bin/yarn
  npm:
    version: 8.19.4
    path: ~/.nvm/versions/node/v16.20.1/bin/npm
  Watchman:
    version: 2023.12.04.00
    path: /opt/homebrew/bin/watchman
Managers:
  CocoaPods:
    version: 1.12.1
    path: /Users/Bryan/.rvm/gems/ruby-2.7.6/bin/pod
SDKs:
  iOS SDK:
    Platforms:
      - DriverKit 23.0
      - iOS 17.0
      - macOS 14.0
      - tvOS 17.0
      - watchOS 10.0
  Android SDK:
    API Levels:
      - "30"
      - "31"
      - "32"
      - "33"
      - "33"
      - "34"
    Build Tools:
      - 27.0.3
      - 28.0.3
      - 29.0.3
      - 30.0.2
      - 30.0.3
      - 31.0.0
      - 32.0.0
      - 33.0.0
      - 33.0.2
      - 34.0.0
    System Images:
      - android-33 | Google APIs ARM 64 v8a
      - android-34 | Google APIs ARM 64 v8a
      - android-UpsideDownCake-ext5 | Google APIs ARM 64 v8a
      - android-UpsideDownCake-ext5 | Google Play ARM 64 v8a
    Android NDK: Not Found
IDEs:
  Android Studio: 2022.3 AI-223.8836.35.2231.10811636
  Xcode:
    version: 15.0.1/15A507
    path: /usr/bin/xcodebuild
Languages:
  Java:
    version: 17.0.6
    path: /usr/bin/javac
  Ruby:
    version: 2.7.6
    path: /Users/Bryan/.rvm/rubies/ruby-2.7.6/bin/ruby
npmPackages:
  "@react-native-community/cli": Not Found
  react:
    installed: 18.2.0
    wanted: 18.2.0
  react-native:
    installed: 0.72.6
    wanted: 0.72.6
  react-native-macos: Not Found
npmGlobalPackages:
  "*react-native*": Not Found
Android:
  hermesEnabled: Not found
  newArchEnabled: Not found
iOS:
  hermesEnabled: Not found
  newArchEnabled: Not found

info React Native v0.73.2 is now available (your project is running on v0.72.6).
info Changelog: https://github.com/facebook/react-native/releases/tag/v0.73.2
info Diff: https://react-native-community.github.io/upgrade-helper/?from=0.72.6
info For more info, check out "https://reactnative.dev/docs/upgrading?os=macos&platform=android".
Bryan@bj-macbook-pro truMoto %
```

### Steps to reproduce

1. open app on android
2. connect to bt device
3. close app
4. reopen app
5. scan for device
note how the device still says connected(most of the time it actually still is but other times its not)

### Formatted code sample or link to a repository

```js
just doing a feeler to see if its just me or more people with the same issue
```

### Relevant log output

```shell
I also get this error when the app doesn't fully close after swiping up on the app to kill it

 WARN  Possible Unhandled Promise Rejection (id: 0):
Error: Unable to deactivate keep awake. However, it probably is deactivated already.
Error: Unable to deactivate keep awake. However, it probably is deactivated already.
    at construct (native)
    at apply (native)
    at _construct
```

### Additional information

I am curious if this is happening to anyone else or just me. 

After killing the app, it should restart to the expo menu where you select the server but it seems to go right to the app where it is still connected to the bt device.  Sometimes, it thinks its connected to the bt device but its not then I can no longer connect , disconnect or start scanning for new devices after I close  and reopen the app.  This happens on prod too not just my local dev environment.

bj97301, 2024-01-29:
@dominik-czupryna-withintent any chance you have experienced this?  Does this project take donations?  I would love to get my android users back up and running asap.

eliaslecomte, 2024-02-14:
This is typically not caused by react-native-ble-plx but due to the bluetooth profile of the bluetooth peripheral. For example a headphone you would want to remain connected. Is the profile correct?

bj97301, 2024-02-14:
> This is typically not caused by react-native-ble-plx but due to the bluetooth profile of the bluetooth peripheral. For example a headphone you would want to remain connected. Is the profile correct?

I'll take a look but it was working right for a while. Not sure what could have changed recently. Was wondering if anyone else was experiencing this as well.

dominik-czupryna-withintent, 2024-02-16:
I will take a look at it, but in general is that through the BLE-PLX instance you control the phone's BT module, and if you do not end the connection, the phone's BT module will wait for further operations on the device. I'm not sure if won't be better to leave it to be implemented via lifecycle (https://developer.android.com/guide/components/activities/activity-lifecycle) by the developer because he knows best what should happen with his device.

bj97301, 2024-02-16:
> I will take a look at it, but in general is that through the BLE-PLX instance you control the phone's BT module, and if you do not end the connection, the phone's BT module will wait for further [...]

Thank you I really appreciate that. Also, I tried this on delay version recently and I get the same results. It might be more intuitive to have the default behavior be the same across devices.  iOS seems to disconnect every time.

intent-kacper-cyranowski, 2024-07-24:
Hi @bj97301 
This issue should be solved in 3.2.1. We have added support for invalidating native module on android, which also handles disconnecting from devices in case of closing the app. 

For anyone struggling with a problem that looks similar, please open separate issue.

closed 2024-07-24 by intent-kacper-cyranowski