dotintent/react-native-ble-plx #1172: Scan for devices with phone screen off (Android)

hnrgtk, 2024-02-18:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

My app scan for devices and gets advertisings from previously connected devices. I installed react-native-background-actions lib and it works when the app is not in the foreground but stops scanning when the screen is off.
Is there any way to make the scan keep working?

EmmaZachara, 2024-02-21:
Hi! 
Thanks for the question. We will investigate and get back to you soon.

Microwave-WYB, 2024-03-15:
I can confirm this problem. I'm using [expo foreground actions](https://socket.dev/npm/package/expo-foreground-actions/overview/0.4.5) but I still can't get the scanning working when the screen is off.

EmmaZachara, 2024-04-19:
Hi @Microwave-WYB, @hnrgtk. 
We are still working on this and some other issues, so please hold on a bit longer.

Beckmann0o, 2024-10-28:
Hello, I have the same problem. Is there a way to ensure that the library can perform ble scans even if the screen is off?

aliberski, 2025-01-09:
Hello.

Scanning for BLE when the screen is off on Android devices can be challenging due to [system limitations and power management policies](https://stackoverflow.com/questions/48077690/ble-scan-is-not-working-when-screen-is-off-on-android-8-1-0). To enhance the likelihood of successful background scanning, consider the following steps:

1. According to the documentation of the native library that we use for bluetooth connection on android you should add 
```
<uses-permission android:name="android.permission.ACCESS_BACKGROUND_LOCATION" android:maxSdkVersion="30" />
```
in AndroidManifext.xml to be able to scan in the background. [[LINK](https://github.com/dariuszseweryn/RxAndroidBle?tab=readme-ov-file#if-you-want-to-scan-in-the-background-and-support-apis-29--30)]

2. `react-native-background-actions` like @hnrgtk mentioned
3. Some device manufacturers implement aggessive power-saving measures that may affect background operations. It may need to be adjusted in the device settings.

Please let me know if that helped

closed 2025-01-09 by aliberski