dotintent/react-native-ble-plx #1173: [IOS]  ERROR  Invariant Violation: `new NativeEventEmitter()` requires a non-null argument.

george-eucare, 2024-02-22:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Library version

3.1.2

### Platform

IOS

### Setup Description

Get `ERROR Invariant Violation:  new NativeEventEmitter()  requires a non-null argument.` 
when IOS (Simulator and TestFlight App) executed  `new BleManager()` . 
Android is working good.

### Steps to Reproduce

``` jsx
import { BleError, BleManager, Characteristic, Device } from 'react-native-ble-plx';

function useBLE(): BluetoothLowEnergyApi {
  const bleManager = useMemo(() => new BleManager(), []);
  
  return {
    ...
  };
}

export default useBLE;

```
Get `ERROR  Invariant Violation: `new NativeEventEmitter()` requires a non-null argument.`
when executed  `new BleManager()`

### Relevant log output

```shell
`ERROR  Invariant Violation: `new NativeEventEmitter()` requires a non-null argument.`
```

### Contents of the `package.json` file

info.plist:

<key>NSBluetoothPeripheralUsageDescription</key>
<key>NSLocationAlwaysUsageDescription</key>

### Additional Information

Android is working good.

already try `pod install` and `npm install`

EmmaZachara, 2024-02-22:
Hi @george-eucare, 
Thanks for reaching out. We will verify the issue and get back to you soon.

dominik-czupryna-withintent, 2024-03-10:
I tried to reproduce this problem but I am unable to get this error. I have generated an application and implemented basic device scanning and I do not see this error in any build. Would you be able to enable this project (https://github.com/dominik-czupryna-withintent/ble-plx-issues-1173) and confirm what exact steps I need to take to get the error?

BTW in 
```js
import { BleError, BleManager, Characteristic, Device } from 'react-native-ble-plx';

const useBLE = () => {
  const bleManager =  new BleManager();
  
  return {
    ...
  };
}

export default useBLE;
```
you shouldn't declare new BleManager() in lifecycle. You should have single instance of BleManager.

george-eucare, 2024-03-18:
The error has not occurred again; 
it seems like the xcode cache wasn't cleared.  (product -> clean build Folder)
Thank you for your answer!

closed 2024-03-18 by george-eucare

LXCGH, 2025-02-26:
cd /Users/edy/luxiaochen/study/toystoreApp/ios
rm -rf Pods
rm -rf build
pod install