dotintent/react-native-ble-plx #1175: How to activate sensor and monitoring data like blood pressure, heart rate on watch device if we know only service uuid and characteristic uuid

silldepair, 2024-02-25:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

i am new to bluetooth development, i want to monitor watch device that can record user blood pressure and heart rate. i already connect and discover service and characteristic uuid, but when i call monitor there nothing happen, is there any step i missing ? 
here is the service and character uuid
[image]

and here is my code :
device.monitorCharacteristicForService(serviceUuid, charUuid, (error, characteristic) => {
                  __DEV__ && console.log('monitor characteristic eror:', error);
                 if(error){
                    __DEV__ && console.log('errorc',JSON.stringify(error));
                  }else{
                    console.log('===',characteristic?.value);
                  }

                })

dominik-czupryna-withintent, 2024-02-25:
Have you checked whether it was described in the specification for a profile eg. https://www.bluetooth.com/specifications/specs/blood-pressure-profile-1-1-1/ ?

silldepair, 2024-02-26:
sorry i dont understand, can you give more explanation or sample code of how i should do it. i dont understand bluetooth development from left to right

dominik-czupryna-withintent, 2024-03-10:
Currently we have no plans to prepare implementation examples for specific BLE services. We are currently focused on providing ability to manage the BLE connection from the RN. The package has a general implementation example which can be found in repo - https://github.com/dotintent/react-native-ble-plx/tree/master/example.
More detailed documents for BLE profiles/services specifications can be found at bluetooth website - https://www.bluetooth.com/specifications/specs/. To get to know the device you are working with, I recommend using the nRF connect application - https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en&gl=US / https://apps.apple.com/pl/app/nrf-connect-for-mobile/id1054362403?l=en.

closed 2024-03-10 by dominik-czupryna-withintent