dotintent/react-native-ble-plx #1176: Expo 50 compatibility workaround

MichaelRinger, 2024-02-27:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

Since react-native-ble-plx is currently only supported up to expo 49 does anyone know how to get it working with Expo 50?

Thanks for any answers in advance.

EmmaZachara, 2024-02-27:
Hi @MichaelRinger, 
Thanks for reaching out! Soon, our developer will get back to you with more information and answers.

dominik-czupryna-withintent, 2024-03-01:
I have tried the expo 50 app on several devices and see no errors. Could you describe what exactly is the problem? Here is the code I used: https://github.com/dominik-czupryna-withintent/ble-plx-issues-1176

MichaelRinger, 2024-03-01:
Thanks, I was attempting to set it up with the outdated `@config-plugins/react-native-ble-plx` package, which was the only part causing version conflicts. Installing only `react-native-ble-plx` worked.

closed 2024-03-01 by MichaelRinger