dotintent/react-native-ble-plx #1177: > > **For anyone interested with Android side:** Looks like Android doesn't required any specific steps like iOS. You can create bluetooth operation on background with [react-native-background-actions](https://github.com/Rapsssito/react-native-background-actions) package. This packages allows you to create background service on Android **even app killed/ terminated (swipe up)**

lsantosmorecolab, 2024-03-05:
> > **For anyone interested with Android side:** Looks like Android doesn't required any specific steps like iOS. You can create bluetooth operation on background with [react-native-background-actions](https://github.com/Rapsssito/react-native-background-actions) package. This packages allows you to create background service on Android **even app killed/ terminated (swipe up)**
> > But like iOS; you **must** provide a UUID for scanning function of manager. This means you can't scan everything on background mode.
> > Basically:
> > ```js
> > const task = async () => {
> >     await new Promise(async () => {
> >         manager.startDeviceScan(
> >             [UUID],  // UUID of your devices 
> >             {scanMode: ScanMode.LowPower}, // I haven't checked yet but LowLatency mode probably
> >                                            // drains so much battery, so maybe OS kill our background service. 
> >                                            // So better not to be so aggressive
> >              listener  // Do whatever you want when the device has found
> >          )
> >     });
> > };
> > 
> > await BackgroundService.start(task, options);
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Speaking of it; When you working on background with heavy tasks (_such as like our topic, ble scans_) your user should give some permission (_I'm not talking about AndroidManifest "uses-permission" thing_) to your application on device Settings menu. You can look them on [dontkillmyapp](https://dontkillmyapp.com/)
> > You can check your task does scanning, made connections and did some operation with `adb shell dumpsys bluetooth_manager` command. If you saw `Ongoing Scans` label under your package name; then your background ble operations are working!
> > **PLUS:** You can check if your task running or not with `adb shell dumpsys activity services <your-package-name>`
> > Hope it helps!
> 
> Thanks brother, it works!

Why this doesnt work for me?
I have this on the same way you do - all permitions, AndroidManifest and Gradle changes to support that, but i really cant run the BleManager in Background mode. The scanner is mounted, but never running...?

When in foreground he can find  devices, but in background he only "starts" the scanner ("entrei", but never really starts the scanner so I can find a new devices...

Really apreciate any help!

_Originally posted by @lsantosmorecolab in https://github.com/dotintent/react-native-ble-plx/issues/529#issuecomment-1976420222_

dominik-czupryna-withintent, 2024-03-14:
We are aware of issues with background modes on Android and this has been added to our road map of issues that we will address. Given our categorization of problems, this will be addressed next month. Unfortunately, we are unable to help with this problem right now.

lsantosmorecolab, 2024-03-15:
@dominik-czupryna-withintent Thank you for your reply!

I believe I've already found a response to the issue.

The BleManager can only scan for BLE devices or normal Bluetooth devices in the background with the help of a library like background-fetch, but it can only do so in the foreground, not in the background (when the app is minimized). This is because the scanner is an active action.

For example, if I start the scanner in the background (using background-fetch), it will run and you can interact with other views and features in your app. However, when you minimize the app (e.g., switch to another app or return to the home screen), the scanner stops. It doesn't crash, but it won't find any devices either.

PS: I'm sharing my research here, but I'm not 100% sure and I'm open to new perspectives. Cheers!

naked-brandon, 2024-04-19:
Is this also an issue On IOS as i am facing the same thing there i cannot seem to find a way to have the ble manager run in background mode after pushing to testflight

lsantosmorecolab, 2024-04-22:
> Is this also an issue On IOS as i am facing the same thing there i cannot seem to find a way to have the ble manager run in background mode after pushing to testflight

Hello! I hope you are doing well.

Based on what I have studied and learned, **Bluetooth is considered an 'active and blocking feature,'** which means that by default, **we cannot run the Bluetooth scanner in the background** due to its high battery consumption and other operating system requirements.

In my project, I have implemented the scanner to run in the foreground.

**_

> But if someone was doing it, please reply to me. I really need to run it in the background, or improve my knownlegth 👍 

_**

JatinBhuva, 2024-06-20:
> @dominik-czupryna-withintent Thank you for your reply!
>
> I believe I've already found a response to the issue.
>
> The BleManager can only scan for BLE devices or normal Bluetooth devices in the [...]

Suppose I'm already connected with BLE device and then I minimise app and go to home or other app, in this can will I continue to get data from Bluetooth device or connection will terminated?

UmarXcorebit, 2025-01-15:
@JatinBhuva Have you got any solution around this?

UmarXcorebit, 2025-01-15:
@smitha-2020 We are experiencing an issue when running our app on two different mobile devices, where each device is tied to its own hardware subscription. When we cancel the subscription on one mobile device, the subscription is also canceled on all other mobile devices. Could you help clarify why this is happening or suggest a solution?