dotintent/react-native-ble-plx #1178: Read data from a scale.

jrubiano01, 2024-03-11:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

Good day, I'm using the sample App provided from the repository to validate its operation when establishing a connection with a Scale, both are connecting, but when I get on the Scale to get my weight and it shows the results, I can't receive them or evidence them on the console, why does this happen? Or is it that some devices don't allow that data to be shared via Bluetooth?.

Here is a link to the scale that was purchased.

https://www.transtekcorp.com/products/gbf-2008-series-smart-digital-body-fat-scale-transtek/

EmmaZachara, 2024-03-14:
Hi @jrubiano01, 
Thank you for reaching out. We will review your question and get back to you soon.

dominik-czupryna-withintent, 2024-03-14:
Sample app does not have implemented support for weight scale services. Have you implemented support for the weight service you are interested in?

closed 2024-05-05 by dominik-czupryna-withintent