dotintent/react-native-ble-plx #1179: Device disconnected after write

kurogalaxy, 2024-03-13:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

After scanning and connecting the device, I call the following

`await plxBleManager
      .writeCharacteristicWithoutResponseForDevice(
        deviceId,
        serviceUUIDs.server,
        serviceUUIDs.write,
        data,
      )
      .then(algo => {
        console.log('Write successful');
      })
      .catch(error => {
        console.log('Error', error);
      });`
      
The first time I run the code I get the 'write successful' text but second time returns a
`[BleError: Device D8:3A:DD:83:AE:DC is not connected]`

I also have the following method but it won't return anything at all

`const onDisconnect = (device: Device) => {
    device.onDisconnected(error => {
      console.log('device was disconnected', error);
    });
  };`
  
Am I missing something? I'd really appreciate any help.
  
Testing on Android 13.

EmmaZachara, 2024-03-14:
Hi @kurogalaxy, 
Thank you for raising this question. We will review it and get back to you soon.

nhaduongc, 2024-03-19:
I got the current issue now. when I sync data from my BLE device, the BLE device send data to the app(Android) the progress take arround 10 - 30 second. But in the latest version 3.1.2 It's fire disconnection event in this case.  I'm not sure it is the same case. look like when the BLE device do some busy task. that make the library fire disconnect event

fustaro, 2024-04-01:
same happening to me after updating from 2.0.3 to 3.1.2 on Android, module fires disconnected event a few seconds after connecting.

fustaro, 2024-04-01:
for me it basically disconnects after the timeout period, despite it actually connecting successfully and I can send and receive data fine before it happens. E.g. below will connect fine for 30 seconds then disconnect for no reason.

```
const connectedDevice = await this.manager.connectToDevice(device.id, {
      timeout: 30000
});
```

ex3ndr, 2024-04-29:
I have the same problem, in my case iOS works fine, but Android disconnects.

EmmaZachara, 2024-04-29:
@fustaro, @ex3ndr, @nhaduongc 
Thanks for your patience; we'll need a bit more time to get back to you with a solution.

jklimke, 2024-06-03:
Is there any progress here ? I am also experiencing a disconnect after writing and leading several characteristics (so somehow a heavy read). Unfortunately i have to write a little piece of data after each connect.

gamilrr, 2024-06-22:
Any update on this issue?. I could make it work putting a small delay (around 50ms) before sending every chunk of data, it happens to me with the writeWithResponse call.

mikasius, 2024-10-22:
Any update on this issue? The same issue, disconnects after 20-30 seconds on Android.
- Android 12
- React-Native 0.74.5

niffler92, 2025-08-09:
same problem here