dotintent/react-native-ble-plx #1180: @expo/config-plugins version range

habaieba, 2024-03-14:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

"@expo/config-plugins": "~7.2.0",

### Current Behavior

"@expo/config-plugins": "^7.2.5"

### Library version

3.1.2

### Device

iPhone 14

### Environment info

```shell
Here are my dependencies in my package.json :

"dependencies": {
    "@config-plugins/react-native-ble-plx": "^7.0.0",
    "@react-native-firebase/analytics": "^18.8.0",
    "@react-native-firebase/app": "^18.8.0",
    "@react-native-firebase/auth": "^18.8.0",
    "@react-native-firebase/crashlytics": "^18.8.0",
    "@react-native-firebase/firestore": "^18.8.0",
    "@react-native-firebase/functions": "^19.0.0",
    "@react-native-picker/picker": "2.4.10",
    "@types/react": "~18.2.14",
    "expo": "~49.0.15",
    "expo-asset": "~8.10.1",
    "expo-build-properties": "~0.8.3",
    "expo-camera": "~13.4.4",
    "expo-checkbox": "~2.4.0",
    "expo-constants": "~14.4.2",
    "expo-crypto": "~12.4.1",
    "expo-dev-client": "~2.4.13",
    "expo-linking": "~5.0.2",
    "expo-router": "^2.0.0",
    "expo-splash-screen": "~0.20.5",
    "expo-status-bar": "~1.6.0",
    "expo-updates": "~0.18.19",
    "react": "18.2.0",
    "react-native": "0.72.10",
    "react-native-ble-plx": "^3.1.0",
    "react-native-gesture-handler": "~2.12.0",
    "react-native-safe-area-context": "4.6.3",
    "react-native-screens": "~3.22.0",
    "typescript": "^5.1.3"
},
```

### Steps to reproduce

1. npm ls @expo/config-plugins

### Formatted code sample or link to a repository

```js
Here are my dependencies in my package.json :

"dependencies": {
    "@config-plugins/react-native-ble-plx": "^7.0.0",
    "@react-native-firebase/analytics": "^18.8.0",
    "@react-native-firebase/app": "^18.8.0",
    "@react-native-firebase/auth": "^18.8.0",
    "@react-native-firebase/crashlytics": "^18.8.0",
    "@react-native-firebase/firestore": "^18.8.0",
    "@react-native-firebase/functions": "^19.0.0",
    "@react-native-picker/picker": "2.4.10",
    "@types/react": "~18.2.14",
    "expo": "~49.0.15",
    "expo-asset": "~8.10.1",
    "expo-build-properties": "~0.8.3",
    "expo-camera": "~13.4.4",
    "expo-checkbox": "~2.4.0",
    "expo-constants": "~14.4.2",
    "expo-crypto": "~12.4.1",
    "expo-dev-client": "~2.4.13",
    "expo-linking": "~5.0.2",
    "expo-router": "^2.0.0",
    "expo-splash-screen": "~0.20.5",
    "expo-status-bar": "~1.6.0",
    "expo-updates": "~0.18.19",
    "react": "18.2.0",
    "react-native": "0.72.10",
    "react-native-ble-plx": "^3.1.0",
    "react-native-gesture-handler": "~2.12.0",
    "react-native-safe-area-context": "4.6.3",
    "react-native-screens": "~3.22.0",
    "typescript": "^5.1.3"
},
```

### Relevant log output

```shell
///////////////
├─┬ expo-constants@14.4.2
│ └─┬ @expo/config@8.1.2
│   └── @expo/config-plugins@7.2.5 deduped
├─┬ expo-splash-screen@0.20.5
│ └─┬ @expo/prebuild-config@6.2.6
│   └── @expo/config-plugins@7.2.5 deduped
├─┬ expo-updates@0.18.19
│ └── @expo/config-plugins@7.2.5
├─┬ expo@49.0.23
│ ├─┬ @expo/cli@0.10.17
│ │ └── @expo/config-plugins@7.2.5 deduped
│ └── @expo/config-plugins@7.2.5 deduped
└─┬ react-native-ble-plx@3.1.2
  └── @expo/config-plugins@7.8.4
```

EmmaZachara, 2024-03-14:
Hi @habaieba, 
Thank you for raising the issue. We will verify it and get back to you soon.

EmmaZachara, 2024-04-19:
Hi @habaieba, 
We are still working on this and some other issues, so please hold on a bit longer.

jbrumwell, 2024-04-21:
Just ran into this as well

```
jason@pop-os:~/Documents/lanbri/focused$ npx expo-doctor
✔ Check Expo config for common issues
✔ Check package.json for common issues
✔ Check native tooling versions
✔ Check dependencies for packages that should not be installed directly
✔ Check for common project setup issues
✔ Check for issues with metro config
✔ Check npm/ yarn versions
✔ Check Expo config (app.json/ app.config.js) schema
✔ Check that packages match versions required by installed Expo SDK
✔ Check that native modules do not use incompatible support packages
✔ Check for legacy global CLI installed locally
✖ Check that native modules use compatible support package versions for installed Expo SDK

Detailed check results:

Expected package @expo/config-plugins@~7.2.2
Found invalid:
  @expo/config-plugins@7.9.1
  (for more info, run: npm why @expo/config-plugins)
Advice: Upgrade dependencies that are using the invalid package versions0.

One or more checks failed, indicating possible issues with the project.
jason@pop-os:~/Documents/lanbri/focused$ npm why @expo/config-plugins@7.9.1
@expo/config-plugins@7.9.1
node_modules/react-native-ble-plx/node_modules/@expo/config-plugins
  @expo/config-plugins@"^7.2.5" from react-native-ble-plx@3.1.2
  node_modules/react-native-ble-plx
    react-native-ble-plx@"^3.1.2" from the root project
```

dominik-czupryna-withintent, 2024-04-22:
Hi @jbrumwell and @habaieba 
I tried with the minimum expo package and no error appears, it seems to be a simple version expected by the packages conflict. I will bump config-plugins version in the next release. Repo with the code I used for the test: https://github.com/dominik-czupryna-withintent/ble-plx-issues-1180

intent-kacper-cyranowski, 2024-06-19:
Hi everyone,
Do you still need help with the issue?

closed 2024-07-11 by intent-kacper-cyranowski