dotintent/react-native-ble-plx #1185: Can't write

Yangnnnn, 2024-03-28:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

got this error when I try to write 
{"message":"Characteristic 24eb85c3-1114-46fd-a9a3-1559361c6a95 write failed for device E95A6C97-D69F-D474-1F5C-9B8129185426 and service 24eb85c0-1114-46fd-a9a3-1559361c6a95","errorCode":401,"attErrorCode":14,"iosErrorCode":null,"androidErrorCode":null,"reason":"Unlikely error.","name":"BleError"}

EmmaZachara, 2024-04-19:
Hi @Yangnnnn, thank you for your question, we'll get back to you as soon as we can.

dominik-czupryna-withintent, 2024-04-22:
Hi @Yangnnnn 
I have few questions:

What device are you connecting from?
What device are you connecting to?
How your BLE-PLX implementation looks like?
What steps did you take when the error occurred?
Did the error appear once or does it occurs all the time?
Did you check the error on several devices or on one?
Does the problem occurs on iOS or Android or both?

It's hard for me to recreate the issue without it.

intent-kacper-cyranowski, 2024-06-19:
Hi @Yangnnnn 
Do you still need help with the issue?

closed 2024-06-20 by intent-kacper-cyranowski