dotintent/react-native-ble-plx #1187: Is expo a required dependency to use this library?

EthanYeh, 2024-04-17:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [ ] I'm running the latest version

### Question

Hi, is it possible to use this library on bare react-native (without expo)? When reading the installation [guide](https://github.com/dotintent/react-native-ble-plx?tab=readme-ov-file#configuration--installation) in the README, it seems like we need to install the `expo` package first, and then list this library as a plugin in `expo`. On the [yarn package site](https://classic.yarnpkg.com/en/package/react-native-ble-plx), it also lists `@expo/config-plugins` as a dependency.

A previous related question is [here](https://github.com/dotintent/react-native-ble-plx/issues/388) but it received no answer

EmmaZachara, 2024-04-19:
Hi @EthanYeh, thanks for your question; we'll get back to you soon.

dominik-czupryna-withintent, 2024-04-22:
Hi @EthanYeh 

In case of installation on react native CLI, you just installing the package and it’s works. In the case of installation on expo, additional configuration and prebuilding are necessary. That’s why you can find much more information about the expo in the readme.

EthanYeh, 2024-04-22:
Ah I see, I'm resolved. Thank you!

closed 2024-04-22 by EthanYeh