dotintent/react-native-ble-plx #1189: app crash after reconnect device

omerabdelhady62, 2024-04-18:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [ ] I'm running the latest version

### Question

i am create function connect and setup monitor and disconnect device and reconnect to device 
steeps : - 
*  BLEServicePlx.connectToDevice(mac)
*  BLEServicePlx.discoverAllServicesAndCharacteristicsForDevice(mac)
*  BLEServicePlx.setupMonitor()
* BLEServicePlx.disconnectDeviceById(mac) 
 and loop this steep all the time ,
 The application terminated after several times

EmmaZachara, 2024-04-19:
Hi, thank you for your question; we'll get back to you soon with answers

intent-kacper-cyranowski, 2024-06-14:
Hi @omerabdelhady62 
Sorry for late response, is your issue is still relevant?

intent-kacper-cyranowski, 2024-06-20:
Hi @omerabdelhady62 
Do you still need help with the issue?

intent-kacper-cyranowski, 2024-07-01:
Closing due to inactivity

closed 2024-07-01 by intent-kacper-cyranowski

omerabdelhady62, 2024-07-02:
@intent-kacper-cyranowski 
hi ،  sorry for delay for response , this problem because using local storage wrong when disconnect the device function .