dotintent/react-native-ble-plx #1190: Is it compatible with Expo 50?

Santa0727, 2024-04-21:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

Hello,

I am using expo 50.
```
"dependencies": {
    "expo": "~50.0.14",
...

```

When I tried to run `npx expo install @config-plugins/react-native-ble-plx`, I got the following error:
```
› Installing 1 other package using npm
> npm install --save @config-plugins/react-native-ble-plx
npm ERR! code ERESOLVE
npm ERR! ERESOLVE unable to resolve dependency tree
npm ERR!
npm ERR! While resolving: expo-ble@1.0.0
npm ERR! Found: expo@50.0.17
npm ERR! node_modules/expo
npm ERR!   expo@"~50.0.14" from the root project
npm ERR!
npm ERR! Could not resolve dependency:
npm ERR! peer expo@"^49" from @config-plugins/react-native-ble-plx@7.0.0       
npm ERR! node_modules/@config-plugins/react-native-ble-plx
npm ERR!   @config-plugins/react-native-ble-plx@"*" from the root project      
npm ERR!
npm ERR! Fix the upstream dependency conflict, or retry
npm ERR! this command with --force or --legacy-peer-deps
npm ERR! to accept an incorrect (and potentially broken) dependency resolution.
...
```

I can see - it is compatible with Expo 49 in the readme, but I can't install the package with Expo 50. Isn't it compatible with 50?

Thank you,

dominik-czupryna-withintent, 2024-04-21:
Hi @Santa0727
We checked compatibility with expo 50 and found no problems. In issue #1176 (https://github.com/dotintent/react-native-ble-plx/issues/1176) is an example using expo 50.

flexbox, 2024-05-01:
This bugs me 😅 
> You shouldn't use additional @config-plugins.
@dominik-czupryna-withintent Do you think it can somehow added anyways?

I was able to successfully make a build 10 days ago with `"expo": "~50.0.4",`

When I want to build my app (`"expo": "~50.0.17",`) with `eas`, I encounter this error https://github.com/expo/expo-cli/issues/4391
```
Error: [ios.dangerous]: withIosDangerousBaseMod: Unrecognised filter type - 127
```
[image]

`npx expo prebuild` runs fine on my local machine.

Here are my investigations with `yarn why @expo/config-plugins` 

```
yarn why @expo/config-plugins
yarn why v1.22.21
[1/4] 🤔  Why do we have the module "@expo/config-plugins"...?
[2/4] 🚚  Initialising dependency graph...
[3/4] 🔍  Finding dependency...
[4/4] 🚡  Calculating file sizes...
=> Found "@expo/config-plugins@7.9.1"
info Has been hoisted to "@expo/config-plugins"
info Reasons this module exists
   - Hoisted from "expo#@expo#config-plugins"
   - Hoisted from "react-native-ble-plx#@expo#config-plugins"        <----- for some reason it's there
   - Hoisted from "expo#@expo#cli#@expo#config-plugins"
   - Hoisted from "expo#@expo#config#@expo#config-plugins"
   - Hoisted from "expo-splash-screen#@expo#prebuild-config#@expo#config-plugins"
info Disk size without dependencies: "1.82MB"
info Disk size with unique dependencies: "3.91MB"
info Disk size with transitive dependencies: "8.42MB"
info Number of shared dependencies: 60
=> Found "expo-updates#@expo/config-plugins@7.8.4"
info This module exists because "expo-updates" depends on it.
info Disk size without dependencies: "1.73MB"
info Disk size with unique dependencies: "3.82MB"
info Disk size with transitive dependencies: "8.32MB"
info Number of shared dependencies: 60
✨  Done in 0.24s.
```

On my `app.config.ts` I added this

```typescript
plugins: [
    "expo-router",
    [
      "react-native-ble-plx",
      {
        isBackgroundEnabled: true,
        modes: ["peripheral", "central"],
        bluetoothAlwaysPermission:
          "Allow $(PRODUCT_NAME) to detect your bluetooth device"
      }
    ],  
```

@Santa0727 
Can you please let us know if `eas build --platform=ios` is working on your side?

Santa0727, 2024-05-01:
Hi @flexbox,
Thank you for your response.
The build command works on my side, and it builds ipa file.

I've used expo managed workflow on my local. Is it impossible to use without ejecting expo?

dominik-czupryna-withintent, 2024-05-05:
@flexbox 
I mean you shouldn't install https://www.npmjs.com/package/@config-plugins/react-native-ble-plx because react-native-ble-plx is shipped with an Expo config plugin on its own.

@Santa0727
You don't need to eject expo, you just need to prebuild the native code because without using native code this package won't work.

kennyyen, 2024-05-23:
Hi, I am not sure now, where should we add these configurations ?
{
        isBackgroundEnabled: true,
        modes: ["peripheral", "central"],
        bluetoothAlwaysPermission:
          "Allow $(PRODUCT_NAME) to detect your bluetooth device"
}

flexbox, 2024-05-23:
@kennyyen 
in your `app.json` or `app.config.js` https://docs.expo.dev/config-plugins/introduction/

```json
{
  "expo": {
    "plugins": 
      [
        "react-native-ble-plx", // https://github.com/dotintent/react-native-ble-plx?tab=readme-ov-file#api
        {
          isBackgroundEnabled: true,
          modes: ["peripheral", "central"],
          bluetoothAlwaysPermission: "Allow $(PRODUCT_NAME) to detect your controller with bluetooth"
        }
      ],
  }
}

```

kennyyen, 2024-05-23:
Thank you for the response, I did tried to put them in my app.json, and when I do
`npx expo install react-native-ble-plx`
I get this 
```
Skipping config plugin check: Plugin is an unexpected type: object
```

kennyyen, 2024-05-23:
Thank you for the clarification ! It works just fine as before !

flexbox, 2024-06-12:
@kennyyen great news!
If your problem is solved can you please close the issue? 
It sparks joy for maintainers ✨

closed 2024-06-12 by Santa0727