dotintent/react-native-ble-plx #1193: BLEmanger is undefine

blood-romantic, 2024-04-30:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

The BleManager should not be **undefine** when importing

`import {BleManager, BleError} from 'react-native-ble-plx';`

### Current Behavior

The BleManager is **undefine** when importing

`import {BleManager, BleError} from 'react-native-ble-plx';`

### Library version

3.1.2

### Device

Pixel 7

### Environment info

```shell
info Fetching system and libraries information...
(node:77921) Warning: Accessing non-existent property 'padLevels' of module exports inside circular dependency
(Use `node --trace-warnings ...` to show where the warning was created)
(node:77921) [DEP0040] DeprecationWarning: The `punycode` module is deprecated. Please use a userland alternative instead.
System:
  OS: macOS 13.1
  CPU: (10) arm64 Apple M1 Pro
  Memory: 2.14 GB / 32.00 GB
  Shell:
    version: 3.2.57
    path: /bin/bash
Binaries:
  Node:
    version: 21.5.0
    path: /opt/homebrew/bin/node
  Yarn:
    version: 1.22.17
    path: /usr/local/bin/yarn
  npm:
    version: 10.2.4
    path: /opt/homebrew/bin/npm
  Watchman:
    version: 2023.12.04.00
    path: /opt/homebrew/bin/watchman
Managers:
  CocoaPods: Not Found
SDKs:
  iOS SDK:
    Platforms:
      - DriverKit 22.2
      - iOS 16.2
      - macOS 13.1
      - tvOS 16.1
      - watchOS 9.1
  Android SDK:
    API Levels:
      - "28"
      - "30"
      - "31"
      - "33"
      - "34"
    Build Tools:
      - 30.0.2
      - 30.0.3
      - 31.0.0
      - 33.0.0
      - 33.0.1
      - 34.0.0
    System Images:
      - android-29 | Google APIs ARM 64 v8a
      - android-33 | Google APIs ARM 64 v8a
      - android-33 | Google APIs Intel x86_64 Atom
      - android-33 | Google Play ARM 64 v8a
    Android NDK: Not Found
IDEs:
  Android Studio: 2022.3 AI-223.8836.35.2231.10406996
  Xcode:
    version: 14.2/14C18
    path: /usr/bin/xcodebuild
Languages:
  Java:
    version: 17.0.9
    path: /usr/bin/javac
  Ruby:
    version: 2.6.10
    path: /usr/bin/ruby
npmPackages:
  "@react-native-community/cli": Not Found
  react:
    installed: 18.2.0
    wanted: 18.2.0
  react-native:
    installed: 0.73.3
    wanted: 0.73.3
  react-native-macos: Not Found
npmGlobalPackages:
  "*react-native*": Not Found
Android:
  hermesEnabled: Not found
  newArchEnabled: Not found
iOS:
  hermesEnabled: Not found
  newArchEnabled: false
```

### Steps to reproduce

```
import {BleManager, BleError} from 'react-native-ble-plx';

export let ble = new BleManager();
```

The BleManager is **undefined** here

### Formatted code sample or link to a repository

```js
Not Yet
```

### Relevant log output

```shell
Not Yet
```

### Additional information

React Native: 0.73.3
JDK: 17
Nodejs: 21.5.0
Android SDK: 34

AndroidManifest.xml

```
<manifest xmlns:android="http://schemas.android.com/apk/res/android">

    <uses-permission android:name="android.permission.INTERNET" />

       <!-- Android >= 12 -->
   <uses-permission android:name="android.permission.BLUETOOTH_SCAN" android:usesPermissionFlags="neverForLocation"/>
   <uses-permission android:name="android.permission.BLUETOOTH_CONNECT" />
   <!-- Android < 12 -->
   <uses-permission android:name="android.permission.BLUETOOTH" android:maxSdkVersion="30" />
   <uses-permission android:name="android.permission.BLUETOOTH_ADMIN" android:maxSdkVersion="30" />

   <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" android:maxSdkVersion="30" />

   <!-- Add this line if your application always requires BLE. More info can be found on:
       https://developer.android.com/guide/topics/connectivity/bluetooth-le.html#permissions
     -->
   <uses-feature android:name="android.hardware.bluetooth" android:required="true"/>
   <uses-feature android:name="android.hardware.bluetooth_le" android:required="true"/>

    <application
      android:name=".MainApplication"
      android:label="@string/app_name"
      android:icon="@mipmap/ic_launcher"
      android:roundIcon="@mipmap/ic_launcher_round"
      android:allowBackup="false"
      android:theme="@style/AppTheme">
      <activity
        android:name=".MainActivity"
        android:label="@string/app_name"
        android:configChanges="keyboard|keyboardHidden|orientation|screenLayout|screenSize|smallestScreenSize|uiMode"
        android:launchMode="singleTask"
        android:windowSoftInputMode="adjustResize"
        android:exported="true">
        <intent-filter>
            <action android:name="android.intent.action.MAIN" />
            <category android:name="android.intent.category.LAUNCHER" />
        </intent-filter>
      </activity>
    </application>
</manifest>
```

build.gradle

```
buildscript {
    ext {
        buildToolsVersion = "34.0.0"
        minSdkVersion = 24
        compileSdkVersion = 34
        targetSdkVersion = 34
        ndkVersion = "25.1.8937393"
        kotlinVersion = "1.8.0"
    }
    repositories {
        google()
        mavenCentral()
    }
    dependencies {
        classpath("com.android.tools.build:gradle")
        classpath("com.facebook.react:react-native-gradle-plugin")
        classpath("org.jetbrains.kotlin:kotlin-gradle-plugin")
    }
}

allprojects {
    repositories {
      maven { url 'https://www.jitpack.io' }
    }
}

apply plugin: "com.facebook.react.rootproject"

```

EmmaZachara, 2024-04-30:
Hi @blood-romantic,
Thanks for reaching out. We'll check the issue and get back to you with a solution proposal soon.

flexbox, 2024-05-01:
@blood-romantic 
Can you double-check what happens when you add some options? 

```typescript
const options: BleManagerOptions = {
  restoreStateIdentifier: "BleInTheBackground",
  restoreStateFunction: (restoredState) => {
    if (restoredState == null) {
      console.log("BleManager was constructed for the first time.");
    } else {
      console.log(
        "BleManager BleManager was restored.",
        restoredState.connectedPeripherals
      );
    }
  }
};

export let ble = new BleManager(options);
```

cixio, 2024-05-03:
> ```shell
> npmPackages:
>   "@react-native-community/cli": Not Found
>   react:
>     installed: 18.2.0
>     wanted: 18.2.0
>   react-native:
>     installed: 0.73.3
>     wanted: 0.73.3
>   react-native-macos: Not Found
> ```

Seems you need to install the module first:
`npm i react-native-ble-plex`

blood-romantic, 2024-05-04:
@cixio the module has been install, i'm not sure why show Not Found

> > ```shell
> > npmPackages:
> >   "@react-native-community/cli": Not Found
> >   react:
> >     installed: 18.2.0
> >     wanted: 18.2.0
> >   react-native:
> >     installed: 0.73.3
> >     wanted: 0.73.3
> >   react-native-macos: Not Found
> > ```
> 
> Seems you need to install the module first: `npm i react-native-ble-plx`

cixio, 2024-05-04:
can you verify that the folder modules/react-native-ble-plx exists?

dominik-czupryna-withintent, 2024-05-05:
@blood-romantic 
Could you create a new project and install only react-native-ble-plx and check if problems still exist or even better show this repository? Before I start investigating this problem, I would like to exclude any potential environmental issues.

blood-romantic, 2024-05-06:
Close this issue, after reinstall the dependency, it is ok

closed 2024-05-06 by blood-romantic