dotintent/react-native-ble-plx #1194: 🐛 [Android] Can't monitor a characteristic

Vishalvishwakarma11, 2024-04-30:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed

### Expected Behavior

I expect, as the documentation says, that if the characteristic is notifiable, I can monitor it to be informed whenever the value of the characteristic has changed. 

### Current Behavior

My characteristics can't be monitored. I'm having this problem on Android.
When I try to monitor the characteristic, I directly receive the following BleError :
[BleError: Characteristic 6f68b37a-58ab-4cbf-b85c-cbdb183ea613 notify change failed for device ? and service 079fe9d4-92a0-4081-a633-7dd74736f131]

### Library version

3.1.2

### Device

device:Motorola Edge 40, Android version 13

### Environment info

```shell
System:
    OS: macOS 13.6.1
    CPU: (8) arm64 Apple M1
    Memory: 48.08 MB / 16.00 GB
    Shell: 5.9 - /bin/zsh
  Binaries:
    Node: 16.16.0 - ~/.nvm/versions/node/v16.16.0/bin/node
    Yarn: 1.22.19 - ~/.nvm/versions/node/v16.16.0/bin/yarn
    npm: 8.11.0 - ~/.nvm/versions/node/v16.16.0/bin/npm
    Watchman: 2023.11.13.00 - /opt/homebrew/bin/watchman
  Managers:
    CocoaPods: 1.15.2 - /usr/local/bin/pod
  SDKs:
    iOS SDK:
      Platforms: DriverKit 22.4, iOS 16.4, macOS 13.3, tvOS 16.4, watchOS 9.4
    Android SDK: Not Found
  IDEs:
    Android Studio: 2022.2 AI-222.4459.24.2221.10121639
    Xcode: 14.3.1/14E300c - /usr/bin/xcodebuild
  Languages:
    Java: 21.0.1 - /usr/bin/javac
  npmPackages:
    @react-native-community/cli: Not Found
    react: 18.2.0 => 18.2.0 
    react-native: 0.71.11 => 0.71.11 
    react-native-macos: Not Found
  npmGlobalPackages:
    *react-native*: Not Found
```

### Steps to reproduce

1. … first i connect to my app with Bluetooth device. here the code, 
  const connectToDevice = async device => {
    try {
      stopScan();
      const connectedDevice = await bleManager.connectToDevice(device.id);
      const deviceIdString = JSON.stringify(connectedDevice.id);
      await AsyncStorage.setItem(
        'deviceForString',
        JSON.stringify(connectedDevice),
      );
      await setLogger(`Device id ${connectedDevice.id} is connected`);
    } catch (error) {
      setError('Error connecting to the device');
    }
  };
2. After that want to subscribe Bluetooth device for notification so, I implemented this, 
const device = deviceForNotify;
    // console.log('checking 3', device);
    try {
      await device.discoverAllServicesAndCharacteristics();
      console.log('Discovering services and characteristics...');
      const services = await device.services();
      const foundService = services.find(
        service => service.uuid === SERVICE_UUID,
      );

      const foundCharacteristics = await foundService.characteristics();

      const finalCharacteristic = foundCharacteristics?.find(
        char => char.uuid === CHARACTERISTIC_UUID,
      );
      if (!foundService) {
        throw new Error('Characteristic not found.');
      }

      console.log('Enabling notifications for characteristic...');

      bleManager.monitorCharacteristicForDevice(
        '63:0C:5B:20:C0:42', // even I provide hardcoded value 
        finalCharacteristic.serviceUUID,
        '6f68b37b-58ab-4cbf-b85c-cbdb183ea613', // even I provide hardcoded value 
        (errorNew, characteristic) => {
          if (errorNew) {
            console.error('Error monitoring characteristic 442:', errorNew);
          } else {
            console.log('Received characteristic value:', characteristic);
          }
        },
      );

      finalCharacteristic?.monitor((error, characteristic) => {
        if (error) {
          console.error('Error monitoring characteristic 455:', error);
        } else {
          console.log('Received characteristic value:', characteristic.value);
        }
      });
      // console.log(
      //   'Notifications enabled for characteristic:',
      //   CHARACTERISTIC_UUID,
      // );
    } catch (error) {
      console.error('Error starting notification:', error);
    }

### Formatted code sample or link to a repository

```js
.
```

### Relevant log output

```shell
It should worked.
```

varun761, 2024-05-30:
Hi. I am facing the same issue and update on this.

intent-kacper-cyranowski, 2024-06-14:
Hi @Vishalvishwakarma11
Sorry for late response, is your issue is still relevant?

kacper-cyra, 2024-06-20:
Hi @Vishalvishwakarma11
Do you still need help with the issue?

intent-kacper-cyranowski, 2024-06-20:
Hi @Vishalvishwakarma11 
Do you still need help with the issue?

intent-kacper-cyranowski, 2024-06-25:
Closing due to inactivity

closed 2024-06-25 by intent-kacper-cyranowski

durgatejam, 2024-12-20:
Is this issue resolved?? Same happening for me..