dotintent/react-native-ble-plx #1195: Connect socket mobile s700 barcode scanner.

krunalshoppiko, 2024-05-02:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

Can we connect socket mobile s700 barcode scanner and scan barcode from this device and get barcode detail?

cixio, 2024-05-03:
There is an SDK from socketmobile you can use with react native - so you don't need to use this module to write your own.

https://docs.socketmobile.com/react-native-capture/en/latest/

krunalshoppiko, 2024-05-03:
> There is an SDK from socketmobile you can use with react native - so you don't need to use this module to write your own.
> 
> https://docs.socketmobile.com/react-native-capture/en/latest/

Ya but i don't want to use this SDK. any other way or example?

cixio, 2024-05-03:
Here are the docs: https://docs.socketmobile.com/main/en/ if it works with another lib / module it will work with this one. But this is not an issue and should be closed.

dominik-czupryna-withintent, 2024-05-05:
Hi @krunalshoppiko

I'm sorry, but we are not able to check every specific device and prepare an example for it (especially when we are talking about such a specific device as this one). If the device supports BLE, you should be able to connect your device using this package, but in the package mentioned by @cixio we can read that bluetooth classic is used to read the barcode.

[image]

If the support on the device side was implemented with Bluetooth classic, you will have to use a different package.

If you encounter more specific bugs in the package during implementation, feel free to report them.

closed 2024-05-05 by dominik-czupryna-withintent

krunalshoppiko, 2024-05-06:
> Hi @krunalshoppiko
> 
> I'm sorry, but we are not able to check every specific device and prepare an example for it (especially when we are talking about such a specific device as this one). If the device supports BLE, you should be able to connect your device using this package, but in the package mentioned by @cixio we can read that bluetooth classic is used to read the barcode.
> 
> <img alt="Screenshot 2024-05-05 at 19 47 28" width="689" src="https://private-user-images.githubusercontent.com/143402866/328037707-80d26293-e374-4ec3-90ae-2c212e7ac963.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ5NjM2NDIsIm5iZiI6MTcxNDk2MzM0MiwicGF0aCI6Ii8xNDM0MDI4NjYvMzI4MDM3NzA3LTgwZDI2MjkzLWUzNzQtNGVjMy05MGFlLTJjMjEyZTdhYzk2My5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwNlQwMjQyMjJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT04OWExMWNiNDJlMTQ4NTlkMmJkOTFiNWI5NTcwYTEyZmFiYWYyMzY5ZTQxNzc3ZTE5NDM3MGU0NTA3MDFhMDIwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.SeREGan4vohYtg1fDJrk1o44R6BnZeki4h7Ov5Jvpvg">
> If the support on the device side was implemented with Bluetooth classic, you will have to use a different package.
> 
> If you encounter more specific bugs in the package during implementation, feel free to report them.

Thanks.. @dominik-czupryna-withintent