dotintent/react-native-ble-plx #1196: can't read Characteristic

blood-romantic, 2024-05-06:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

got this error when I try to read

[BleError: Characteristic f0e60004-4b8c-41d8-a61c-eeafa4eb6077 read failed for device 94:54:93:17:63:73 and service f0e60001-4b8c-41d8-a61c-eeafa4eb6077]

blood-romantic, 2024-05-07:
React native: 0.73.3
react-native-ble-plx: 3.1.2

josmithua, 2024-05-20:
Looks like you're using the same ID for the service and the characteristic

intent-kacper-cyranowski, 2024-06-13:
Hi @blood-romantic 
Could you provide more context? I would need information on how your ble service looks like and how you try to write to it. The more code and context will you provide the easier it will be to find the issue.

flexbox, 2024-06-17:
Hey @blood-romantic 

If your problem is solved can you please close the issue? 
It sparks joy for maintainers ✨

intent-kacper-cyranowski, 2024-07-01:
Hi @blood-romantic 
Do you still need help with the issue?

closed 2024-07-11 by intent-kacper-cyranowski

niharika-141, 2024-12-13:
may i know how the issue is solved

niharika-141, 2024-12-13:
even i getting the dame issue 

  Error reading characteristic: [BleError: Characteristic 233bf001-5a34-1b6d-975c-000d5690abe4 read failed for device 00:09:1F:87:DF:1B and service 233bf000-5a34-1b6d-975c-000d5690abe4]

ERROR  Error reading characteristic: [BleError: Device 00:09:1F:87:DF:1B is not connected]
 ERROR  Connection failed: [TypeError: Cannot read property 'charCodeAt' of undefined]