dotintent/react-native-ble-plx #1197: BLE device automatically disconnecting

chris-goodchild, 2024-05-09:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

Hi, I'm not certain this is a bug but I'm also unable to reproduce this issue using another client, so here goes. 

On iOS, I'm seeing the BLE device disconnect very quickly after pairing. The sequence of events is as follows:

1. Initialize BLE
2. Scan for devices
3. Select the device
4. Connect and stop scan
5. Detect all services and characteristics for the device
6. Write characteristic to obtain a list of available WiFi SSIDs
7. Write characteristic to send WiFi credentials
8. Disconnect the device

Unless I trigger the first write immediately the device disconnects. It seems to disconnect unless there's an operation almost immediately which seems odd. The docs don't suggest there's a need to keep the connection open by any means, so am I missing something? When we test the BLE device using another client performing the same steps it works fine, no disconnects. It's only when I use this library it drops out. I've seen other issues related to Android but nothing for iOS.

madej10, 2024-05-14:
@chris-goodchild thank you for reporting the issue, we're adding it to the pipeline.  We'll let you know if we're seeing the same.

chris-goodchild, 2024-05-15:
Thanks so much @madej10 - just to add some additional findings:

- The `connectToDevice` method succeeds and returns the device state.
- The device is then automatically disconnected and the handler for `onDeviceDisconnected` is invoked with a `null` error argument and, again, the device state as before. I am not manually disconnecting the device anywhere.
- Often, when I check my bluetooth settings I can see that the device is still connected to my iPhone despite the library reporting it as disconnected.
- Attempts to reconnect seem to work but then the device disconnects immediately again.

More often than not, as I am setting up a monitor after discovering services and characteristics, I'm also seeing an error `Characteristic [UUID] not found`. I have double-checked my service and characteristic UUIDs are present and correct, so I assume this error is a result of the device disconnecting?

FlorinPopaPersonal, 2024-05-17:
I am encountering a similar behaviour on android. The ble device disconnects after a few seconds without activity

madej10, 2024-05-17:
Thanks @FlorinPopaPersonal we're preparing a release, after which we'll look into this issue on both platforms.

madej10, 2024-05-23:
Hi all, we've started seeing the issue and are looking for a way to fix it. Thanks for your patience and for bringing it to our attention.

PierreHenri123, 2024-05-27:
Hello. I got the same issue, both Android and Ios. Thanks for your investigating. It will be really appreciated.

joanmarcs, 2024-06-05:
The problem persist in 3.2.0. We only encounter that problem in Android 14. Not in Android 12.

Our app is able to connect to the device. When we are connected, we ask for the state of the device every seconds. It works fine in Android 12 as we can get the state unlimitedly but in Android 14 we get disconnected after few seconds. Also, the connection is not as estable as it is when we make the connection in Android 12, it is in a slow way and with some errors between but it eventually connects.

madej10, 2024-06-10:
hi all, this is our top priority - hopefully we can at least get you an update today. Thank you all for your patience

joanmarcs, 2024-06-10:
We have found some Android 14 devices that are working fine, seems to be a problem about samsungs and pixels.

intent-kacper-cyranowski, 2024-06-11:
Hello everyone, 
I was investigating this issue and I have to admit that I'm having trouble reproducing this error. In the case of iOS, I once managed to reach a situation where an error occurred when connecting to the device, but that was when I started trying to force its occurrence by repeatedly resetting the BLE instance and performing operations on it during the resets 😕.
I also checked it on Android and standard flow also causes no problems.
I tried to reproduce this problem according to the scenario suggested in the first post:
- searched for the device
- connected with the device
- discovered services
then I tried the following scenario 3 times:
- wait 5 or 10 seconds
- save the data for the characteristics
- read the characteristics (checking whether the data is saved correctly)
Related code, which can be checked using the example app, can be found here:
https://github.com/dotintent/react-native-ble-plx/blob/fix/1197_device_automatically_disconnecting/example/src/screens/MainStack/DisconnectBeforeWrite/DisconnectBeforeWrite.tsx

Despite waiting between each step, nothing breaks and everything works as intended. So I have few questions:
Do you have any additional suggestions on what I should do to reproduce this problem?
How (and whether) do you use BLE instance destroy function in your applications?
How long do your apps stay active?
How often are requests sent to the device (how long is the interval between subsequent requests)?
Do we assume that the user turns off (moves to background) the application after connecting?

a396901990, 2024-06-25:
we had a similar issue with Samsung, but we are using v2.0.3

madej10, 2024-07-01:
@a396901990 @amit13091992 could you please reply to the questions posted by @intent-kacper-cyranowski in the previous comment to help us narrow it down:

> Despite waiting between each step, nothing breaks and everything works as intended. So I have few questions:
> Do you have any additional suggestions on what I should do to reproduce this problem?
> How [...]

Lesha1213, 2024-07-02:
Also have this problem on Samsung. May be the reason is https://www.youtube.com/watch?v=KH0tryRDhMw ?

intent-kacper-cyranowski, 2024-07-02:
Hi @joanmarcs 
I tried to reproduce your issue using https://github.com/dotintent/react-native-ble-plx/blob/fix/1197_device_automatically_disconnecting/example/src/screens/MainStack/DisconnectBeforeWrite/DisconnectBeforeWrite.tsx without setting the timeout on android 14 device with no success.

The message you received in [#1157 (comment)](https://github.com/dotintent/react-native-ble-plx/issues/1157#issuecomment-2148997260) means only that connection was successfully closed, the rest of the error is related to unhandled promise.

Because problem seems to be device specific I would need of you to [setLogLevel](https://dotintent.github.io/react-native-ble-plx/#blemanagersetloglevel) to `LogLevel.Verbose`, and provide filtered logs from Logcat, gathered on a device with the issue. The filters should be `package:mine (tag:BluetoothGatt | tag:ReactNativeJS | RxBle)`

I would also ask you to create separate issue co we could continue our conversation there, as this one is related to iOS.

eliaslecomte, 2024-07-04:
For me that is:
```
2024-07-04 09:30:07.340  9402-9797  RxBle#Clie...ationQueue com...urope.control.assistant.debug  D  QUEUED   ScanOperationApi21(56802824)
2024-07-04 09:30:07.341  9402-9815  RxBle#Clie...ationQueue com...urope.control.assistant.debug  D  STARTED  ScanOperationApi21(56802824)
2024-07-04 09:30:07.342  9402-9815  RxBle#Clie...ationQueue com...urope.control.assistant.debug  I  RUNNING  ScanOperationApi21{ANY_MUST_MATCH -> nativeFilters=[BluetoothLeScanFilter [mDeviceName=null, MAC=null, mUuid=..., mUuidMask=null, mSolicitedUuid=null, mSolicitedUuidMask=null, mServiceDataUuid=null, mServiceData=null, mServiceDataMask=null, mManufacturerId=-1, mManufacturerData=null, mManufacturerDataMask=null], BluetoothLeScanFilter [mDeviceName=null, MAC=null, mUuid=..., mUuidMask=null, mSolicitedUuid=null, mSolicitedUuidMask=null, mServiceDataUuid=null, mServiceData=null, mServiceDataMask=null, mManufacturerId=-1, mManufacturerData=null, mManufacturerDataMask=null]]}
2024-07-04 09:30:07.343  9402-14433 RxBle#ScanOperation     com...urope.control.assistant.debug  I  Scan operation is requested to start.
2024-07-04 09:30:07.344  9402-14433 RxBle#Scan...ationApi21 com...urope.control.assistant.debug  D  No library side filtering —> debug logs of scanned devices disabled
2024-07-04 09:30:07.351  9402-9815  RxBle#Clie...ationQueue com...urope.control.assistant.debug  D  FINISHED ScanOperationApi21(56802824) in 11 ms
2024-07-04 09:30:07.396  9402-9402  RxBle#Client            com...urope.control.assistant.debug  I  ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=BRC1H 43:23:99}, rssi=-68, timestampNanos=852829057810892, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...], isConnectable=CONNECTABLE}
2024-07-04 09:30:07.413  9402-9402  RxBle#Client            com...urope.control.assistant.debug  I  ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=BRC1H 43:23:99}, rssi=-68, timestampNanos=852829085807246, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...], isConnectable=CONNECTABLE}
2024-07-04 09:30:07.453  9402-9402  RxBle#Client            com...urope.control.assistant.debug  I  ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=BRC1H 43:23:99}, rssi=-68, timestampNanos=852829124941101, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...], isConnectable=CONNECTABLE}
2024-07-04 09:30:07.499  9402-9402  RxBle#Client            com...urope.control.assistant.debug  I  ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=BRC1H 43:23:99}, rssi=-68, timestampNanos=852829170055684, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...], isConnectable=CONNECTABLE}
2024-07-04 09:30:07.536  9402-9402  RxBle#Client            com...urope.control.assistant.debug  I  ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=BRC1H 43:23:99}, rssi=-67, timestampNanos=852829208258601, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...], isConnectable=CONNECTABLE}
2024-07-04 09:30:08.472  9402-9402  RxBle#Client            com...urope.control.assistant.debug  I  ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=BRC1H 43:23:99}, rssi=-60, timestampNanos=852830139733184, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...], isConnectable=CONNECTABLE}
2024-07-04 09:30:09.358  9402-9402  RxBle#Client            com...urope.control.assistant.debug  I  ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=BRC1H 43:23:99}, rssi=-54, timestampNanos=852831027069121, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...], isConnectable=CONNECTABLE}
2024-07-04 09:30:09.389  9402-9402  RxBle#Client            com...urope.control.assistant.debug  I  ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=BRC1H 43:23:99}, rssi=-53, timestampNanos=852831052887350, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...], isConnectable=CONNECTABLE}
2024-07-04 09:30:09.463  9402-9402  RxBle#Client            com...urope.control.assistant.debug  I  ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=BRC1H 43:23:99}, rssi=-54, timestampNanos=852831130719329, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...], isConnectable=CONNECTABLE}
2024-07-04 09:30:09.515  9402-9402  RxBle#Client            com...urope.control.assistant.debug  I  ScanResult{bleDevice=RxBleDeviceImpl{MAC='XX:XX:XX:XX:XX:XX', name=BRC1H 43:23:99}, rssi=-54, timestampNanos=852831180362141, callbackType=CALLBACK_TYPE_ALL_MATCHES, scanRecord=[...], isConnectable=CONNECTABLE}
2024-07-04 09:30:09.776  9402-14433 RxBle#ScanOperation     com...urope.control.assistant.debug  I  Scan operation is requested to stop.
2024-07-04 09:30:09.802  9402-14441 RxBle#Clie...ationQueue com...urope.control.assistant.debug  D  QUEUED   ConnectOperation(203521677)
2024-07-04 09:30:09.803  9402-9815  RxBle#Clie...ationQueue com...urope.control.assistant.debug  D  STARTED  ConnectOperation(203521677)
2024-07-04 09:30:09.804  9402-9815  RxBle#Clie...ationQueue com...urope.control.assistant.debug  I  RUNNING  ConnectOperation{MAC='XX:XX:XX:XX:XX:XX', autoConnect=true}
2024-07-04 09:30:09.816  9402-14433 RxBle#BleC...tionCompat com...urope.control.assistant.debug  V  Connecting without reflection
2024-07-04 09:30:09.825  9402-9815  RxBle#Clie...ationQueue com...urope.control.assistant.debug  D  FINISHED ConnectOperation(203521677) in 22 ms
[... cut ...] ading: https://github.com/ReactiveX/RxJava/wiki/What's-different-in-2.0#error-handling | java.util.concurrent.TimeoutException: The source did not signal an event for 10000 milliseconds and has been terminated.
2024-07-04 09:30:20.136  9402-9815  RxBle#Clie...ationQueue com...urope.control.assistant.debug  D  STARTED  DisconnectOperation(146704759)
```

I'm not intentionally disconnecting, this is just an app that worked with react-native-ble-plx 2, updated for new android platform requirements.

joanmarcs, 2024-07-04:
> Hi @joanmarcs I tried to reproduce your issue using [...]

Done here: https://github.com/dotintent/react-native-ble-plx/issues/1222

intent-kacper-cyranowski, 2024-07-04:
Hi @eliaslecomte 
Thanks for your time and logs.
I tried to reproduce it with rapid writes, but with no luck. Could you create separate issue with minimal repro, as issue seems to be different from what @joanmarcs faces.

I would also ask everyone else that needs help with issue similar issue, yet unrelated to iOS to create separate issue and provide relevant information

MahmoudNafiseh, 2024-07-29:
@intent-kacper-cyranowski I had this issue recently, and found out its due to setting the `timeout` property in the `connectToDevice` option.

What fixed it:

```
	bleManager
	.connectToDevice(deviceId, {
		autoConnect: false,
		requestMTU: 96,
		timeout: Platform.OS === 'android' ? undefined : 45000,
	})
```

Setting the `timeout` to a `45000` made my device disconnect after 45 seconds, even though I was reading/writing data.

I was on a Samsung Galaxy FE S21. Maybe this is the issue and could possibly help?

enriquezc, 2024-08-14:
Any update on when the next release will be? We're seeing this issue on Android and would love some resolution

ApplicationNinja, 2024-09-15:
I can confirm this same issue, using PLX version 3.2.1, Google Pixel 9 connected to an ESP32 device, occurs exactly at 30 seconds.

AdriBee, 2024-09-22:
Experiencing the same issue here. I am not using an ESP32, but another MCU (Dialog/Renesas one). The connection drops after about 10 seconds, then it reconnects, and it drops it again continuously. It is only happening on Android for me, iOS is working fine.

codewizardVN, 2024-10-07:
> Experiencing the same issue here. I am not using an ESP32, but another MCU (Dialog/Renesas one). The connection drops after about 10 seconds, then it reconnects, and it drops it again continuously. It is only happening on Android for me, iOS is working fine.
> 
> I recently upgraded the app to RN 0.74 and went from ble-plx 2.0 to 3.2.1. This library was working fine with the same code on Android, but now it is not.

I am having the same problem as you. Did you find a solution?

AdriBee, 2024-11-05:
Hello @chris-goodchild  @codewizardVN ! The problem I had was only with Android. On iOS, I did not experience the problem, so there could be two separate issues. I am unaware of the iOS one, but here is the fix for the Android.

Okay, here is a quick fix that you can do. This is fixed on the upcoming update on this library, but as a work around for the ones who need it now, here is what I did to fix it.

The main problem is on the BLE PLX library, and it needs to be fixed on this file:
node_modules/react-native-ble-plx/android/src/main/java/com/bleplx/adapter/BleModule.java

Change this:
-      connect = connect.timeout(timeout, TimeUnit.MILLISECONDS);
For this:
+      connect = connect.timeout(
+        Observable.timer(timeout, TimeUnit.MILLISECONDS),
+        item -> Observable.never()
+      );

Now, here is the problem. Changing this once, will work, but this change is not permanent, as when you install your node reqs again, this change will get overrided. To overcome this and make is easier, I used https://www.npmjs.com/package/patch-package. This package basically will create .patch files that will automatically override your installation every time you run a yarn install or whatever you use to install all your reqs. 

The code I am providing here is already fixed on the upcoming update of the BLE PLX. I would recommend searching across the commit history on this library to see if there is any fix for the iOS app side, and if there is, you can apply a similar process to fix that witht he patch-package as well. 

Hope this helps!

jakecurreri, 2024-11-22:
> @intent-kacper-cyranowski I had this issue recently, and found out its due to setting the `timeout` property in the `connectToDevice` option.
>
> What fixed it:
>
> ```
> 	bleManager [...]

This worked for our Android issue. Every `timeout` (in our case, 5 seconds), the application would attempt to reconnect, creating an infinite loop. Thanks @MahmoudNafiseh.

aliberski, 2025-01-22:
It seems the issue is resolved (using the timeout param). I will close it for now. Please reopen or create a new issue if it's still relevant.

closed 2025-01-22 by aliberski