dotintent/react-native-ble-plx #1199: Question

rnigro-rwb, 2024-05-09:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

With expo support added in v3.1.0, `@expo/config-plugins` has been added as a required dependency.  My project does not use expo and therefore I do not need this, but because it is a dependency of the package, it now gets pulled in.

Is it possible to utilize v3.1.0+ without requiring `@expo/config-plugins`?

madej10, 2024-05-14:
@rnigro-rwb hi! thanks for the report. we'll get back to you soon. thanks for your patience 🙏

intent-kacper-cyranowski, 2024-06-13:
Hi @rnigro-rwb 
From what I know `@expo/config-plugins` is being installed with the package (by js package manager), yet because you are not using any of plugin files the `@expo/config-plugins` is not being bundled in your final js file, and so should not affect your project.
The way of adding plugin for expo is established by expo itself.
I hope it answers your question in case of any problems please elaborate on the issue

intent-kacper-cyranowski, 2024-06-30:
Hi @rnigro-rwb 
Do you still need help with the issue?