dotintent/react-native-ble-plx #1202: Android setup

chris-goodchild, 2024-05-24:
### Prerequisites

- [X] I checked the documentation and FAQ without finding a solution
- [X] I checked to make sure that this issue has not already been filed
- [X] I'm sure that question is related to the library itself and not Bluetooth Low Energy or Classic in general. If that so, please post your question on [StackOverflow](https://stackoverflow.com/questions/tagged/react-native-ble-plx?sort=active).
- [X] I'm running the latest version

### Question

Hi, I have read the docs and seen a related issue specific to Expo but this is running a raw RN project. Is there a need to manually import and add the BlePlxPackage in Android in `MainApplication.java`? I installed the package and checked that we have all of the necessary config in our AndroidManifest.xml but when running the app on Android we are seeing the error `Cannot read property 'createClient' of null` from instantiating `BleManager`. 

I have checked your example project which contains no such manual installation setup, but I'm wondering why the module cannot be found unless we're missing some steps.

dominik-czupryna-withintent, 2024-05-26:
I tried with the latest RN template and I don't see any problems - you can check all the changes I made to run the scan: https://github.com/dominik-czupryna-withintent/i1202/commit/883bab543dd3dcbf9f12a250cfa6cd2d5073c01e 
Maybe something is cached?

chris-goodchild, 2024-06-06:
My bad, we have a custom setup and needed to manually register the module. Thanks.

closed 2024-06-06 by chris-goodchild

CSWITH89, 2025-01-22:
@chris-goodchild Hey there. Having the same issue with a bare React Native project. Are you able to share how you resolved the problem?